tuya-cloudcutter/tuya-cloudcutter #230: flashing not working anymore

WiredLife, 2022-12-23:
i tried 2 different devices but it looks like this:

> pi@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  57.37MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> e93a1d4a68a4
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> a6d43f7274e2
Step 3/10 : FROM base AS python-deps
 ---> a6d43f7274e2
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> ebec948641d2
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 0de2dcdebcd3
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 96eaa92c2410
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 8af34d853e4d
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 8af34d853e4d
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 55861fc00f23
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> ae33fff7352d
Successfully built ae33fff7352d
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] Select your custom firmware file: OpenBK7231T_UG_1.15.221.bin
 > OpenBK7231T_UG_1.15.221.bin

[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: LSC
   Immax
   KHSUIN
   KOGAN
   Kanlux
   Klarstein
   LEDVANCE
 > LSC
   Lampsero
   Lenovo
   Lightway
   Lonsonho
   Lumary
   MOES

[?] Select the article number of your device: 3004919 970710 Smart LED RGB Tunable White GU10 380Lumen
   3001700 970739 Warm White Smart Filament E27 806Lumen
[... 161 lines cut ...]
{'t': 1671786206}
[LOG] ==== End client request body ===
!!! Endpoint default response not found - tuya.device.dynamic.config.get
tuya.device.dynamic.config.get - response: {'success': True, 't': 1671786207}
[I 221223 09:03:27 web:2271] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=G2YO1PgLsCWc5hKNcLAJ&et=1&t=1671786206&v=1.0&sign=512b22fddf20df935a61c7e3dd4bb471 (10.42.42.19) 106.39ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1671786210} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[LOG] Client request PATH: /d.json - QS: a=tuya.device.dynamic.config.get&devId=G2YO1PgLsCWc5hKNcLAJ&et=1&t=1671786211&v=1.0&sign=c36cbf6f482a2d92b8c5dc3fb424e627
[LOG] ==== Client request body ===
{'t': 1671786211}
[LOG] ==== End client request body ===
!!! Endpoint default response not found - tuya.device.dynamic.config.get
tuya.device.dynamic.config.get - response: {'success': True, 't': 1671786211}
[I 221223 09:03:31 web:2271] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=G2YO1PgLsCWc5hKNcLAJ&et=1&t=1671786211&v=1.0&sign=c36cbf6f482a2d92b8c5dc3fb424e627 (10.42.42.19) 103.59ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1671786212} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1671786216} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[LOG] Client request PATH: /d.json - QS: a=tuya.device.dynamic.config.get&devId=G2YO1PgLsCWc5hKNcLAJ&et=1&t=1671786220&v=1.0&sign=f8e4594649dca87522afe54eed1e347a
[LOG] ==== Client request body ===
{'t': 1671786220}
[LOG] ==== End client request body ===
!!! Endpoint default response not found - tuya.device.dynamic.config.get
tuya.device.dynamic.config.get - response: {'success': True, 't': 1671786221}
[I 221223 09:03:41 web:2271] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=G2YO1PgLsCWc5hKNcLAJ&et=1&t=1671786220&v=1.0&sign=f8e4594649dca87522afe54eed1e347a (10.42.42.19) 98.35ms`

Cossid, 2022-12-23:
> !!! Endpoint default response not found - tuya.device.dynamic.config.get
> !!! Endpoint default response not found - tuya.device.uuid.pskkey.get
> !!! Endpoint default response not found - tuya.device.dynamic.config.get
> !!! Endpoint default response not found - tuya.device.dynamic.config.get

Indicates you are missing files from your `/device-profiles/schema/` directory.

WiredLife, 2022-12-23:
the profile was a default one, not build by myself
i think the default ones are now included in the docker image?
anyway, i learned its better to git clone into a new directory instead of using git pull

Cossid, 2022-12-23:
That directory is static, and the files should always be there.  It's not a profile, but the schema/files all profiles use.

`git pull` is usually sufficient, but we did a major restructuring of the profiles about a month ago.  They should continue to be clean after that point.

Assuming a clean checkout resolved your issue, I'm closing this issue.

closed 2022-12-23 by Cossid