tuya-cloudcutter/tuya-cloudcutter #232: GL-C-009W single color LED strip CBU BK7231N

openshwprojects, 2022-12-23:
Hello, here is another firmware dump for you:
[GL-C-009W_CBU.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10296123/GL-C-009W_CBU.zip)
This device:
[image]
Soldering access to that one is limited:
[image]

As usual, I don't need the exploit as I always flash by wires, just providing dump to help non-soldering part of the community, but I am interested to see what you can do with that bin. Which bootloader version is it, is it supportable?
Have fun!

Cossid, 2022-12-23:
This device is fully supportable.  Profile added as Tuya Generic GL-C-009W LED Dimmer

N only has a single bootloader so far, BK7231N_1.0.1, which is what this has.

closed 2022-12-23 by Cossid

openshwprojects, 2022-12-23:
Thank you for the quick reply! 
Out of curiosity, would that profile also work for the RGB and RGBCW versions of this strip?

kuba2k2, 2022-12-23:
Each firmware version needs a separate dump. If the RGB and RGBCW versions use the same firmware version, it can work. If not, another profile needs to be used.

Cossid, 2022-12-23:
If they have the same firmware version (2.0.0), yes probably.  We currently have "oem-bk7231n-strip-ty" support for 1.1.5 and 2.0.0.  If they don't work, they can be supported if you supply full firmware dumps.