tuya-cloudcutter/tuya-cloudcutter #233: Moes MS-105 1 Gang 1/2 Way Dimmer Switch Module

ElGranPepo, 2022-12-23:
I always get the weird devices!!
Device has no switches or LEDs, but it does have a very annoying buzzer!
will start as Bluetooth pairing only, need to turn switch 6 times, then six times again to get longs beeps and WIFI pairing.
CB2S Chip (BK7231N) tried all three  profiles in LightLeak and none worked.

Can the classic profiles be tested as well?

Thanks!

kuba2k2, 2022-12-23:
Classic profiles are for cloudcutter only, they don't work in the Android app yet.

What is the firmware version of that device?

ElGranPepo, 2022-12-23:
pictures:

[image]
[image]
[image]

ElGranPepo, 2022-12-23:
> Classic profiles are for cloudcutter only, they don't work in the Android app yet.
> 
> What is the firmware version of that device?

I looked in the Tuya Dev portal and i didnt see any parameter for firmware, how do i find out?

kuba2k2, 2022-12-23:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md#finding-your-device

ElGranPepo, 2022-12-23:
> https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md#finding-your-device

Thanks!  V2.0.2

kuba2k2, 2022-12-23:
There is a profile for 2.0.2. And 2.0.2 seems to be exploitable using Lightleak. Try the XOR profile, others are not usually useful.

ElGranPepo, 2022-12-23:
> There is a profile for 2.0.2. And 2.0.2 seems to be exploitable using Lightleak. Try the XOR profile, others are not usually useful.

Will try, what's the current profile name?

Cossid, 2022-12-23:
For Lightleak, it is `Lightleak - BK7231N - Variant 1 (XOR)`
For cloudcutter, you would choose `by firmware version and name` and scroll to the 2.0.2 versions.

ElGranPepo, 2022-12-23:
so after joining the device to the tuya app and removing it (wipe data) i cant get it to go into pairing mode.  not sure if there might be some kind of block in the update.

also i did a little searching a noticed something interesting on the devices i have on tuya and their versions:

Generic tuya switches  1.16
tuya bluetooth gateway: 1.01
tuya bluetooth only Recessed lamps: 4.5
tuya wifi flood lights: 1.2.16

none have the same version!  not sure if there are any known exploits for the bluetooth pairing/devices.

wiil open another dimmer switch and se if that one works.

Cossid, 2022-12-23:
> so after joining the device to the Tuya app and removing it (wipe data) i cant get it to go into WIFI pairing mode. (will only go into Bluetooth mode) not sure if there might be some kind of block in the update.

Nothing cloudcutter does should prevent you from putting the device back into wifi pairing mode.  You should be able to cloudcut a device over and over again if you're not flashing 3rd party firmware.

> also i did a little searching a noticed something interesting on the devices i have on tuya and their versions:
>
> Generic Tuya switches 1.16 Tuya Bluetooth gateway: 1.01 Tuya Bluetooth only Recessed [...]

This is completely normal and expected.  This is the reason we have many profiles in cloudcutter, because there are so many different ones.

At this point, I'm not sure what you're expecting out of this ticket, there seems to be no end-goal here.  I'm closing this issue.  Please keep issues as specific and related to cloudcutter as possible.

closed 2022-12-23 by Cossid

ElGranPepo, 2022-12-24:
End goal would be is a dump posted by me.  didn't think any of my questions or posts were unrelated to the general topic of Turya CloudCutter and hoped they would help others.  I'm still having issues dumping this device and hoping others might provide info and experience when dumping dimmer switches.

also can you confirm if the name of the ticket would be OK for the dump?

Once again thank you for putting so much time into this great project!

Cossid, 2022-12-24:
> also can you confirm if the name of the ticket would be OK for the dump?

Looks fine except the `/` is not valid in a filename, you'd need to use a something different like the word or, or a dash.

ElGranPepo, 2023-01-21:
thank you for your help, dump attached:
[MOES MS-105.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10473113/MOES.MS-105.zip)

Cossid, 2023-01-21:
A profile has been added as Moes MS-105 Dimmer Switch

ElGranPepo, 2023-01-23:
there is a MOES and a Moes under the manufacturer link,

Cossid, 2023-01-23:
Fixed, both are capitalized now (may need to delete from your /device-profiles/ directory though if cached)