tuya-cloudcutter/tuya-cloudcutter #236: Treatlife ST30 Smart LED TV Backlight System Dump

IeSTrErCHiAlIoLf, 2022-12-27:
[treatlife_st30_led_backlight_lightleak.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10310473/treatlife_st30_led_backlight_lightleak.zip)

Here is a system dump grabbed by Lightleak

IeSTrErCHiAlIoLf, 2022-12-27:
Looking for some help with a profile for cloudcutter for this device

Cossid, 2022-12-28:
bk7231tools is having an issue decrypted the app partition.  I will follow up once we are able to confirm/resolve the bk7231tools issue.

Cossid, 2022-12-28:
bk7231tools have been fixed, a profile has been added as `TreatLife ST30 Smart LED Strip Lights`

IeSTrErCHiAlIoLf, 2022-12-29:
Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: TreatLife
   Qiachip
   Rabit
   STITCH
   Spectrum
   Stitch
   Topgreener
 > TreatLife
   Tuya Generic
   UME
   UltraPro
   Woox
   iHome
   ihome

[?] Select the article number of your device: ST30 Smart LED Strip Lights
   DP20 Smart Dimmer Plug
   DS02F Fan Controller
   DS02S Dimmer Switch
   DS03 Fan and Light Smart Switch
   NL10 Bedside Lamp
   SL10 A19 RGBCT Bulb 1.0.1
   SL10 A19 RGBCT Bulb 1.0.2
   SL10 A19 RGBCT Bulb 1.0.8
   SL20
   SS01S Single Pole Smart Light Switch
   ST10 Lightstrip
 > ST30 Smart LED Strip Lights

[?] Select the firmware version and name: 1.0.2 - BK7231T / HYS_03_010_RGB_xhsx0clcbovz5xre
 > 1.0.2 - BK7231T / HYS_03_010_RGB_xhsx0clcbovz5xre

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
..
Found access point name: "SmartLife-C4A4", trying to connect..
Device 'wlan0' successfully activated with '877c6c31-9354-40cf-89cd-1f5d05945251'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/s9oXzMfxXEyd.deviceconfig
Saved device config in /work/configured-devices/s9oXzMfxXEyd.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
...........
Found access point name: "SmartLife-C4A4", trying to connect..
Device 'wlan0' successfully activated with '877c6c31-9354-40cf-89cd-1f5d05945251'.
Connected to access point.
====================================================================
[!] The profile you selected did not result in a successful exploit.
====================================================================

Cossid, 2022-12-29:
We're finding that devices that have already had a firmware upgrade done via Tuya are not exploiting properly.  We'll look into it to see if we can find a cause.

IeSTrErCHiAlIoLf, 2022-12-29:
ok. I did not choose to upgrade firmware ever, is there a way I can tell if one happened?

Cossid, 2022-12-29:
Not really aside from examining the dump.  The indicator for us is that there is no RBL indicator, so the app partition may be in a slightly different location than expected.

kuba2k2, 2022-12-29:
The app partition can't be in a different location. We have a dump from Lightleak, no if the profile doesn't work, it means it has wrong offsets. Maybe the profile-builder can't properly handle this particular binary, for some reason.

Cossid, 2022-12-29:
It's identifying the correct offsets relative the the decrypted app partition.  I can verify they're matching the same thing they would match on a normal app partition.  This certainly appears to be exclusive to firmwares that have OTA'd from Tuya (see also https://github.com/tuya-cloudcutter/lightleak/issues/3 ) and possibly having some sort of different offset.

I have a test case from my own light strip that had the 1.0.0/1.1.0 update process I will test later (I serial flashed that one originally).

Cossid, 2022-12-29:
Can you try one more time?  Sometimes the old SSID hangs around after exploit, and the exploiting device detects it and connects quick enough to properly fail it seems.

Please try one more time and post the full log again.

IeSTrErCHiAlIoLf, 2022-12-29:
Success! I had to try a few more times, but this profile did finally work! I am not on OpenBeken and working on configuring it. Thank you @Cossid

closed 2022-12-29 by IeSTrErCHiAlIoLf