tuya-cloudcutter/tuya-cloudcutter #237: Treatlife SK50 Smart Plug

rickyelopez, 2022-12-27:
Hey again,

Just bought some of these smart plugs and looks like the `run_detach` script works with the `1.1.8 Atarm WHS-2 Infrared Switch` profile. However, I am not able to flash firmware with the `run_flash` script for some reason.

Pretty much everything goes as expected until I start seeing traffic in the terminal after starting the AP, at which point I see a bunch of traffic and then
```
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1672172858} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
```
getting printed a few times, more traffic, then repeat. Eventually the script exits, but the firmware on the plug does not get flashed...

Any suggestions? I'm going to try a few other device profiles I guess, but I don't have any other ideas. I'll also try another plug from the 4-pack.

Also, I still have to use the `setup_checks.sh` script from tuya-cloudcutter before attempting to use this script though, otherwise when the cloudcutterflash AP is started, the devices never connect. Would be great if the check for port 53 could be added to this tool.

rickyelopez, 2022-12-27:
Full log from `run_flash.sh`:
```
wlp0s20f3: interface state UNINITIALIZED->ENABLED
wlp0s20f3: AP-ENABLED
Dec 27 20:34:06 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 27 20:34:06 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlp0s20f3) a0:92:08:05:d5:b6
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 tags: wlp0s20f3
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlp0s20f3) 10.42.42.17 a0:92:08:05:d5:b6
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  2
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlp0s20f3) a0:92:08:05:d5:b6
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 tags: wlp0s20f3
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlp0s20f3) 10.42.42.17 a0:92:08:05:d5:b6
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  2
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 DHCPREQUEST(wlp0s20f3) 10.42.42.17 a0:92:08:05:d5:b6
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 tags: wlp0s20f3
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 DHCPACK(wlp0s20f3) 10.42.42.17 a0:92:08:05:d5:b6 wlan0
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  5
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 27 20:34:09 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 27 20:34:28 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.17
Dec 27 20:34:28 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
[I 221227 20:34:28 web:2239] 200 POST /v2/url_config (10.42.42.17) 48.18ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.active&et=1&t=36&uuid=k8siurL7RXLC&v=4.4&sign=1162d484718b18a5c97ed2d7a9a73933
[LOG] ==== Client request body ===
{'token': 'AAAAAA', 'softVer': '1.1.8', 'productKey': 'keyjup78v54myhan', 'protocolVer': '2.2', 'baselineVer': '40.00', 'productKeyStr': 'keyjup78v54myhan', 'devId': '9tYHKM6EyjSAOqVNk3Sp', 'hid': 'a0920805d5b6', 'modules': '[{"otaChannel":9,"softVer":"1.1.8","online":true}]', 'devAttribute': 579, 'cadVer': '1.0.3', 'cdVer': '1.0.0', 'options': '{"isFK":true,"otaChannel":0}', 't': 36}
[LOG] ==== End client request body ===
tuya.device.active - response: {'result': {'schema': '[{"type":"obj","mode":"rw","property":{"type":"bool"},"id":1},{"type":"obj","mode":"rw","property":{"min":0,"max":86400,"scale":0,"step":1,"type":"value"},"id":9},{"type":"obj","mode":"rw","property":{"range":["off","on","memory"],"type":"enum"},"id":38}]', 'devId': 'hQZ2gEOaky8rPu8mlDl9', 'resetFactory [... cut ...]
tuya.device.dynamic.config.get - response: {'result': {'ackId': '0-0', 'validTime': 1800, 'time': 1672173298, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}}, 't': 1672173298, 'success': True}
[I 221227 20:34:58 web:2239] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=hQZ2gEOaky8rPu8mlDl9&et=1&t=1672173298&v=2.0&sign=2645a9ee78f79840878d064f367ac060 (10.42.42.17) 8.53ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1672173303} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
```

Cossid, 2022-12-28:
It sounds like you are on an older version, as `run_detach.sh` and `run_flash.sh` have been deprecated in favor of the combined `tuya-cloudcutter.sh` and are using a response file that hasn't been used since we migrated from legacy to classic profiles.

Please update and provide full logs on the current scripts.

rickyelopez, 2022-12-28:
Oh damn! I didn't even think to check, my bad. I'll update and try again. Thanks!

rickyelopez, 2022-12-28:
Ok so with the latest master I'm now getting a padding error once I get to the flashing state.

```
wlp0s20f3: interface state UNINITIALIZED->ENABLED
wlp0s20f3: AP-ENABLED
Dec 28 02:33:51 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 28 02:33:51 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlp0s20f3) a0:92:08:05:e9:44
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 tags: wlp0s20f3
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  2
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlp0s20f3) a0:92:08:05:e9:44
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 tags: wlp0s20f3
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  2
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 DHCPREQUEST(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 tags: wlp0s20f3
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 DHCPACK(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44 wlan0
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  5
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 28 02:33:54 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 28 02:33:56 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.25
Dec 28 02:33:56 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
[LOG (Client)] Request: HTTPServerRequest(protocol='https', host='h3.iot-dns.com', method='POST', uri='/v2/url_config', version='HTTP/1.1', remote_ip='10.42.42.25')
[LOG (Client)] ==== Request body ===
b'{"region":"AA","env":"","uuid":"ph9xxGN95x5L","ts":18,"sign":"V/A5CS1pMN42y51vgAuY1lLegCfd68lUN8yCGouj6P4=","config":[{"key":"httpUrl","need_ca":false},{"key":"mqttUrl","need_ca":false},{"key":"httpsPSKUrl","need_ca":false},{"key":"mqttsPSKUrl","need_ca":false}]}'
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {"caArr":null,"httpUrl":{"addr":"http://10.42.42.1/d.json","ips":["10.42.42.1"]},"mqttUrl":{"addr":"10.42.42.1:1883","ips":["10.42.42.1"]},"ttl":600}
[I 221228 02:33:56 web:2271] 200 POST /v2/url_config (10.42.42.25) 48.44ms
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying with other key choice
[... cut ...]
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.
```
The last bit just repeats over and over until some limit is hit I guess, then the script feezes up. Any suggestions for this?

Forgot to mention in the OP that I confirmed v1.1.8 from the Smart Life app.

Cossid, 2022-12-28:
Please clear out your `device-profiles` directory, git update and try again.  It still sounds like there are incompatible files present.

rickyelopez, 2022-12-28:
No dice :( totally clean repo (`git clean -dffx`), pulled latest hash (`b833a6250dad439cdc75aa0e`), still getting the same padding error (I think it's the same?)

```
[I 221228 02:59:34 web:2271] 200 POST /v2/url_config (10.42.42.25) 49.24ms
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying is false - bailing out
[E 221228 02:59:34 web:1798] Uncaught exception POST /d.json?a=tuya.device.active&et=1&t=37&uuid=ph9xxGN95x5L&v=4.4&sign=e390c84fe46e479bbb4f7dd6f228f52d (10.42.42.25)
    HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.active&et=1&t=37&uuid=ph9xxGN95x5L&v=4.4&sign=e390c84fe46e479bbb4f7dd6f228f52d', version='HTTP/1.1', remote_ip='10.42.42.25')
    Traceback (most recent call last):
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "/src/.venv/lib/python3.9/site-packages/tornado/web.py", line 1711, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      File "/src/cloudcutter/protocol/handlers.py", line 96, in post
        request_body = self.__decrypt_request_body(key_choice)
      File "/src/cloudcutter/protocol/handlers.py", line 137, in __decrypt_request_body
        decrypted = self.cipher.decrypt(body, key_choice).decode('utf-8')
      File "/src/cloudcutter/crypto/tuyacipher.py", line 30, in decrypt
        return self.__decrypt_with_retry(data, key_choice)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 59, in __decrypt_with_retry
        return self.__decrypt_with_retry(data, key_choice, False)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 56, in __decrypt_with_retry
        raise e
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.
```

Cossid, 2022-12-28:
What profile are you using?  The logs you are providing show only the very end of the process.  Full logs would be more helpfu.

rickyelopez, 2022-12-28:
Sorry, here's more. I still only included a few of the errors since I think it's just the same thing repeated, but lmk if you want more:

```
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.1.8 - BK7231T / oem_bk7231s_rnd_switch
   1.1.6 - BK7231T / oem_bk7231s_strip_ty
   1.1.6 - BK7231T / oem_bk7231s_switch1_zero_kjd
   1.1.7 - BK7231N / oem_bk7231n_plug
   1.1.7 - BK7231T / oem_bk7231s_light_pir_ty
   1.1.71 - BK7231T / bk7231t_common_user_config_ty
   1.1.8 - BK7231N / oem_bk7231n_plug
 > 1.1.8 - BK7231T / oem_bk7231s_rnd_switch
   1.1.9 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.2.1 - BK7231N / oem_bk7231n_control_switch
   1.2.16 - BK7231N / oem_bk7231n_light_ty
   1.2.7 - BK7231N / oem_bk7231n_light_ty
   1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.3.2 - BK7231T / oem_bk7231s_dltj

[?] Select the brand of your device: TreatLife
   Aoycocr
 > TreatLife

[?] Select the article number of your device: SS01S Single Pole Smart Light Switch
 > SS01S Single Pole Smart Light Switch

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Redirecting to /bin/systemctl start NetworkManager.service
Scanning for open Tuya SmartLife AP
....
Found access point name: "A-E944", trying to connect..
Device 'wlp0s20f3' successfully activated with '12004754-6753-4e08-bd55-afad88b6eb8f'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/nybskw7knf8U.deviceconfig
Saved device config in /work/configured-devices/nybskw7knf8U.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Redirecting to /bin/systemctl start NetworkManager.service
Scanning for open Tuya SmartLife AP
.......
Found access point name: "A-E944", trying to connect..
Device 'wlp0s20f3' successfully activated with '12004754-6753-4e08-bd55-afad88b6eb8f'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Cutting device off from cloud..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
Redirecting to /bin/systemctl stop NetworkManager.service
Using WLAN adapter: wlp0s20f3
Dec 28 02:59:09 dnsmasq[15]: started, version 2.80 cachesize 150
Dec 28 02:59:09 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Dec 28 02:59:09 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Dec 28 02:59:09 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlp0s20f3
Dec 28 02:59:09 dnsmasq[15]: read /etc/hosts - 2 addresses
Configuration file: /dev/stdin
Using interface wlp0s20f3 with hwaddr 34:7d:f6:65:25:57 and ssid "cloudcutterflash"
wlp0s20f3: interface state UNINITIALIZED->ENABLED
wlp0s20f3: AP-ENABLED
Dec 28 02:59:13 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 28 02:59:13 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec 28 02:59:16 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlp0s20f3) a0:92:08:05:e9:44
Dec 28 02:59:16 dnsmasq-dhcp[15]: 1220565492 tags: wlp0s20f3
Dec 28 02:59:16 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44
Dec 28 02:59:16 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
[... 122 lines cut ...]
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "/src/.venv/lib/python3.9/site-packages/tornado/web.py", line 1711, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      File "/src/cloudcutter/protocol/handlers.py", line 96, in post
        request_body = self.__decrypt_request_body(key_choice)
      File "/src/cloudcutter/protocol/handlers.py", line 137, in __decrypt_request_body
        decrypted = self.cipher.decrypt(body, key_choice).decode('utf-8')
      File "/src/cloudcutter/crypto/tuyacipher.py", line 30, in decrypt
        return self.__decrypt_with_retry(data, key_choice)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 59, in __decrypt_with_retry
        return self.__decrypt_with_retry(data, key_choice, False)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 56, in __decrypt_with_retry
        raise e
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.
```

Cossid, 2022-12-28:
You're using the wrong profile, you should be using the plug one.  The only reason you're not getting a better error is because in the past, you have exploited the device with the correct profile, so it now has the A- prefix.

If you did that with a new device, it would fail earlier saying the exploit was not successful.

rickyelopez, 2022-12-28:
Hmmm are you sure? The plug profile says BK7231N but from what I can tell this device uses WB2S which is a BK7231T? Yeah I probably shouldn't have used that profile to begin with, but I didn't notice it was the wrong chip...

Cossid, 2022-12-28:
Unfortunately, there is no way for me to know that.  If you used oem_bk7231n_plug to exploit it in the first place, then your device likely has a BK7231N inside.  It is not uncommon for the same model to have a version with either board inside.

Whatever profile actually changes the device to an A- AP prefix is what is inside your device.  I can't determine it from your previous runs.  But if you get bad padding, that means the profile you're currently using is not correct.

rickyelopez, 2022-12-28:
Interesting... Ok so if the detach step passed then that confirms that it uses the chip from that profile? Alrighty, I'll build some fw for the BK7231N and try the flash again with the original profile.

Yeah I'm aware of that, I just don't have anything to go on other than what I found online since I haven't opened one of these. If I mess it up, I'll have to open it and will figure out the correct chip that way I guess.

Cossid, 2022-12-28:
Correct.  You can't detach with the wrong profile, it will fail and just not work.  Whatever is associated with what you used to successfully detach is what the firmware on your device actually is.

rickyelopez, 2022-12-28:
Amazing, switched to the cb2s and it flashed successfully, now to find out if that's actually the right module :laughing:

Thanks for your help, _again_! Looks like you should be good to add the profiles for these two devices.

Cossid, 2022-12-28:
We only add profiles if we have the correct schema as well, otherwise users can just use the existing profiles matching by firmware version/name as you already did.

The correct schema is the only thing that verifies they are working correctly in detached mode without 3rd party firmware.

rickyelopez, 2022-12-28:
Hmmm I may have spoken too soon, looks like the flashing is still failing, differently this time though:

```
65492 DHCPDISCOVER(wlp0s20f3) a0:92:08:05:e9:44
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 tags: wlp0s20f3
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 DHCPOFFER(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 next server: 10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  1 option: 53 message-type  2
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 client provides name: wlan0
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 DHCPDISCOVER(wlp0s20f3) a0:92:08:05:e9:44
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 tags: wlp0s20f3
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 DHCPOFFER(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 next server: 10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  1 option: 53 message-type  2
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 client provides name: wlan0
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 DHCPREQUEST(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 tags: wlp0s20f3
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 DHCPACK(wlp0s20f3) 10.42.42.25 a0:92:08:05:e9:44 wlan0
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 next server: 10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  1 option: 53 message-type  5
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 58 T1  6h
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec 28 03:56:13 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec 28 03:56:49 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.25
Dec 28 03:56:49 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
[LOG (Client)] Request: HTTPServerRequest(protocol='https', host='h3.iot-dns.com', method='POST', uri='/v2/url_config', version='HTTP/1.1', remote_ip='10.42.42.25')
[LOG (Client)] ==== Request body ===
b'{"region":"AA","env":"","uuid":"gkXJDaaofxeH","ts":58,"sign":"0742dFkJgzCo22JbkUlATiJB26EE6cBH8NO2nBPdiiA=","config":[{"key":"httpUrl","need_ca":false},{"key":"mqttUrl","need_ca":false},{"key":"httpsPSKUrl","need_ca":false},{"key":"mqttsPSKUrl","need_ca":false}]}'
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {"caArr":null,"httpUrl":{"addr":"http://10.42.42.1/d.json","ips":["10.42.42.1"]},"mqttUrl":{"addr":"10.42.42.1:1883","ips":["10.42.42.1"]},"ttl":600}
[I 221228 03:56:49 web:2271] 200 POST /v2/url_config (10.42.42.25) 49.19ms
[LOG (Server)] Response:  {'result': {'schema': '[{"type":"obj","mode":"rw","property":{"type":"bool"},"id":1},{"type":"obj","mode":"rw","property":{"min":0,"max":86400,"scale":0,"step":1,"type":"value"},"id":9},{"type":"obj","mode":"rw","property":{"range":["off","on","memory"],"type":"enum"},"id":38}]', 'devId': 'bid87KOXVcniKnO2Y61p', 'resetFactory': False, 'timeZone': '+02:00', 'capability': 1025, 'secKey': 'ocF7KiSEgRs5Miml', 'stdTimeZone': '+01:00', 'schemaId' [... cut ...]
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=567587-589983/589984
[I 221228 03:57:19 web:2271] 206 GET /files/bk7231n_app.ota.ug.bin (10.42.42.25) 0.33ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=567587-589983/589984
[I 221228 03:57:23 web:2271] 206 GET /files/bk7231n_app.ota.ug.bin (10.42.42.25) 0.37ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=567587-589983/589984
```

Looks like it gets all the way to the end, but the device keeps running stock firmware?

Cossid, 2022-12-28:
This isn't a cloudcutter issue (it's doing everything it can correctly at this point).  This is usually a wifi/performance issue.  Make sure the target device is as close to the cloudcutter device as possible.

rickyelopez, 2022-12-28:
Gotcha. I'm literally right next to it. Will keep trying, thanks!

rickyelopez, 2022-12-28:
nice yeah a few more retries did it. Thanks!

closed 2022-12-28 by rickyelopez