tuya-cloudcutter/tuya-cloudcutter #239: unknown pattern error on build_profile

blakadder, 2022-12-28:
```
+] Searching for known exploit patterns
Traceback (most recent call last):
  File "D:\github\tuya-cloudcutter\profile-building\build_profile.py", line 45, in <module>
    haxomatic.run(app_file)
  File "D:\github\tuya-cloudcutter\profile-building\haxomatic.py", line 150, in run
    walk_app_code()
  File "D:\github\tuya-cloudcutter\profile-building\haxomatic.py", line 87, in walk_app_code
    raise RuntimeError('Unknown pattern, please open a new issue and include the bin.')
RuntimeError: Unknown pattern, please open a new issue and include the bin.
```
[Hihome_WPP---16R-Plug.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10311353/Hihome_WPP---16R-Plug.zip)

Cossid, 2022-12-28:
bk7231tools is having an issue decrypted the app partition.  I will follow up once we are able to confirm/resolve the bk7231tools issue.

Cossid, 2023-01-03:
Closing as bk7231tools have been fixed, and you have already PR'd a profile for this device.

closed 2023-01-03 by Cossid