tuya-cloudcutter/tuya-cloudcutter #240: New device, no error but doesn't seem to flash.

protocol113, 2022-12-29:
The script completes fully and shows that it flashed the latest openbeken Cctr binary but the device reboots back to the tuya firmware after the script closes. 
[KMC_smart-plug-mini-107.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10317569/KMC_smart-plug-mini-107.zip)

Cossid, 2022-12-29:
You're going to have to provide some information.  Full logs of what is happening would be most helpful.

Cossid, 2022-12-30:
Profile has been added as KMC 30153 Smart Plug Mini.

Make sure you are flashing the correct firmware, this device is BK7231T.  According to Amazon reviews, there is also a BK7231N version of the same model.  Closing this issue due to lack of information.  Feel free to re-open if you provide more information.

closed 2022-12-30 by Cossid