tuya-cloudcutter/tuya-cloudcutter #241: Cloudcutter run without Access Point mode.

thanasiskalos, 2022-12-30:
Hello

I have successfully run cloudcutted for a Tuya RGB bulb. The issue is that the DPS values seems to be wrong as I might have selected a wrong profile. I have figure out that by running various tinytuya commands as well as try to add it on localtuya on Home Assistant. 

Since the device cannot be reset  or enter the AP mode any more  is there any way to run tuya-cloucutter again without the WIFI AP section ? 
Since I have ID, Local Key of the device and the IP I am assuming I can run a firmware change directly.
Is that possible?
 
Best Regards and keep up the good work.

Cossid, 2022-12-30:
If you didn't flash a 3rd part firmware, you should be able to put the device back into AP mode like normal and cut it again using another profile/schema.

thanasiskalos, 2022-12-30:
Hello
I have not flash any f/w . The reset process I guess is the same ( 6 On/OFF) ?

Regards

Cossid, 2022-12-30:
Correct, you're still running Tuya firmware, so the process for putting it into AP mode is unchanged.  You can cut as many times as you like,  you just can't hook back up to Smart Life because CloudCutter changes the API address to something that can't work to disconnect it from the cloud.

You usually have to pause for a couple seconds after 3x, then finish with 3x more.

Cossid, 2023-01-03:
Closing as answered.
To reiterate, cutting a device without flashing a 3rd party firmware does not change the procedure or ability to put a device in AP mode.  If you experience an issue after cutting, such as an incorrect schema, you can cut the device again using the same procedure originally used.

closed 2023-01-03 by Cossid