tuya-cloudcutter/tuya-cloudcutter #243: CB3S single switch module with RF with amazon link

openshwprojects, 2023-01-01:
Hello,
a firmware dump from our user.
This device:
https://www.amazon.de/gp/product/B08L65N7BW/ref=ppx_yo_dt_b_asin_image_o01_s00?ie=UTF8&th=1
[image]
[image]
Real photo:
[image]

There are supposedly two versions of the dump, can anyone check?
Here is full attachment:

[cb3s_amazon_switch.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10329241/cb3s_amazon_switch.zip)

Cossid, 2023-01-01:
A new device profile has been added as Tuya Generic 1CH Relay With RF

The two dumps appeared to be the same as best I could tell.  However, this was actually a different SDK identifier, and I would appreciate if a user could test that the cloudcutter profile actually works, as it is unverified.

Cossid, 2023-01-03:
Closing as complete, as the original user likely won't be able to test/verify this profile.  If anybody tries it in the future, please comment here or start a new issue.

closed 2023-01-03 by Cossid

openshwprojects, 2023-01-07:
Sorry, I didn't look here, but it's as you said. I don't think he'll be willing to test, the non-technical people even don't have USB AP dongles needed for that. 
Thank you for your service.

bohregard, 2023-08-08:
I have this exact device and tried the profile `tuya-generic-1ch-relay-with-rf` and it doesn't complete successfully.

Here is the output:

```
Selected Device Slug: tuya-generic-1ch-relay-with-rf
Selected Profile: oem-bk7231n-control-switch-1.1.5-sdk-2.3.3-40.00
Selected Firmware: ESPHome-Kickstart-v23.07.13_bk7231n_app.ota.ug.bin

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
....
Found access point name: "SmartLife-E8A8", trying to connect...
Device 'wlan0' successfully activated with '12e30391-6384-46ca-a999-d0332624be9f'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/VkLbsnHqNuJs.deviceconfig
Saved device config in /work/configured-devices/VkLbsnHqNuJs.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
...
Found access point name: "SmartLife-E8A8", trying to connect...
Device 'wlan0' successfully activated with 'd28aa715-cfe8-4539-ac9e-3fab127582dc'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
```

Cossid, 2023-08-08:
@bohregard That means your device has a different firmware.  You need to start by finding out which firmware you device does have: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has