tuya-cloudcutter/tuya-cloudcutter #246: Tuya Temp Sensor

Liionboy, 2023-01-03:
I firmware dump from Tuya Temp Sensor:
[temp_sensor.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10338199/temp_sensor.zip)

https://www.aliexpress.com/item/1005004857160939.html?spm=a2g0o.order_list.order_list_main.49.5c461802voyAJ0

[image]

Cossid, 2023-01-03:
This has been added as Tuya Generic Temperature Sensor.

Note:  This is compiled with the low-powered version of Tuya Firmware.  You will likely get poor battery performance from 3rd party firmware without power saving features.

closed 2023-01-03 by Cossid

Liionboy, 2023-01-03:
when i try to flash i got this error:

Found access point name: "SmartLife-FA9D", trying to connect..
Device 'wlp0s20f3' successfully activated with 'afcd5a0a-8f13-41ba-81f9-987e7967d7e1'.
Error: SmartLife-FA9D - no such connection profile.

Cossid, 2023-01-03:
Are you running elevated?  You must run with sudo for network modifications to take.
NetworkManager is also required.

Liionboy, 2023-01-03:
yes , i am running with sudo, i will try with root to see if work

Liionboy, 2023-01-03:
root@adrian-huawei-d16:/home/adrian/tuya-cloudcutter# ./tuya-cloudcutter.sh 
Checking UDP port 53... Occupied by systemd-resolve with PID 771621.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  23.69MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 70fa42c638b3
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 0f208bfac20a
Step 3/10 : FROM base AS python-deps
 ---> 0f208bfac20a
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> bb81d4a3af8a
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> c3c82cbcdf94
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 1435d69885ee
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 953c01e36efb
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 953c01e36efb
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 10b426718394
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 17bb88dd3e09
Successfully built 17bb88dd3e09
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.276.bin
   OpenBK7231N_UG_1.15.275.bin
 > OpenBK7231N_UG_1.15.276.bin

[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.1.11 - BK7231N / oem_bk7231n_temp_hum_sensor
   1.0.8 - BK7231T / oem_bk7231s_ty_ffc_db_ldv
   1.0.9 - BK7231N / soc_bjg
   1.0.9 - BK7231T / oem_bk7231s_ty_ffc_db_ldv
   1.1.0 - BK7231T / HYS_03_018_bk7231t_tjsama0rocl95kdn
   1.1.0 - BK7231T / oem_bk7231s_rnd_switch
   1.1.1 - BK7231T / HYS_03_018_bk7231t_a4y9plclmzpct6ol
 > 1.1.11 - BK7231N / oem_bk7231n_temp_hum_sensor
   1.1.2 - BK7231T / oem_bk7231s_control_switch
   1.1.2 - BK7231T / oem_bk7231s_light_ty
   1.1.2 - BK7231T / oem_bk7231s_rnd_switch
   1.1.2 - BK7231T / oem_bk7231s_ty_ffc_db_ldv
   1.1.2 - BK7231T / oem_bk7231t_light3_laser_nanxin
   1.1.3 - BK7231T / oem_bk7231s_double_switch_rmd

[?] Select the brand of your device: Tuya Generic
 > Tuya Generic

[?] Select the article number of your device: Temperature Sensor
 > Temperature Sensor

Selected Device Slug: tuya-generic-temperature-sensor
Selected Profile: oem-bk7231n-temp-hum-sensor-1.1.11-sdk-2.3.1-40.00
Selected Firmware: /work/custom-firmware/OpenBK7231N_UG_1.15.276.bin
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP

Found access point name: "SmartLife-FA9D", trying to connect..
Device 'wlp0s20f3' successfully activated with 'afcd5a0a-8f13-41ba-81f9-987e7967d7e1'.
Error: SmartLife-FA9D - no such connection profile.
..
Found access point name: "SmartLife-FA9D", trying to connect..
Device 'wlp0s20f3' successfully activated with 'afcd5a0a-8f13-41ba-81f9-987e7967d7e1'.
Error: SmartLife-FA9D - no such connection profile.
.
Found access point name: "SmartLife-FA9D", trying to connect..
Device 'wlp0s20f3' successfully activated with 'afcd5a0a-8f13-41ba-81f9-987e7967d7e1'.
Error: SmartLife-FA9D - no such connection profile.
..
Found access point name: "SmartLife-FA9D", trying to connect..
Device 'wlp0s20f3' successfully activated with 'afcd5a0a-8f13-41ba-81f9-987e7967d7e1'.
Error:

Cossid, 2023-01-03:
It sounds like you likely have NetworkManger installed, but not activated.  The error is coming from the commands we are sending to nmcli which is the command line interface for NetworkManager.

https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md Step 3 (this part isn't actually specific to raspberry pi)

You might also need the `-r` attribute.

Liionboy, 2023-01-03:
i am not using raspberry pi, i am using my laptop, but on other plugs work great when flashed about 3 weeks ago

Cossid, 2023-01-03:
I can't supply any help beyond that, it isn't cloudcutter code failing, but your underlying OS.  You may need to supply a `-w` argument if you have more than 1 wifi adapter.

The error you are getting is simply the command
`nmcli dev wifi connect SmartLife-FA9D`
failing to connect.

Liionboy, 2023-01-03:
Is worked with -r, thanks

Liionboy, 2023-01-03:
now i need to figured out how to config to report temp, humidy and battery lvl :D