tuya-cloudcutter/tuya-cloudcutter #253: TreatLife SK50 BK7231T System Dump

IeSTrErCHiAlIoLf, 2023-01-09:
[TreatLife_SK50_BK7231T.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10375287/TreatLife_SK50_BK7231T.zip)

Here is a system dump grabbed by BK7231Flasher.
As I mentioned in side conversations, I was not able to pull this dump using lightleak after many tries.
I ripped one of the plugs open and took out the chip.

Looking for some help with a profile for cloudcutter for this device

kuba2k2, 2023-01-09:
Please consider using [bk7231tools](https://github.com/tuya-cloudcutter/bk7231tools) or [ltchiptool](https://github.com/libretuya/ltchiptool) to acquire full firmware dumps. BK7231Flasher (the new GUI tool) produces binaries that don't represent the physical Flash structure, which makes analyzing it harder.

IeSTrErCHiAlIoLf, 2023-01-09:
I can try. I had a windows machine available easily so I pulled this one.

kuba2k2, 2023-01-09:
You don't need to dump this one again, because we can fix this by hand for a single dump, but I'm just giving future advice to avoid invalid dumps floating around.

openshwprojects, 2023-01-09:
It's just strange that you spread misinformation like "fixing by hand" and act like it's a big deal while in reality it's just a line or two of code.... 1 second deal

Cossid, 2023-01-09:
You cannot use the dumps to flash back without using very specific and misleading offsets, which is why they are considered incorrect.  Anybody who thinks they have a valid backup will be greatly troubled if they actually try to use them as such.

Yes, they can be dissected by tools easily, but that doesn't make them useful as backups.

This device has been added as `TreatLife SK50 Smart Plug`

closed 2023-01-09 by Cossid

kuba2k2, 2023-01-09:
Also, it's important to know that flashing a full, 2MiB, bl-app-swapped dump at 0x11000 will go past 0x200000, and **overwrite the bootloader**. Which may not end very well on BK7231T.

IeSTrErCHiAlIoLf, 2023-01-09:
Thank you @Cossid , this works great!

openshwprojects, 2023-01-10:
@kuba2k2  @Cossid I still strongly diseagree with whole "invalid flash! so hard to use!" @kuba2k2 thing, because doing a swap (or adding a commandline swap) takes 2 seconds, so it basically takes more time to complain about than to handle it in code (you could even detect it automatically), but since it's your project then I decided to **abandon bkWriter 1.60** standard in favour of yours. It kinda breaks things for me because we have a legacy tradition of using UA binaries to write at 0x11000 of T and we wanted our reads to start at 0x11000 at T, but... whatever.
**Starting with today's update, GUI tool will provide binaries according to your standard.**
Can you take a moment to verify if that's what you want, just to make sure you're happy with that format and I understood you well?
[test_flashes.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10381953/test_flashes.zip)
Both binaries have RBL tag at 69 530 offset. 
Thank you for your continued service.
@kuba2k2 regarding overwriting bootloader, you are correct. The GUI flash tool prevents

Cossid, 2023-01-10:
Unfortunately, no, that T image is still incorrect.  The general solution has been instead of starting at 0x0 or 0x11000, you start at 0x200000 and read through 0x400000 and it will wrap around back to 0x0 and not fail.  Beken attempts to protect reading their bootloader, but only in simple ways (blocking 0x0-0x10FFF reads).

Yes, we can identify where things are in the dump, but that doesn't make the dump correct.  We're trying to make sure that dumps reflect their actual positioning on actual flash.  Having to use different offsets everywhere for every single file is a mess.

kuba2k2, 2023-01-10:
> Unfortunately, no, that T image is still incorrect

The last ZIP he attached (test_flashes.zip) looks okay, where do you see it's incorrect?

Cossid, 2023-01-10:
> The last ZIP he attached (test_flashes.zip) looks okay, where do you see it's incorrect?

0xfff9a: FAILED TO PARSE - Partition app does not have enough bytes for payload
Missing app RBL container. Using a scan pattern instead

openshwprojects, 2023-01-12:
Maybe because it had obk flashed?

Here's a (hopefully) unmodified Tuya dump made after changes from WB2S donated to me by user:
[WB2S_donated_by_user_uknown_content_BK7231T_QIO_2023-12-1--12-26-39.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10401133/WB2S_donated_by_user_uknown_content_BK7231T_QIO_2023-12-1--12-26-39.zip)

kuba2k2, 2023-01-12:
The problem is most likely related to OBK having a wrong size of output images. If you take a look at your T dump of OBK, you'll see the app RBL header at around 0x100000, whereas in standard Tuya firmwares it's at around 0x129F00. OBK is configured to output smaller partitions (not sure why) around 900 KiB, and Tuya uses something like 1056 KiB instead.

Anyway, if you want to check if your flasher outputs correct images, dump any T chip with the flasher, and dump the same T chip with bk7231tools or ltchiptool. Then compare both binary files - they should be identical (unless the firmware modifies parts of the storage data with each boot, then they might differ slightly).

Cossid, 2023-01-12:
Yes, that WB2S QIO dump appears correct.

openshwprojects, 2023-01-12:
Thank you for your continued assistance. Starting from now, you will get a QIO binaries from our tools. Keep on good work on Cloudcutter!