tuya-cloudcutter/tuya-cloudcutter #254: Not an UG file

sandermeis, 2023-01-10:
I tried the last 5 releases from openbeken, all UG files. Trying to flash an 7231N on a raspberry pi.

> pi@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -v
./tuya-cloudcutter.sh: illegal option -- v
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  22.48MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 4c93b56bfe17
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 8a45702c449f
Step 3/10 : FROM base AS python-deps
 ---> 8a45702c449f
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> e2b544ebe38e
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 92a3d6fac84c
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> a4c98fd58a2b
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> af72225ac5b3
Step 8/10 : FROM python-deps AS cloudcutter
 ---> af72225ac5b3
Step 9/10 : COPY src /src
 ---> Using cache
 ---> f0e79a1313da
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> d272f304abf8
Successfully built d272f304abf8
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.305.bin
   OpenBK7231N_UG_1.15.302.bin
   OpenBK7231N_UG_1.15.303.bin
   OpenBK7231N_UG_1.15.304.bin
 > OpenBK7231N_UG_1.15.305.bin

Firmware /work/custom-firmware/OpenBK7231N_UG_1.15.305.bin is not a UG file.

Cossid, 2023-01-10:
Something in the way you are downloading the file must be incorrect, as the file you listed, `OpenBK7231N_UG_1.15.305.bin`, from their site is indeed a valid file.  We're simply checking the first 4 bytes of the file to make sure it has the header a UG file should have, which their file does.  Something in your download process is resulting in an altered file.

sandermeis, 2023-01-10:
I am downloading with curl -O url

Cossid, 2023-01-10:
Are you getting a 0 byte file?

Cossid, 2023-01-10:
Closing this ticket, as I verified curl -O provides a 0 byte file and adding -L works, and CloudCutter is working as expected (and saved you from making a mistake).

closed 2023-01-10 by Cossid

sandermeis, 2023-01-10:
My bad, that was indeed the case. Rookie mistake.

The script is now however hanging on :

>Device 'wlan0' successfully activated with '24ace973-b4f9-4806-aad8-c546e2982b2a'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlan0
Jan 10 22:22:09 dnsmasq[15]: started, version 2.80 cachesize 150
Jan 10 22:22:09 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Jan 10 22:22:09 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Jan 10 22:22:09 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Jan 10 22:22:09 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr e4:5f:01:f3:85:8a and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
Jan 10 22:22:16 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jan 10 22:22:16 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlan0) d8:1f:12:d1:90:b2 
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 tags: wlan0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlan0) 10.42.42.20 d8:1f:12:d1:90:b2 
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  2
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlan0) d8:1f:12:d1:90:b2 
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 tags: wlan0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlan0) 10.42.42.20 d8:1f:12:d1:90:b2 
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  2
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 DHCPREQUEST(wlan0) 10.42.42.20 d8:1f:12:d1:90:b2 
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 tags: wlan0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 DHCPACK(wlan0) 10.42.42.20 d8:1f:12:d1:90:b2 wlan0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  5
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Jan 10 22:22:19 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Jan 10 22:26:52 dnsmasq-dhcp[15]: 1756340921 available DHCP range: 10.42.42.10 -- 10.42.42.40
[... cut ...]
Jan 10 22:26:52 dnsmasq-dhcp[15]: 1756340921 sent size:  4 option: 58 T1  6h
Jan 10 22:26:52 dnsmasq-dhcp[15]: 1756340921 sent size:  4 option: 59 T2  10h30m
Jan 10 22:26:52 dnsmasq-dhcp[15]: 1756340921 sent size:  4 option:  1 netmask  255.255.255.0
Jan 10 22:26:52 dnsmasq-dhcp[15]: 1756340921 sent size:  4 option: 28 broadcast  10.42.42.255
Jan 10 22:26:52 dnsmasq-dhcp[15]: 1756340921 sent size:  4 option:  3 router  10.42.42.1
Jan 10 22:26:52 dnsmasq-dhcp[15]: 1756340921 sent size:  4 option:  6 dns-server  10.42.42.1

Cossid, 2023-01-10:
Just wait, it usually takes about 60 seconds for the device to timeout at that point, and then everything continues on it's way.

sandermeis, 2023-01-10:
Thank you for taking the time to reply, could you explain what this means?:

>Jan 10 22:46:45 dnsmasq[15]: query[A] a3.tuyaeu.com from 10.42.42.20
Jan 10 22:46:45 dnsmasq[15]: config a3.tuyaeu.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
[LOG (Client)] Request: HTTPServerRequest(protocol='https', host='a3.tuyaeu.com', method='POST', uri='/d.json?a=tuya.device.uuid.pskkey.get&et=1&t=185&uuid=rxCQcJZyw9a0&v=1.0&sign=6fbcdca8d27a4d8b9c499f812df07a6d', version='HTTP/1.1', remote_ip='10.42.42.20')
[LOG (Client)] ==== Request body ===
{'t': 185}
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {'result': {'pskKey': ''}, 't': 1673390806, 'success': True}
[I 230110 22:46:46 web:2271] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=185&uuid=rxCQcJZyw9a0&v=1.0&sign=6fbcdca8d27a4d8b9c499f812df07a6d (10.42.42.20) 94.92ms
Jan 10 22:47:05 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:47:05 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:47:29 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:47:29 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:47:37 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:47:37 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:47:37 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:47:37 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:48:04 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:48:04 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:48:07 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:48:07 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:48:27 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:48:27 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:48:37 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:48:37 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:49:08 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:49:08 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:49:22 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:49:22 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:49:42 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:49:42 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:49:45 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:49:45 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:50:10 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:50:10 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
Jan 10 22:50:26 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.20
Jan 10 22:50:26 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1

Cossid, 2023-01-10:
I'm honestly not sure.  The only device I've seen respond like this is one I wasn't able to convert using CloudCutter, not because cloudcutter was doing anything incorrectly, but because the rom had such broken wifi, it wasn't really capable of completing anything.

This looks to me like your MQTT server might not be properly running.  It's trying to connect to mqtt (m2.) and failing, and repeating.  Can you post a full start-to-finish log?

Are you running elevated?

sandermeis, 2023-01-11:
Yes I am, I tried it again today with different results:

>pi@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  23.08MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 4c93b56bfe17
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 8a45702c449f
Step 3/10 : FROM base AS python-deps
 ---> 8a45702c449f
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> e2b544ebe38e
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 92a3d6fac84c
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> a4c98fd58a2b
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> af72225ac5b3
Step 8/10 : FROM python-deps AS cloudcutter
 ---> af72225ac5b3
Step 9/10 : COPY src /src
 ---> acbeca0dc18d
Step 10/10 : WORKDIR /src
 ---> Running in 5c74ab1143b3
Removing intermediate container 5c74ab1143b3
 ---> 0a34b4539745
Successfully built 0a34b4539745
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.305.bin
 > OpenBK7231N_UG_1.15.305.bin

[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.6 - BK7231T / oem_bk7231s_strip_ty
   1.1.6 - BK7231T / oem_bk7231s_switch1_zero_kjd
   1.1.7 - BK7231N / oem_bk7231n_ceiling_light_ty
   1.1.7 - BK7231N / oem_bk7231n_plug
   1.1.7 - BK7231T / oem_bk7231s_light_pir_ty
   1.1.71 - BK7231T / bk7231t_common_user_config_ty
 > 1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.8 - BK7231T / oem_bk7231s_rnd_switch
   1.1.9 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.2.1 - BK7231N / oem_bk7231n_control_switch
   1.2.16 - BK7231N / oem_bk7231n_light_ty
   1.2.7 - BK7231N / oem_bk7231n_light_ty
   1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty

[?] Select the brand of your device: Tuya Generic
   Atarm
[... 156 lines cut ...]
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {'t': 1673435465, 'success': True}
[I 230111 11:11:05 web:2271] 200 POST /d.json?a=tuya.device.dynamic.config.ack&devId=m32hCcyHYSwbOvhhohZk&et=1&t=1673435465&v=2.0&sign=25dd9d7ffc4205a283113dc538b992bf (10.42.42.20) 96.53ms
[LOG (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.timer.count&devId=m32hCcyHYSwbOvhhohZk&et=1&t=1673435465&v=4.0&sign=c565edc8f8fdcce2348676b76f18e922', version='HTTP/1.1', remote_ip='10.42.42.20')
[LOG (Client)] ==== Request body ===
{'devId': 'm32hCcyHYSwbOvhhohZk', 'lastFetchTime': '0', 't': 1673435465}
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {'result': {'devId': 'm32hCcyHYSwbOvhhohZk', 'count': 0, 'lastFetchTime': 0}, 't': 1673435465, 'success': True}
[I 230111 11:11:05 web:2271] 200 POST /d.json?a=tuya.device.timer.count&devId=m32hCcyHYSwbOvhhohZk&et=1&t=1673435465&v=4.0&sign=c565edc8f8fdcce2348676b76f18e922 (10.42.42.20) 101.30ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1673435470} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[LOG (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.upgrade.get&devId=m32hCcyHYSwbOvhhohZk&et=1&t=1673435469&v=4.4&sign=e1aae51e8971e24f9da0e5835a863dd5', version='HTTP/1.1', remote_ip='10.42.42.20')
[LOG (Client)] ==== Request body ===
{'type': 0, 't': 1673435469}
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {'result': {'url': 'http://10.42.42.1:80/files/OpenBK7231N_UG_1.15.305.bin', 'hmac': '172085F59EE7FCA2AEC60656E87162E00DCB9EC69600A0337EAAFF751C3D0A65', 'version': '9.0.0', 'size': '605616', 'type': 0}, 'success': True, 't': 1673435470}
[I 230111 11:11:10 web:2271] 200 POST /d.json?a=tuya.device.upgrade.get&devId=m32hCcyHYSwbOvhhohZk&et=1&t=1673435469&v=4.4&sign=e1aae51e8971e24f9da0e5835a863dd5 (10.42.42.20) 123.26ms
[LOG (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.upgrade.status.update&devId=m32hCcyHYSwbOvhhohZk&et=1&t=1673435470&v=4.1&sign=3b7239a3bfa95846ee25f28e8b190126', version='HTTP/1.1', remote_ip='10.42.42.20')
[LOG (Client)] ==== Request body ===
{'type': 0, 'upgradeStatus': 2, 't': 1673435470}
[LOG (Client)] ==== End request body ===
!!! Endpoint default response not found - tuya.device.upgrade.status.update
[LOG (Server)] Response:  {'success': True, 't': 1673435470}
[I 230111 11:11:10 web:2271] 200 POST /d.json?a=tuya.device.upgrade.status.update&devId=m32hCcyHYSwbOvhhohZk&et=1&t=1673435470&v=4.1&sign=3b7239a3bfa95846ee25f28e8b190126 (10.42.42.20) 94.82ms
pi@raspberrypi:~/tuya-cloudcutter $

Cossid, 2023-01-11:
About all we can recommend in this case is to keep trying.

Your device made a request, and the firmware was sent
> [I 230111 11:11:03 web:2271] 200 GET /files/OpenBK7231N_UG_1.15.305.bin (10.42.42.20) 16399.75ms
> [DEVICE OTA] Responding to device OTA HTTP request range: bytes=0-605615/605616

But then it requested it again, but there is no way for us to know why.  Usually if you keep trying it will eventually work.

sandermeis, 2023-01-11:
It did indeed work after several times, thanks for the replies and for sparing me the soldering!!

FYI in the host specific solutions .md, /tuya-cloudcutter.sh is still named /run_detach.sh

Cossid, 2023-01-11:
Noted, I'll get that updated.  Glad everything worked out.