tuya-cloudcutter/tuya-cloudcutter #257: Multiple devices in AP mode

lowey71, 2023-01-13:
Have quite a few hard wired downlights that would like to convert, but without having to electrically isolate each lamp, would it be possible to handle multiple devices in AP mode?

For a test, I have 2 downlights on one switch.  Placing them into AP mode and attempting to flash seems to result in cloudcutter combing both of the AP names found and fails.  Example:

using profile: arlec-ald095ha-rgbct-downlight

**Found access point name: "GRID-7598
GRID-65A8", trying to connect..
Error: No network with SSID 'GRID-7598
GRID-65A8' found.
Error: GRID-7598
GRID-65A8 - no such connection profile.**

Is it possible to say select an individual device? Or just grab the first seen and attempting to process that one only, then the other can be done later etc? 

Thanks

Cossid, 2023-01-13:
It's pretty difficult to handle multiple at the same time because of devices resetting not possibly not being in the same state.

Selecting all available APs like that was a bug.  I commited a patch that will just take the first one.  Please git update and try again.

lowey71, 2023-01-13:
Thanks.  Partially worked, in the sense that the head -n1 picks up the first detected.  Attempts to exploit but in the 2nd stage, will pick up the other light.

Workaround was placing another grep to just look for the 4 letter id I want to flash to ensure it will only attempt fully exploiting the one unit.  Then repeat for the 2nd (or in my case, remove as only working on 2 at once at this stage so only one left to exploit).

Cossid, 2023-01-13:
Ah, yeah, that makes sense, it will reset the AP and rescan, so the order is never maintained.  We can't just scan for the same AP name, as cutting can change the prefix.  I'll look into parsing out just the last 4 and see if we can use that instead for the second scan.  Should be possible.  I'll also look into a pass in parameter to filter the list as well.

Cossid, 2023-01-14:
I've decided against a pass in argument for now as the dash (-) could lead to ambiguity and confusion.  I've added a check that will record the last 5 characters of the AP SSID on a successful connection, which shouldn't change, and only reconnect to that on subsequent tries for the active session.

So now, it will try the first one it sees in the list, and stick to that one (even if it changes to the A- prefix) for the active session.

closed 2023-01-14 by Cossid