tuya-cloudcutter/tuya-cloudcutter #258: Firmware for Aubess mini switch 5A

yvesdm3000, 2023-01-14:
This is the firmware for an Aubess mini switch, described here:

https://www.elektroda.com/rtvforum/viewtopic.php?p=20385212#20385212

[readResult_BK7231N_QIO_2023-13-1--19-27-56.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10418707/readResult_BK7231N_QIO_2023-13-1--19-27-56.zip)

Can you extract the keys and add it to the database ? Thanks.

Yves

Cossid, 2023-01-14:
Unfortunately, that is a bad dump, all bytes contained within are null bytes.

kuba2k2, 2023-01-14:
@openshwprojects ~~can I suggest using bk7231tools now?~~

yvesdm3000, 2023-01-14:
Oops, that is sad. I have overwritten it with an OpenBeken firmware.

closed 2023-01-14 by yvesdm3000

kuba2k2, 2023-01-14:
So we lost a firmware dump for cloudcutter thanks to the tool :slightly_smiling_face:

yvesdm3000, 2023-01-14:
Looks that way... The tool from openbeken even did a checksum verification that said was OK.

openshwprojects, 2023-01-14:
We are sorry for inconvenience, but the tool will not proceed once a certain step goes wrong. The checksum check is also done correctly (which also means it will fail if you selected wrong platform because it's different by one between T and N). Was the whole operation correctly completed? Check file on your HDD to see if it's also 0x00 and please provide steps how to reproduce the issue. It has never happened before so it would be good to know how to reproduce.

Don't worry about the flash dumps, I provide a lot of them for you, even tough I don't even have cloudcutter installed.

kuba2k2, 2023-01-15:
I'd say that the program created the 2MB file and either didn't write anything for some reason, or wrote an array of 0x00 chunks, because they were cleared by something. 

I'm not worrying about the dumps, just letting you know that there's a bug in the tool.

yvesdm3000, 2023-01-15:
> I'd say that the program created the 2MB file and either didn't write anything for some reason, or wrote an array of 0x00 chunks, because they were cleared by something.
> 
> I'm not worrying about the dumps, just letting you know that there's a bug in the tool.

Yes thank you. Playing with this device was just a warm-up to work on others. I now know I need to take extra care for the dump file.

Yves