tuya-cloudcutter/tuya-cloudcutter #260: tuya-cloudcutter script error

Ricardo1320-del, 2023-01-15:
Hello,

I get the error message below:

`
pi@raspberrypi:~/tuya-cloudcutter $ sudo docker build --network=host -t cloudcutter .
Sending build context to Docker daemon   22.5MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 04059e6996c2
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> b5f0278115ab
Step 3/10 : FROM base AS python-deps
 ---> b5f0278115ab
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 8a4010b29318
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 341eabb42d65
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 3205d141dc41
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 17366d623317
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 17366d623317
Step 9/10 : COPY src /src
 ---> Using cache
 ---> f3cedc55199d
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> c0e819246942
Successfully built c0e819246942
Successfully tagged cloudcutter:latest
pi@raspberrypi:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Error: NetworkManager is not running.
[!] Unable to auto-detect wifi adapter.  Please use the '-w' argument to pass in a wifi adapter.
See '{./tuya-cloudcutter.sh} -h' for more information.
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
./tuya-cloudcutter.sh: line 86: run_in_docker: command not found
pi@raspberrypi:~/tuya-cloudcutter $
`
Docker is properly installed, i can use docker build. But when i start script, run_in_docker can't be found.

Cossid, 2023-01-15:
You have 2 errors present explaining what is failing.
You don't have Network Manager installed or running, and as a result, it couldn't control your wifi adapter.

Please see
https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md
for more details.

But beyond that, it's not finding pipenv inside your docker.  You don't need to run docker build separately, the script will run it on it's own.

closed 2023-01-15 by Cossid

donburch888, 2023-01-20:
I also got the "run_in_docker: command not found" error. 

Changing to the LITE version of Raspberry Pi OS (as stated in the instructions) fixed that issue for me.