tuya-cloudcutter/tuya-cloudcutter #263: Issue while building docker image

JorgMuskens, 2023-01-17:
Hi,
Hope someone can help me. While running the docker build i am running into an issue which i am unable te solve. 

```
pi@piusb:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -r
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  22.95MB
Step 1/10 : FROM python:3.9-slim-buster AS base
3.9-slim-buster: Pulling from library/python
75680cfdd833: Pull complete
97dd79ea073a: Pull complete
44e20250de7c: Pull complete
01d185b489bc: Pull complete
83bfa0c24126: Pull complete
Digest: sha256:ac4662237ccc729004bc85a463f4948fe47e0f44ddd0011130043d39574fa6f2
Status: Downloaded newer image for python:3.9-slim-buster
 ---> 04059e6996c2
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> [Warning] The requested image's platform (linux/arm/v7) does not match the detected host platform (linux/arm/v6) and no specific platform was requested
 ---> Running in 8a5f51c4a507
The command '/bin/sh -c apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto' returned a non-zero code: 139
Failed to build Docker image, stopping script
pi@piusb:~/tuya-cloudcutter $
```

It looks like network connectivity is an issue but i am unsure if this is the real case. When i comment out the apt-get update entry the script continues. So i think the update generates an error in attempting to update. I am running from a raspberry pi zero and used the build process detailed on this page

hope someone can point me to getting this fixed

thanks Jorg

Cossid, 2023-01-17:
Yes, connectivity is an issue, and this won't work on an original Pi Zero unless you have a USB wifi adapter (Pi Zero original has neither ethernet nor wifi).  Specifically wifi is needed for AP mode.

It looks like you might be using Raspbian Buster?  I think CloudCutter has only been tested on bullseye so far.

Using Zero with wifi only is also tricky and requires USB for SSH:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md#pi-zero-2w-with-ssh-over-usb

JorgMuskens, 2023-01-19:
I resolved the issue with updating the Dockerfile with the proper base. I was using buster as that stated in the Dockerfile. On your request i updatet to bullseye. Initial tries failed till i realized that i had to pull the bullseys python as base release. Now it is working. You could update the dockerfile to prevent these kind of issue by using for example an sh file which if statements which could read the os version.

closed 2023-01-19 by JorgMuskens