tuya-cloudcutter/tuya-cloudcutter #264: wb3l bk7231t CW ceiling lamp 18W

openshwprojects, 2023-01-19:
Hello, this device:
[image]
Very hard to flash by wires, but at least I just bought only 1 piece sooo other users will benefit from my flash?
[image]
Had to trick-desolder it:
[image]
[image]
And finally:
[image]

Here are the dumps:

[backups-cw-lamp-wb3l.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10458294/backups-cw-lamp-wb3l.zip)
I hope you'll find them useful, please tell if it's a new device or not.

Cossid, 2023-01-19:
This is a new device that shares a common existing light profile.

A new device has been added as Kobi Oprawa Smart LED Moon.

closed 2023-01-19 by Cossid

kuba2k2, 2023-01-19:
I think "oprawa" is a Polish word for "a light fixture" :smile: