tuya-cloudcutter/tuya-cloudcutter #266: Tuya devices are cut from cloud, but not returned to WLAN

donburch888, 2023-01-21:
I installed CloudCutter on a RasPi 4 using a new SDcard.  After a few minor hiccups (like where is "run_in_docker" ? - caused by using the desktop version OS) I got it working and successfully cut my first Arlec PC191HA (for which you already have a profile) using the firmware version and name option. 

Note that the PC191HA power plug takes a 5-10 second press of the power button to go into its pairing mode with the LED fast blinking.  Cloudcutter does not detect this mode, so have to press power for another 5-10 seconds to get it to start its slow blinking mode - which cloudcutter detects and continues.  At the "==> Turn the device off and on again once." prompt, tuya-cloudcutter.sh script does not wait long enough for this double reset - but after giving a "Error: Connection activation failed: (53) The Wi-Fi network could not be found" message it does seem to continue detecting and appears to work correctly.  

It took me several tries to get the timing and procedure right.  I also found that entering my user password several time gets tedious and prone to keyboard errors; but prefixing with sudo solved that.  End result was that cloudcutter gave new Device_id and Local_keys. I re-installed the device in Home Assistant localTuya integration.  Excellent result; first device cut from the cloud; now I can take time investigating LibreTuya firmware option.  

But, the next day ...
--
The next day I rebooted my RasPi and proceeded to liberate my other 4 PC191HA units. Since I planned to repeat the command I added the -p option to the shell command as well as -s <ssid> <password> (note that password is 11 alphanumeric-only characters)

> sudo ./tuya-cloudcutter.sh -s dagoba password -p arlec-pc191ha-smart-plug

Device_id and Local_key were being given, despite noticing many python trace dumps, including:
```
[E 230120 00:39:59 web:1798] Uncaught exception POST /d.json?a=tuya.device.dynamic.config.get&devId=bxKoJEsJMnx6VezL7C1x&et=1&t=219&v=1.0&sign=8c9080fb93f26164b9f8f18f050f115e (10.42.42.12)
    HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.dynamic.config.get&devId=bxKoJEsJMnx6VezL7C1x&et=1&t=219&v=1.0&sign=8c9080fb93f26164b9f8f18f050f115e', version='HTTP/1.1', remote_ip='10.42.42.12')
    Traceback (most recent call last):
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/src/.venv/lib/python3.9/site-packages/tornado/web.py", line 1711, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      File "/src/cloudcutter/protocol/handlers.py", line 96, in post
        request_body = self.__decrypt_request_body(key_choice)
      File "/src/cloudcutter/protocol/handlers.py", line 138, in __decrypt_request_body
        decrypted = self.cipher.decrypt(body, key_choice).decode('utf-8')
      File "/src/cloudcutter/crypto/tuyacipher.py", line 30, in decrypt
        return self.__decrypt_with_retry(data, key_choice)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 59, in __decrypt_with_retry
        return self.__decrypt_with_retry(data, key_choice, False)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 56, in __decrypt_with_retry
        raise e
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.
[E 230120 00:39:59 web:2271] 500 POST /d.json?a=tuya.device.dynamic.config.get&devId=bxKoJEsJMnx6VezL7C1x&et=1&t=219&v=1.0&sign=8c9080fb93f26164b9f8f18f050f115e (10.42.42.12) 104.97ms
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying is false - bailing out
[E 230120 00:39:59 web:1798] Uncaught exception POST /d.json?a=tuya.device.info.sync&devId=bxKoJEsJMnx6VezL7C1x&et=1&t=219&v=1.0&sign=8f13296bb746d84941d04e54205317fb (10.42.42.12)
    HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.info.sync&devId=bxKoJEsJMnx6VezL7C1x&et=1&t=219&v=1.0&sign=8f13296bb746d84941d04e54205317fb', version='HTTP/1.1', remote_ip='10.42.42.12')
    Traceback (most recent call last):
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/src/.venv/lib/python3.9/site-packages/tornado/web.py", line 1711, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      File "/src/cloudcutter/protocol/handlers.py", line 96, in post
        request_body = self.__decrypt_request_body(key_choice)
      File "/src/cloudcutter/protocol/handlers.py", line 138, in __decrypt_request_body
        decrypted = self.cipher.decrypt(body, key_choice).decode('utf-8')
      File "/src/cloudcutter/crypto/tuyacipher.py", line 30, in decrypt
        return self.__decrypt_with_retry(data, key_choice)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 59, in __decrypt_with_retry
        return self.__decrypt_with_retry(data, key_choice, False)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 56, in __decrypt_with_retry
        raise e
[... 60 lines cut ...]

```
pi@raspberrypi:~/tuya-cloudcutter $ tail -f /var/log/syslog
Jan 21 10:42:17 raspberrypi NetworkManager[19996]: <info>  [1674258137.1741] dhcp4 (wlan0): state changed unknown -> bound, address=192.168.1.97
Jan 21 10:42:17 raspberrypi avahi-daemon[405]: Registering new address record for 192.168.1.97 on wlan0.IPv4.
Jan 21 10:42:17 raspberrypi NetworkManager[19996]: <info>  [1674258137.1853] device (wlan0): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
Jan 21 10:42:17 raspberrypi NetworkManager[19996]: <info>  [1674258137.1974] device (wlan0): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
Jan 21 10:42:17 raspberrypi NetworkManager[19996]: <info>  [1674258137.1991] device (wlan0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
Jan 21 10:42:17 raspberrypi NetworkManager[19996]: <info>  [1674258137.2124] dns-mgr: Writing DNS information to /sbin/resolvconf
Jan 21 10:42:17 raspberrypi NetworkManager[19996]: <info>  [1674258137.2352] device (wlan0): Activation: successful, device activated.
Jan 21 10:42:17 raspberrypi NetworkManager[19996]: <info>  [1674258137.2382] manager: startup complete
Jan 21 10:42:27 raspberrypi systemd[1]: NetworkManager-dispatcher.service: Succeeded.
Jan 21 10:42:37 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
```

Now, while I have been updating this, my most recent device has come online ?  What did I do ? Is it like Windows where you have to go make a cup of tea for a new network to start working ?

OK, run cloudcutter for my last device ... all appears OK, got device_id and local_key ... but not available on my WLAN :-( Turned device off and on several times over 10 minutes, and still nothing. 

I guess i should leave my last device, in case you can suggest something else. 
What other information can I provide ?

In the mean-time I have 2 paths which will take time to get my head around:
- I have 3 other Tuya devices which don't appear to already have CloudCutter profiles
- investigate LibreTuya firmware

Cossid, 2023-01-21:
There are several things missing or incomplete here.  Please provide complete logs.  Only showing the ending error doesn't show us the process of getting to the error, and we can't give helpful advice if the error is just cascading down to the end.

Yes, `sudo` is the preferred way to run because of the system level control needed to handle the wifi adapter.  I'll update documentation for that.  The input parameters should be working correctly, but the `-p` parameter does have a quirk of you either need to select it through the menu first once, or you need to download and place the files manually because of the way the API works.  You shouldn't need quotes for anything unless there are spaces present.

If you get a padding error, the exploit is not working and the values shown at the end are what cloudcutter _tried_ to set, but failed.  I'll see if I can add a failure/stop sooner in the process.  This is where not seeing what came before it blocks us from seeing what part went wrong.  Padding error is almost always either the wrong profile, or using keys belonging to another device.

Yes, you should only be doing one device at a time, as we cannot resume a session for a device, it must go through from start to finish because we're not using firmware we control, but the stock firmware which has set actions built in beyond our control.

donburch888, 2023-01-21:
Thanks Cossid for your reply - and of course to you all for CloudCutter.

Sorry that my terminal scrollback buffer wasn't long enough to get the start of that processing; but as I said, I did work out that the errors were because I had not noticed that my previous cloudcutcut attempts were unsuccessful, and had left the devices powered on. 

The issue I am having is that tuya-cloudcutter runs without giving any error, gives Device_id and Local_key ... but the device is not returned to my WLAN.  

I perform it one more time to get the log, and true to Murphy's Law, it works this time.  Maybe the RasPi needs to be rebooted between each run of cloudcutter ?  And since the device _did_ connect to my WLAN this time, I guess there's no point giving you the log, or leaving this issue open. 

That was the last of my Arlec PC191HA devices, there don't appear to be profiles for my other Tuya devices, and the procedure for LightLeak looks more involved than I feel capable of at this time, so I'll just let them lie. 
The Tuya devices are cheap and readily available, but it seems my wi-fi is an ongoing issue, so in future I will be buying Zigbee devices with ESPHome.

Cossid, 2023-01-21:
> The issue I am having is that tuya-cloudcutter runs without giving any error, gives Device_id and Local_key ... but the device is not returned to my WLAN.

`padding is incorrect` is an error though (and a fatal one to the process).  I pushed changes that should make that more clear.

> Maybe the RasPi needs to be rebooted between each run of cloudcutter ?

Shouldn't need to.  Occasional reboots maybe, but certainly not between each run.  I've done dozens of bulbs back-to-back without issue.  But doing 1 device at a time is certainly the best way to go.  Because of the nature of CloudCutter being a man-in-the-middle that needs to go through a linear progression, having multiple devices could throw off the linear portion and cause unexpected results.

It sounds like you're at the end of your process, and I'm not sure there are additional changes that can be made.  As a result, I'll close this issue as complete.  Feel free to re-open it with complete logs if you continue to have issues with other devices (or create a new issue).

closed 2023-01-21 by Cossid