tuya-cloudcutter/tuya-cloudcutter #267: cloud cutter did not complete - bulb is partially freed

tjbcatch22, 2023-01-23:
Hoping someone can comment on this and provide their thoughts on how to complete the cloud cut.
I ran cloudcutter on a bulb made by Schwaiger, model SWLc19.  There is not much info on the bulb on the internet.  I used a generic profile (tuya-generic-e27-rgbct-wb2l-m1-bulb).
The bulb was set up and working fine in Smart Life app prior to running cloud cutter.

The process went fine up to when the bulb was to reboot into AP mode.  I restarted the process and now seems like I can not longer "see" the bulb in the Smart Life App and cannot complete the process in Cloudcutter.  I can get the bulb into AP mode (slow blink) or setup (fast blink).  Cloudcutter seems to connect and disconnect from the bulb when rerunning the process and finally fails with (below).  The access point of the bulb is A-669B:
Found access point name: "A-669B", trying to connect..
Device 'wlp1s0' successfully activated with '64dedeb3-81e7-484e-9fd8-4d6dbb0541f4'.
Error, could not connect to SSID.
Failed to connect, please run this script again

Running nmap with the bulb in AP mode shows:
PORT   STATE         SERVICE
53/udp open|filtered domain
67/udp open|filtered dhcps

The appears to notice it is being connected to in AP mode by either the Smart Life app or cloudcutter because is stops blinking (slow) when the programs try to connect and then remains on.

And cloudcutter will only allow running using "From device-profiles" and only allow the "tuya-generic-e27-rgbct-wb2l-m1-bulb" profile.  The other choices result in a bunch of errors with the program crashing.

Any thoughts or suggestions on how to finish cloudcutter with the bulb?

Cossid, 2023-01-23:
Try running with the `-r` parameter?

SmartLife devices only allow 1 device to be connected at a time, I'm not sure if nmap is causing you additional problems.

tjbcatch22, 2023-01-24:
Thank you for your response.  Unfortunately -r wasn't able to do it.  I did see the additional message "Wiping NetworkManager configs", but ultimately the process would cycle through 
"Found access point name: "A-669B", trying to connect..
Device 'wlp1s0' successfully activated with '1df95a15-6592-4c6b-932b-c78a78bdb829'."  a few times  and finally end with:
"Error, could not connect to SSID.
Failed to connect, please run this script again"

I have all the devices right next to each other.

BTW, I provided the nmap in case anyone thought of another way into the bulb while the bulb is in AP mode.  Those are the ports open when the bulb is in AP mode.  I 'm not running nmap when I'm running cloudcutter.

Cossid, 2023-01-24:
Can you please manually connect to the AP and run the command

`nmcli -f IP4.GATEWAY con show "A-669B" | awk -F ' ' '{print $2}'` and provide what it responds with?
We check for a successful connection to a Tuya device by expecting the device gateway IP to be 192.168.175.1

One of two things is happening;
1) it's not connecting at all, and that would be outside the control of cloudcutter.
2) You're getting a different gateway IP or localization has a different format since that is a string match.

tjbcatch22, 2023-01-24:
With the bulb in AP mode (slow blink) I can connect to it as if it were a regular AP
When running
`nmcli -f IP4.GATEWAY con show "A-669B" | awk -F ' ' '{print $2}'`
`192.168.175.1`

And...
`iwconfig
wlp1s0    IEEE 802.11  ESSID:"**A-669B**"  
          Mode:Managed  Frequency:2.437 GHz  Access Point: D4:A6:51:E9:66:9A   
          Bit Rate=54 Mb/s   Tx-Power=22 dBm`

`wlp1s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet **192.168.175.100**  netmask 255.255.255.0  broadcast 192.168.175.255`

I bolded my ip addess and the Bulb as an AP after connecting.  The gateway is 192.168.175.1.  I can ping it.

However, I noticed in my network manager that there are numerous A-669B access points with a number appended.  i.e. "A-669B 1" "A-669B 2", etc.
I connect to the first one (no number appended), ran cloudcutter.  It failed with
`Found access point name: "A-669B", trying to connect..
Device 'wlp1s0' successfully activated with '5505cfe6-89ac-4193-a13e-98c0b7fc6e90'.
Error, could not connect to SSID.
Failed to connect, please run this script again`

And when i looked at my AP connection after the fail, I was attached to "A-669B 31"  When that occurred, I' can still ping 192.168.175.1, but the nmcli command does not return anything.

Its like there is an initial connection, then there is a new connection to the AP with the number appended and the program fails.

Is it possible for hard code the AP name into the script?  I've been looking but haven't seen where to do so.

Cossid, 2023-01-24:
That is the issue `-r` is meant to resolve.  Your system is creating multiple profiles in `/etc/NetworkManager/system-connections/` and -r attempts to clear them out.  Where you not running elevated or as a user that doesn't have permissions to change that directory?

I'd recommend removing all files containing `A-669B` from `/etc/NetworkManager/system-connections/` manually and see if that works.

The only change we can make to cloudcutter would be to try to parse the guid of the connected network and use that instead, but unfortunately, that leads to more internationalization issues we are trying to avoid.  It looks like there may be a way we can force the profile name to be consistent, I'll look into that.

> Is it possible for hard code the AP name into the script? I've been looking but haven't seen where to do so.

That won't help because your system is appending a number, that that's the crux of the issue.  The script is recording the AP's broadcast SSID, but your system names it something extra, and you can't hard-code it because it appears it is creating a new one each time (you'd have to predict, and cloudcutter makes multiple connections because of the different states required, so it would be maybe right once but then also wrong once).

Cossid, 2023-01-24:
I have pushed a change that will try to create a profile name that matches the broadcast AP SSID, please do a `git update` and try again?

tjbcatch22, 2023-01-24:
Thank you.  I'm on it.  Will keep you posted

tjbcatch22, 2023-01-24:
Success !  You are Awesome !  I clear out the AP names as you suggested above.  During the process 4 more came back, however, now with many characters following (i.e cd65001a-f573-421c-a17f-14f7ef2c020d)
I ran the updated version and the processed completed with the bulb !
`Device should be successfully onboarded on WiFi AP!`

and I can see it in TinyTuya with the ip, device id, local key (which is blank), version and type.
`Scan Complete!  Found 1 devices.`

Thank you !

closed 2023-01-24 by Cossid