tuya-cloudcutter/tuya-cloudcutter #269: Error(new linux user)

magoone7, 2023-01-27:
Uncertain of what i am truly doing, but i have docker installed and am running ubuntu.

>Checking UDP port 53... Occupied by systemd-resolve with PID 10061.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  22.48MB
Step 1/10 : FROM python:3.9-slim-buster AS base
Get "https://registry-1.docker.io/v2/": dial tcp: lookup registry-1.docker.io: Temporary failure in name resolution
Failed to build Docker image, stopping script

tjbcatch22, 2023-01-27:
This happened to me.  Looking through the closed issues, I found a reference to edit `safety_checks.sh` and comment out the reference to port 53 to look like this: `#check_port udp 53 "resolve DNS queries"`
It worked for me.

Cossid, 2023-01-27:
The safety check is there for a reason, as CloudCutter does require listening on port 53 to intercept requests from tuya devices to exploit them.

When this happens, this means your machine has a DNSMasq server running and is likely proxying all DNS requests through itself.  I haven't been able to reproduce a case for this, but it is possible it's only binding on certain adapters in which case it may not be an issue.  Alternately, you can try adding dns nameservers to /etc/resolv.conf other than local addresses (1.1.1.1, 9.9.9.9, 8.8.8.8, etc)

You can try disabling the safety check for port 53, but you may or may not run into other errors further down the line.

magoone7, 2023-01-28:
> This happened to me. Looking through the closed issues, I found a reference to edit `safety_checks.sh` and comment out the reference to port 53 to look like this: `#check_port udp 53 "resolve DNS queries"` It worked for me. I made sure to have docker installed correctly prior to that.

Could you explain docker install correctly? Incase i did it wrong.

magoone7, 2023-01-28:
> The safety check is there for a reason, as CloudCutter does require listening on port 53 to intercept requests from tuya devices to exploit them.
>
> When this happens, this means your machine has a [...]

Ill try out some of the dns setting in the config and see if it changes anything.

closed 2023-01-28 by magoone7