tuya-cloudcutter/tuya-cloudcutter #270: Connection to AP not possible

wu7zh, 2023-01-27:
Hello,
so I ran the script to cut a GU10 RGBCCT Bulb (LVWIT brand) from the cloud. Everything worked fine and I got the keys in my configured-devices folder, it also connects to my Wi-Fi.

Afterwards I tried to flash a custom firmware (OpenBeken), and the script worked up until the second time you reset the bulb. At this point it found the AP but could't connect, after a while it timed out.
Upon trying again the script now can't even connect on the first time (Still finds the AP though).

Did I brick my device or can I save this situation somehow?

For reference I am using Kali Linux on VMware with a USB WiFi adapter.

tjbcatch22, 2023-01-27:
This happened to me.  I went into `/etc/NetworkManager/system-connections` and found multiple instances of the AP (i.e. a number was appended to each new instance of the AP and seemed to cause an issue with reconnecting to the bulb).  
I removed those multiple instances, put the bulb back into AP mode, may have restarted the script if needed, might have also toggled back to the shell to watch if additional AP instances were being created - and deleted them - except for the original.
I was able to connect to the bulb and complete the script.

wu7zh, 2023-01-27:
Thank you so much, that solved it!

Now I run into another problem though, the script runs through without error, but the bulb is still running Tuya firmware and not OpenBeken, any idea what could be causing this?

Cossid, 2023-01-27:
Some devices just have unreliable network and it takes a few tries before it completes.  Just start the process over and eventually, it will work.  As always, try to keep the cloudcutter and target devices as close to each other as possible.

wu7zh, 2023-01-27:
Okay I will try it some more times and hold my laptop right next to the bulb, will report back later!
Thanks for the help and fast replies already

wu7zh, 2023-01-27:
Tried it about 15 more times now, and every time it's still on Tuya firmware.
Here is the log, help would be appreciated!

Selected Device Slug: feit-bpa8000rgbw2p-rgbct-bulb
Selected Profile: oem-bk7231s-light-ty-2.9.16-sdk-1.0.8-40.00
Selected Firmware: /work/custom-firmware/OpenBK7231T_UG_1.15.391.bin
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP

Found access point name: "A-88B2", trying to connect..
Device 'wlan0' successfully activated with 'fef69fa7-ceba-4196-b4e3-ff95b4c7ec81'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/TdhVXAxQZZ2U.deviceconfig
Saved device config in /work/configured-devices/TdhVXAxQZZ2U.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
.
Found access point name: "A-88B2", trying to connect..
Device 'wlan0' successfully activated with '45bfb87a-4fdf-41f3-9430-4eeabf699c1c'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlan0
Jan 27 17:12:43 dnsmasq[15]: started, version 2.80 cachesize 150
Jan 27 17:12:43 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Jan 27 17:12:43 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Jan 27 17:12:43 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Jan 27 17:12:43 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr 3a:ef:89:40:57:47 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
Jan 27 17:12:45 dnsmasq-dhcp[15]: 1876173215 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jan 27 17:12:45 dnsmasq-dhcp[15]: 1876173215 client provides name: wlan0
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 DHCPDISCOVER(wlan0) fc:67:1f:59:88:b2 
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 tags: wlan0
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 DHCPOFFER(wlan0) 10.42.42.37 fc:67:1f:59:88:b2 
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 next server: 10.42.42.1
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  1 option: 53 message-type  2
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 54 server-identifier  10.42.42.1
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 51 lease-time  12h
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 58 T1  6h
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 59 T2  10h30m
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option:  1 netmask  255.255.255.0
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 28 broadcast  10.42.42.255
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option:  3 router  10.42.42.1
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option:  6 dns-server  10.42.42.1
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 available DHCP range: 10.42.42.10 -- 10.42.42.40
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 client provides name: wlan0
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 DHCPDISCOVER(wlan0) fc:67:1f:59:88:b2 
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 tags: wlan0
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 DHCPOFFER(wlan0) 10.42.42.37 fc:67:1f:59:88:b2 
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 next server: 10.42.42.1
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  1 option: 53 message-type  2
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 54 server-identifier  10.42.42.1
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 51 lease-time  12h
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 58 T1  6h
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 59 T2  10h30m
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option:  1 netmask  255.255.255.0
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option: 28 broadcast  10.42.42.255
Jan 27 17:12:48 dnsmasq-dhcp[15]: 1876173215 sent size:  4 option:  3 router  10.42.42.1
[... 68 lines cut ...]
[LOG (Client)] ==== End request body ===
[... cut ...]
[I 230127 17:13:06 web:2271] 200 POST /d.json?a=tuya.device.timer.count&devId=23cm8phRCMi7kK31bAuy&et=1&t=1674839586&v=4.0&sign=35cd396e6b463096697d42a42b1cb646 (10.42.42.37) 103.64ms
[I 230127 17:13:07 web:2271] 200 GET /files/OpenBK7231T_UG_1.15.391.bin (10.42.42.37) 10371.14ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=0-480431/480432
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1674839588} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.

Cossid, 2023-01-27:
That log shows a completely normal and successful firmware send.  Unfortunately, if the device rejects the firmware, we can't see that happen or the reason it might give.  I'd recommend redownloading the firmware to make sure it is valid and not corrupt.

CloudCutter spend 10 seconds sending the firmware file and completed (200) and the device confirmed it received all bytes (bytes=0-480431/480432), at which point, Tuya's OS reboots and verifies that what it downloaded is valid and then installs it.  If it fails for any reason, we cannot see the message as to why.

wu7zh, 2023-02-03:
Closing as I solved the problem.

After copying the file [tuya.device.dynamic.config.get.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/device-profiles/schema/tuya.device.dynamic.config.get.json) into the profile I am using it worked.
After that it worked perfectly on 6/6 bulbs without any disconnects or problems. Before this the problem persisted for any of them, so I am fairly certain that this file is what was missing.

Cossid, 2023-02-03:
> After copying the file [tuya.device.dynamic.config.get.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/device-profiles/schema/tuya.device.dynamic.config.get.json) into the [...]

I'm pretty sure that was just coincidental and simply running again would have provided the same result.  That file you copied is the one that is currently be used, and it would never be loaded from the directory you copied it to.

wu7zh, 2023-02-03:
Oh okay, then it was a pretty big coincidence, as i tried it like 50 times before, with restarting, reinstalling the scripts etc etc