tuya-cloudcutter/tuya-cloudcutter #271: Expected functioning of device profiles on a possibly generic bulb

tjbcatch22, 2023-01-27:
I have successfully flashed a generic device profile, to my generic bulb, however, the bulb does not have its full functionality.  I tried a few other profiles and get the same reduced functionality.  Would this be expected for any device profile that I would try?

Specifics:
I have a "Schwaiger A19 60W Equivalent Color Selectable Dimmable LED Light Bulb", (see here for more info: `https://www.menards.com/main/p-1642874274122411.htm`).  Successfully flashed using `tuya-generic-e27-rgbct-wb2l-m1-bulb`.

However, When using Tinytuya bulb.py, the bulb will only turn on/off and apply its color modes ( White - full bright on;  Colour - turns red and is the only color I get; Scene - bulb dims a little and softens color; Music - turns red again).  All other functions do not work (Dimming, colors).  The bulb has main module 1.2.16

I tried a few other profiles with that module (i.e. Feit, Nous) and get the same reduced function.

The bulb, of course, had full function: colors, dimming, scenes, etc while on the Smart Life app.

Do I keep flashing different profiles to hopefully find one with full function or should I not expect full function with this particular bulb?

Cossid, 2023-01-27:
Yes, that's not uncommon, which is why we name profiles as specific devices with verified schemas, and only recommend using the base profile/by firmware version for 3rd party flashing.

Devices can be configured with an LED driver (2 or 3 pretty common ones) or PWM, and they would have different schemas, even if they use the same firmware.  The configuration of the LED driver is on the bulb itself, and not something CloudCutter can change, we can only apply a schema, that needs to match to how the bulb is configured.

`tuya-generic-e27-rgbct-wb2l-m1-bulb` is a PWM bulb, so if it didn't work, your device likely has an LED driver.  Some of the other ones you appear to have tried have LED drivers, but probably had different configurations that yours.

Thanks for the heads-up on that bulb, I'll actually pick up a couple today or this weekend while they're on sale and see what I can find.

tjbcatch22, 2023-01-27:
Thank you Cossid.  Those bulbs are a nice price on sale.

I tried this profile: `kanlux-33642-s-a60-rtbcct-bulb` and noticed, then running Tinytuna, the dimming is working a little bit.  Bulb dims low ~ 10%, goes to ~ 40-50%, and also raised to full.  Just sharing this info in case its useful.

I also noticed when the Tinytuya bulb.py runs the `Current Status of Bulb` test, initially it did not find the bulb, but still was able to turn the bulb on, off, and the scenes.  
When I ran the script again, the `Current Status of Bulb` found the bulb and provided a result of status, was still able to turn the bulb on,off, and the scenes - but was also able to dim the bulb a little bit.

Cossid, 2023-01-28:
Picked up a couple, dumped with LightLeak and added a new profile as Schwaiger SWLC19 RGBCT Bulb, give it a try.

tjbcatch22, 2023-01-28:
Awesome! I gave it a try using Tinytuya bulb.py.  The dimming worked better from 40 to 100%.  None of the colors worked however, until the Scene and Music mode, turning the bulb to red.  This is the same functionality as the other profiles  I'm going to try an unopened/unused bulb tomorrow, in case its something with this particular bub, and adjust the bulb.py script to focus on the colors to see what I can get working.

tjbcatch22, 2023-01-28:
I might add...  I did not create a Tuya Account as in the instructions at `jasonacox/tinytuya`.  (I have an aversion to clouds vs running my devices locally if I can)
So, I'm wondering if my reduced function with the bulbs is due to that.  
Were you able to get the Schwaiger totally working?

Cossid, 2023-01-28:
I don't use TinyTuya or LocalTuya.  I do recall from when I used it in the past some bulbs didn't work with set_colour and instead needed to use set_hsv instead.  I did put ESPHome on one, but ran into other issues not related to the bulb, but it does appear this bulb has a BP5758D LED driver and I was able to get R/G/B and WW working correctly using a previous build I had from a different bulb, but found this bulb doesn't use WW/CW channels (sometimes referred to as RGBWW), but instead Temperature/Brightness (sometimes referred to as RGBCT), so this is a true RGBCT bulb.

At this point, there is nothing cloudcutter can do any differently.  I'd recommend possibly opening up an issue in TinyTuya instead.

closed 2023-01-28 by Cossid

Cossid, 2023-01-28:
Actually, it appears this bulb doesn't actually have a cold white channel or cold white LED chips, which means it can only do that via emulation.

Either way, the schema on the `Schwaiger SWLC19 RGBCT Bulb` profile in cloudcutter is the correct one directly from Tuya for this bulb.

tjbcatch22, 2023-01-28:
What do you use to operate the bulb, if not using tinytuya?
Thank you for creating the schema!

Cossid, 2023-01-28:
I flash 3rd party firmware (LibreTuya/ESPHome, though OpenBeken is also available).

tjbcatch22, 2023-01-28:
Thank you, I'll check those out.  By any chance, is there an already-developed device profile from OpenBeken you know and can point me to, that may work or partially work (so I don't have to start from zero and reinvent wheels to work on the bulb)?

Cossid, 2023-01-28:
See the device_configuration section of this profile: https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/schwaiger-swlc19-rgbct-bulb.json

Here is my ESPHome configuration you should also be able to adapt:
```yaml
bp5758d:
  data_pin: GPIO8
  clock_pin: GPIO7

# Individual outputs
output:
  - platform: bp5758d
    id: output_red
    channel: 3 # Valid channels are 1-5
    current: 28 # In mA, valid values are 0-90, default is 10.
  - platform: bp5758d
    id: output_green
    channel: 2 # Valid channels are 1-5
    current: 28 # In mA, valid values are 0-90, default is 10.
  - platform: bp5758d
    id: output_blue
    channel: 1 # Valid channels are 1-5
    current: 28 # In mA, valid values are 0-90, default is 10.
  - platform: bp5758d
    id: output_warmwhite
    channel: 5 # Valid channels are 1-5
    current: 55 # In mA, valid values are 0-90, default is 10.

light:
  - platform: rgbw
    restore_mode: RESTORE_DEFAULT_OFF
    name: "Light"
    red: output_red
    green: output_green
    blue: output_blue
    white: output_warmwhite
    color_interlock: true
```

I didn't set the upper/lower bounds for brightness, but you can see those in the device_configuration section, I'm not sure how OpenBeken handles stuff like that.

tjbcatch22, 2023-01-28:
Great!

tjbcatch22, 2023-01-30:
@Cossid .  If you were able to dump the bin for the Schwaiger bulb, are you able to post it here? If possible, I'd like to see if I can reinstall it and restore the original firmware, just because.

kuba2k2, 2023-01-30:
If you have a wired UART connection (and I assume you do, since you want to flash stock firmware), you can use [ltchiptool](https://docs.libretuya.ml/docs/flashing/tools/ltchiptool/) to do this - it will automatically use correct flashing offsets, so that only the APP firmware is flashed, and so that your MAC is unchanged.

tjbcatch22, 2023-01-30:
Thank you both !

Cossid, 2023-01-30:
If he's trying to undo cloudcutter, he'd need to flash storage as well.

kuba2k2, 2023-01-30:
Right. That still can be done without changing the MAC address. Storage offset is something like 0x1EB000 iirc.