tuya-cloudcutter/tuya-cloudcutter #272: Device seems updated, but not returned to WLAN

donburch888, 2023-01-29:
I have rebuilt the RasPi 4 I used previously to run CloudCutter, and reinstalled it. 

In your [Host specific instructions](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md) you should change step 4 from "Reboot the pi sudo reboot then reaccess." to 
> 4. run "sudo raspi-config", select "6 Advanced Options" > " AA Network Config" > then select "2 NetworkManager".  When you Finish from raspi-config do allow it to do the reboot to activate the network-manager  

I ran tuya-cloudcutter to check that it is all installed and working, and ran into the problem I reported previously (#266) - namely that it appears to have worked without error ... but the device just doesn't come back to my home WLAN, even after a couple of power cycles. 

Below is the log of my second attempt (with the extra options to avoid having to delete and re-create the device in Home Assistant's LocalTuya integration). I even copy and pasted the wireless SSID and PSK from the WAP to the RasPi CLI to avoid error. The RasPi 4, PC191HA power plug, and my home WAP are all sitting on my study desk, so signal strength is not an issue. 
```
pi@raspi4:~/tuya-cloudcutter $ sudo ./tuya-cloudcutter.sh -r -p arlec-pc191ha-smart-plug -d 4eN4UVnTcULNLkBu4Trb -l BxHcfdQC21WcY93T -s dagoba ssid_password
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  22.49MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 90e1cbf9cebb
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 6da306585bcf
Step 3/10 : FROM base AS python-deps
 ---> 6da306585bcf
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> b4c74812a0be
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 42b6f3ba8f2a
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> dbd9f2c1a43d
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 1e0c3f0097da
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 1e0c3f0097da
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 0c4df533696d
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 680e48219e25
Successfully built 680e48219e25
Successfully tagged cloudcutter:latest
Successfully built docker image
Loading options, please wait...
Selected Device Slug: arlec-pc191ha-smart-plug
Selected Profile: oem-bk7231s-dltj-1.3.5-sdk-1.0.2-40.00
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
......
Found access point name: "A-F21A", trying to connect..
Device 'wlan0' successfully activated with '01b18757-3293-4fc4-ae6e-43d0ec6e3fe7'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Using 4eN4UVnTcULNLkBu4Trb and BxHcfdQC21WcY93T
Exploit run, saved device config to!
output=/work/configured-devices/PCGv9NaXlccK.deviceconfig
Saved device config in /work/configured-devices/PCGv9NaXlccK.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
.
Found access point name: "A-F21A", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
[... 99 lines cut ...]

Looking at the log of my main ASUS WAP (in a different room) I find lots of
```
Jan 28 20:09:32 wlceventd: wlceventd_proc_event(530): eth2: Auth 38:1F:8D:F1:FA:22, status: Successful (0), rssi:0
Jan 28 20:09:32 wlceventd: wlceventd_proc_event(494): eth2: Deauth_ind 38:1F:8D:F1:FA:22, status: 0, reason: Unspecified reason (1), rssi:0
Jan 28 20:09:32 wlceventd: wlceventd_proc_event(530): eth2: Auth 38:1F:8D:F1:FA:22, status: Successful (0), rssi:0
Jan 28 20:09:34 wlceventd: wlceventd_proc_event(494): eth2: Deauth_ind 38:1F:8D:F1:FA:22, status: 0, reason: Unspecified reason (1), rssi:0
Jan 28 20:09:34 wlceventd: wlceventd_proc_event(530): eth2: Auth 38:1F:8D:F1:FA:22, status: Successful (0), rssi:0
Jan 28 21:09:54 wlceventd: wlceventd_proc_event(494): eth2: Deauth_ind 38:1F:8D:F1:FA:22, status: 0, reason: Disassociated due to inactivity (4), rssi:0
Jan 28 21:09:54 wlceventd: wlceventd_proc_event(511): eth2: Disassoc 38:1F:8D:F1:FA:22, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
```
Curiously I do have several devices with MAC addresses starting 38:1F:8D:F1 ... my PC191HA power plugs !  But the one I am trying to cut has MAC address ending F2:1A and I have none ending FA:22 !  Is this possibly a side effect ? Or maybe a different MAC for 5GHz connection (but my study WAP only does 2.4GHz). 

Issue #267 sounds similar, to this and my previous issue; and so I tried following instructions you gave there to connect to the device in slow blink mode giving

> nmcli -f IP4.GATEWAY con show "A-669B" | awk -F ' ' '{print $2}'
> 192.168.175.1

You mentioned updating cloudcutter, so I downloaded and installed it again, and ran again.  

Of course it worked this time !

For other future users who have this issue ... would it be possible to restart the last step (return device to users ssid) without having to keep repeating the whole cloud cutting procedure ?

Sorry to be complaining ... but I am really enthusiastic about using CloudCutter to free devices OTA.  Please keep up the good work !!!

Cossid, 2023-01-29:
> In your [Host specific instructions](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md) you should change step 4 from "Reboot the pi sudo reboot then [...]

I'm pretty certain those result in the same thing, our instructions are just the command line version.

> Below is the log of my second attempt (with the extra options to avoid having to delete and re-create the device in Home Assistant's LocalTuya integration). I even copy and pasted the wireless SSID [...]

Since the password is edited out, are there any non-alpha-numeric characters?  That could be a part of the problem.  It's possible you need to quote your password or just don't supply it at all and enter it when prompted instead.  There are some limitations for input parameters.

> Looking at the log of my main ASUS WAP (in a different room) I find lots of
>
> ```
> Jan 28 20:09:32 wlceventd: wlceventd_proc_event(530): eth2: Auth 38:1F:8D:F1:FA:22, status: Successful (0), rssi:0 [...]

No Tuya devices support 5GHz, so they don't have a separate MAC for that.  Generally speaking, the MAC address shouldn't shift by more than 1 character if put into a different mode (like AP mode).  Are you sure you don't have another device actually connected to Smart Life?

What version of WPA are you using?  If you're using WPA3 or mixed WPA2/WPA3, these devices may not support that.  We don't have control over what the Tuya OS running on the devices support.  The log you provided is a good and complete cut, but once the device has the information, CloudCutter can't do anything more.  If the device doesn't connect to your network, that's specific to your network.  There is absolutely nothing CloudCutter can be doing differently, it's just setting and SSID and Password, resetting security keys Tuya uses to something it knows (and records to configured-devices) and changing the remote Tuya address to something that won't communicate with Tuya.

> Issue #267 sounds similar, to this and my previous issue; and so I tried following instructions you gave there to connect to the device in slow blink mode giving
>
>
> You mentioned updating [...]

I'm not sure what to tell you.  The log you posted prior was correct, and nothing would have changed that by updating CloudCutter.  The only changes to CloudCutter have been error handling, not functional changes to exploiting.  Did you have multiple devices in pairing mode?  That is the only change we've made is to lock onto a single device from phase 1 to phase 2 of CloudCutter.

> For other future users who have this issue ... would it be possible to restart the last step (return device to users ssid) without having to keep repeating the whole cloud cutting procedure ?

Technically possible, but incredibly cumbersome.  It would require passing in an SSID/Password, a profile, and at least 4 additional parameters in order to properly communicate with the device.  Generally speaking, it takes less than 3 minutes to get through the full process, so starting over is generally the easiest instruction to follow.

> Sorry to be complaining ... but I am really enthusiastic about using CloudCutter to free devices OTA. Please keep up the good work !!!

What exactly are you looking for out of this issue/ticket?  I'm not convinced there is anything actionable here.  If you're just documenting, that is fine, but then please close the ticket.  You said the last run worked, so I'm really lost as to what this ticket is aside from documentation/sharing.

donburch888, 2023-01-31:
> I'm pretty certain those result in the same thing, our instructions are just the command line version.

I'm still floundering around in linux, but when I followed your instructions (to _install_ network-manager and adjust config files), when I rebooted my RasPi it didn't start _running_ network manager. I assumed there was an extra step I had missed. 

> password is edited out, are there any non-alpha-numeric characters? 

I paid particular attention to that note in the documentatiom.  Fortunately my password is 11 alphanumeric characters only.  I copied it from the WAP configuration page and pasted to the command line to make sure I didn't mis-type it. 

> Technically possible, but incredibly cumbersome. 

Thanks, that's fine.  I just asked in case it might be easy to add an option.  I've been feeling (very unscientific) that my devices are sluggish going into fast and slow blink mode (sometimes requiring button pressed 10 seconds), and wondered if it hadn't been ready for that last step. 

> What exactly are you looking for out of this issue/ticket?

Yep, nothing specific actionable.  It seems that when I don't meticulously document, things don't work and I get asked for all the details I didn't record; and when I do document it seems a waste to just throw away that effort. I guess that's just how Murphy's Law applies to IT.

closed 2023-01-31 by donburch888

Cossid, 2023-01-31:
> I'm still floundering around in linux, but when I followed your instructions (to _install_ network-manager and adjust config files), when I rebooted my RasPi it didn't start _running_ network manager. I assumed there was an extra step I had missed.

That's part of my `-r` is recommended for that host specific version, as it will issue a restart to NetworkManager which if not started will still start it.

> Yep, nothing specific actionable. It seems that when I don't meticulously document, things don't work and I get asked for all the details I didn't record; and when I do document it seems a waste to just throw away that effort. I guess that's just how Murphy's Law applies to IT.

No problem.  We had a huge backlog for a long time, and tickets stayed open.  I've spent a fair amount of effort into getting the issues session into a good state and prefer to keep it that way.  I fully support using them to document findings and make suggestions, but I don't want anything left lingering in the open state that doesn't have a definite call to action that can be resolved.