tuya-cloudcutter/tuya-cloudcutter #273: Dump for ALDI LightZone/Melitec 8251/01/1

yvesdm3000, 2023-01-29:
This is a dump for Dump for ALDI LightZone/Melitec 8251/01/1
I had to desolder the CB3S, I assume the IR sensor or MCU was interfering with communication

[readResult_BK7231N_QIO_2023-29-1--15-56-12.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10530076/readResult_BK7231N_QIO_2023-29-1--15-56-12.zip)

I will wait until you have processed it until I flash a new firmware (last time it was all zero's for another type of device, this time it is not, I checked !).

kuba2k2, 2023-01-29:
@openshwprojects another report of your tool creating 0x0 files - so it's actually a bug then. It would be good if you investigated that issue.

openshwprojects, 2023-01-29:
@kuba2k2 huh, I can clearly see that you're counting the same (and only one so far?) accident twice, as @yvesdm3000 is the user that had the 0x00 byte issue in the first place. Still, I can add even more failsafes for that in tomorrow update. Better safe than sorry!

@yvesdm3000 , were you able to at least once reproduce the problem? I can check at disk write time for 0x0, but consider that before writing to disk, the **MemoryStream has to pass CRC check**, there are no many other places where it can break.

Cossid, 2023-01-29:
@yvesdm3000 Can you please provide a link to the product?

yvesdm3000, 2023-01-29:
https://www.leuchtenservice-melitec.de/LED-SmartHome-Deckenleuchte-D110-schlankes-Design/8251-01-1-CM

Cossid, 2023-01-29:
> Weird I checked the file and was certainly not all zeroes. I also took a dump with your tool and took a md5sum and it was the same

This one is good, he was just referring to your comment.

Cossid, 2023-01-29:
Added as `Melitec 8251-01-1-CM D110 Ceiling Light`

closed 2023-01-29 by Cossid

yvesdm3000, 2023-01-29:
I could not reproduce the zeroes issue

yvesdm3000, 2023-01-29:
[image]
[image]
[image]

yvesdm3000, 2023-01-29:
[image]

yvesdm3000, 2023-01-29:
[image]

yvesdm3000, 2023-01-29:
[image]

kuba2k2, 2023-01-29:
>  huh, I can clearly see that you're counting the same (and only one so far?) accident twice, as yvesdm3000 is the user that had the 0x00 byte issue in the first place. Still, I can add even more failsafes for that in tomorrow update. Better safe than sorry!

Oh, sorry, I didn't realize that was the same person. My bad then.

openshwprojects, 2023-01-31:
No problem @kuba2k2 , please keep me informed if anything happens, we're available here every day to help and fix issues 

@yvesdm3000 very nice photos, please submit them to teardowns list along with the template, thanks!

yvesdm3000, 2023-02-01:
Your tool seems to indicate this is a BK7231T but a WB3S is a BK7231N ?

Cossid, 2023-02-01:
I think you're getting things mixed up.  Your picture shows CB3S which is BK7231N.  The firmware dump you provided was BK7231N with a BK7231N bootloader and BK7231N SDK.

yvesdm3000, 2023-02-01:
I found it doesn't work on a second one (unopened):

```
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 9cba42adac63
Successfully built 9cba42adac63
Successfully tagged cloudcutter:latest
Successfully built docker image
Loading options, please wait...
Selected Device Slug: lightway-smart-al2017-tgts
Selected Profile: oem-bk7231s-light-ty-2.9.16-sdk-1.0.8-40.00
Selected Firmware: custom-firmware/OpenBK7231N_UG_1.15.405.bin
WARNING: Flashing a firmware for the wrong chip may soft-brick your device requiring a serial flash to recover.
You have selected a profile for chip type BK7231T but the selected firmware filename (custom-firmware/OpenBK7231N_UG_1.15.405.bin) indicates it may be for another chip.
Are you sure want to proceed?  Type 'PROCEED' (case-sensitive) to continue or 'exit' to stop (exit/PROCEED): PROCEED
Proceding with selected profile for chip BK7231T and firmware custom-firmware/OpenBK7231N_UG_1.15.405.bin
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
...............................................
Found access point name: "SmartLife-7179", trying to connect..
Error: Connection activation failed: (10) 802.1X supplicant failed.
...............................................
Found access point name: "SmartLife-7179", trying to connect..
Device 'wlp3s0' successfully activated with '700170e8-a698-4fcd-abac-36e0ebfb1f89'.
..............................................
Found access point name: "SmartLife-7179", trying to connect..
Device 'wlp3s0' successfully activated with 'dd277fb1-7dec-4b58-b15b-265d884363bb'.
.............................................
Found access point name: "SmartLife-7179", trying to connect..
Device 'wlp3s0' successfully activated with '44051354-36a2-4660-9723-984dc904bbb4'.
.............................................
Found access point name: "SmartLife-7179", trying to connect..
Device 'wlp3s0' successfully activated with '18c528ac-af3a-451b-a631-899b55f58be9'.
Error, could not connect to SSID.
Failed to connect, please run this script again
```

yvesdm3000, 2023-02-01:
Yes I meant CB3S sorry

Cossid, 2023-02-01:
You're selecting the incorrect profile.

This was added as `Melitec 8251-01-1-CM D110 Ceiling Light`

yvesdm3000, 2023-02-01:
I guess this does not work :-(

```
yves@T3400:~/src/tuya-cloudcutter$ sudo ./tuya-cloudcutter.sh -p melitec-8251-01-1-cm-d110-ceiling-light -f OpenBK7231N_UG_1.15.405.bin 
Checking UDP port 53... Occupied by systemd-resolve with PID 23341.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  23.82MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 8a1ad5297c50
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 1c1a683e2884
Step 3/10 : FROM base AS python-deps
 ---> 1c1a683e2884
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 7b512c4555b7
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> aa06e534356e
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 578637e6e59d
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 231f4f88619a
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 231f4f88619a
Step 9/10 : COPY src /src
 ---> Using cache
 ---> f6af29e4c185
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 9cba42adac63
Successfully built 9cba42adac63
Successfully tagged cloudcutter:latest
Successfully built docker image
Loading options, please wait...
Selected Device Slug: melitec-8251-01-1-cm-d110-ceiling-light
Selected Profile: bk7231n-common-user-config-ty-2.1.6-sdk-2.3.3-40.00
Selected Firmware: OpenBK7231N_UG_1.15.405.bin
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
.................................
Found access point name: "A-7179", trying to connect..
Device 'wlp3s0' successfully activated with '50fb263b-3b84-4769-b69d-66da756d7940'.
..................................
Found access point name: "A-7179", trying to connect..
Device 'wlp3s0' successfully activated with '713c8f87-9fb5-461f-95c2-31cf953a1727'.
..................................
Found access point name: "A-7179", trying to connect..
Device 'wlp3s0' successfully activated with '867df212-cdb0-4d9f-8a64-dd959243ed01'.
.................................
Found access point name: "A-7179", trying to connect..
Device 'wlp3s0' successfully activated with 'f58b3991-262b-4c64-ac64-e59564807bbd'.
................................
Found access point name: "A-7179", trying to connect..
Device 'wlp3s0' successfully activated with '9fb8275d-2ee9-4fd2-9af1-b02acdc7c986'.
Error, could not connect to SSID.
Failed to connect, please run this script again
yves@T3400:~/src/tuya-cloudcutter$
```

yvesdm3000, 2023-02-01:
It did do something, and changed the AP name to A-7179 and then failed. It is now failing all the time.

Cossid, 2023-02-01:
That's purely a connection problem.  Try running with the `-r` attribute.

yvesdm3000, 2023-02-01:
Looks like a bug in the script (or output of network-manager).
I bypassed the gateway check since it seems to contain a lot of newlines and hence fails when it should succeed:

```
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
............................................................................................................................................
Found access point name: "A-7179", trying to connect..
Device 'wlp3s0' successfully activated with 'efe649e0-eee6-4a9f-adc9-76c0760816d4'.
Gateway: 

192.168.175.1
Connected ending: -7179
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/XBIU07ARA9wL.deviceconfig
Saved device config in /work/configured-devices/XBIU07ARA9wL.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
.......................................................................................................................................................................................................
Found access point name: "A-7179", trying to connect..
Device 'wlp3s0' successfully activated with 'b769b562-a9ba-464a-93b3-d55abf4a3f04'.
Gateway: 

192.168.175.1
Connected ending: -7179
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlp3s0
Feb  1 19:21:53 dnsmasq[14]: started, version 2.80 cachesize 150
Feb  1 19:21:53 dnsmasq[14]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Feb  1 19:21:53 dnsmasq-dhcp[14]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Feb  1 19:21:53 dnsmasq-dhcp[14]: DHCP, sockets bound exclusively to interface wlp3s0
Feb  1 19:21:53 dnsmasq[14]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
Using interface wlp3s0 with hwaddr 88:78:73:d4:b0:2f and ssid "cloudcutterflash"
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
Feb  1 19:21:56 dnsmasq-dhcp[14]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Feb  1 19:21:56 dnsmasq-dhcp[14]: 842012285 client provides name: wlan0
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 DHCPDISCOVER(wlp3s0) a0:92:08:9d:71:79 
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 tags: wlp3s0
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 DHCPOFFER(wlp3s0) 10.42.42.13 a0:92:08:9d:71:79 
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 next server: 10.42.42.1
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  1 option: 53 message-type  2
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option: 51 lease-time  12h
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option: 58 T1  6h
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option: 59 T2  10h30m
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option:  1 netmask  255.255.255.0
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option: 28 broadcast  10.42.42.255
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option:  3 router  10.42.42.1
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option:  6 dns-server  10.42.42.1
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 client provides name: wlan0
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 DHCPDISCOVER(wlp3s0) a0:92:08:9d:71:79 
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 tags: wlp3s0
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 DHCPOFFER(wlp3s0) 10.42.42.13 a0:92:08:9d:71:79 
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 next server: 10.42.42.1
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  1 option: 53 message-type  2
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option: 54 server-identifier  10.42.42.1
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option: 51 lease-time  12h
Feb  1 19:21:59 dnsmasq-dhcp[14]: 842012285 sent size:  4 option: 58 T1  6h
[... 57 lines cut ...]
[I 230201 19:22:20 web:2271] 200 POST /d.json?a=tuya.device.property.save&devId=5Su5so8V06YQwW4wyB03&et=1&t=1675279340&v=1.0&sign=c92bf93a3c5d3922c3f0f1e28f8fea5d (10.42.42.13) 100.67ms
[... cut ...]
!!! Endpoint default response not found - tuya.device.upgrade.status.update
[LOG (Server)] Response:  {'success': True, 't': 1675279345}
[I 230201 19:22:25 web:2271] 200 POST /d.json?a=tuya.device.upgrade.status.update&devId=5Su5so8V06YQwW4wyB03&et=1&t=1675279344&v=4.1&sign=eb8fe2af5381b264f84296d673c4a9ac (10.42.42.13) 96.32ms
[I 230201 19:22:38 web:2271] 200 GET /files/OpenBK7231N_UG_1.15.405.bin (10.42.42.13) 13430.79ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=0-616751/616752
```

Cossid, 2023-02-01:
`-r` likely would have resolved that issue.  That happens when your OS creates multiple profiles for the same AP, and the script can't really fix that.

yvesdm3000, 2023-02-01:
No -r did not solve that, the gateway field still contained newlines. I already 'cleaned' the wifi settings by telling to forget all other AP it knew about. However this pc has multiple network interfaces with wifi on a pcie board. I removed the -r because that took extra time and the target device doesn't keep its AP up for a long time.

Cossid, 2023-02-01:
I'm not sure what to tell you then.  I cannot reproduce this issue.  I can't get it to spit out blank lines under any condition.  I'm not even sure how it can, as that is a filter, it doesn't print anything that doesn't match.  I can only assume it created incomplete profiles on your machine, and `-r` should remove all profiles (provided it has permission to clear those files).

There are no obvious changes I can make right now.

All I can recommend is to clear out /etc/NetworkManager/system-connections/ and reboot.
If you have multiple wifi adapters and they are connected to APs, that really isn't supported.

yvesdm3000, 2023-02-01:
Only 1 wifi adapter, a 2 port ethernet and an sfp adapter. I will look at the source where the gateway is extracted and try to update the filter to accomodate this tomorrow.