tuya-cloudcutter/tuya-cloudcutter #274: Profile for Main module version 3.1.4

siqbal1986, 2023-01-31:
Hi guys. Thankyou very much for this wonderful project. I have some Feit 3 way smart dimmers purchased from Costco. I have been able to cloud cut a few of these devices with main module version 1.1.71 quite easily. 
But I am not able to find a profile to match the dimmers running 3.1.4.
Has anyone seen this firmware and know if it is exploitable. 
My fear is that this is the version in which tuya has fixed the exploit.

If I want to go about figuring out the profile can someone please guide me.

Cossid, 2023-01-31:
It has been incredibly rare to see a firmware version 3.x.x and be Beken.  It's not impossible though.  Even patched devices (We've only seen one so far) have just been an incremental on 1.1.x

Does the device have an FCCID printed on the back?  Can you put the device in AP mode, and do a wifi scan and get the BSSID of the AP?

Usually 3.x.x were ESP devices.

siqbal1986, 2023-01-31:
I have tried using tuya-convert in the hopes that they might be esp devices but that doesn't work. These devices can be put in AP mode and they give ssid  like "SmartLife-XXXX"

siqbal1986, 2023-01-31:
I have them installed but I can try to open one up tomorrow and send you the pictures

Cossid, 2023-01-31:
The BSSID is different than the SSID.  The BSSID will look like a mac address in the form of 00:00:00:00:00:00.  You'd need a 3rd party program such as Wifi Analyzer on windows.  Linux you can usually find it through `nmcli dev wifi`

Even if the device is ESP, Tuya-Convert likely won't work, as that has long since been patched.

siqbal1986, 2023-01-31:
It seems you were correct. It's based on esp and not BK.

Cossid, 2023-01-31:
Yeah, nothing CloudCutter can currently do for this device then.

closed 2023-01-31 by Cossid