tuya-cloudcutter/tuya-cloudcutter #275: Ubuntu 20.04 - Docker Command not found - line 99

craigcurtin-dev, 2023-01-31:
Guys,

New Virtual machine (VMWare Workstation) - running a Ubuntu 20.04 VM - freshly created

Passed through a USB WIFI adapter - verified it is seen and operational in the VM - performed system updates

Logged in as normal user

Installed Docker 

Created Docker group and added user account to group

Rebooted

performed a git-clone of the respository into my home directory - all passed OK

changed into the directory and performed ./tuya-cloudcutter.sh

Got an error: Common.sh line 99 Docker command not found

Any ideas ?

Craig

Cossid, 2023-01-31:
Two things:

First) It is failing because it can't locate the docker binary.  Your bash environment does not seem to have a path to available to docker.  The command that bash script is running at that point is `docker build --network=host -t cloudcutter .`
Second) You need to run elevated (sudo) as it will need to manipulate the wifi interface via NetworkManager, and elevation is required for that.  Make sure your elevated user also has access to docker.

craigcurtin-dev, 2023-02-01:
Thanks for that - turns out i had installed docker but not the docker-cli tools etc.

Also note that the automated script attempts (and does) stop the DNS service when it senses the port is in use - but then the rest of the script fails as docker can not locate the components to download for the image.

Once i commented out the steps in the sh script all ran fine

Turns out my device was not supported as it was based on a realtek chipset so a good learning exercise

Craig

Cossid, 2023-02-01:
> Also note that the automated script attempts (and does) stop the DNS service when it senses the port is in use - but then the rest of the script fails as docker can not locate the components to download for the image.

Unfortunately, there is little we can do there.  If port 53 is in use, that means the machine is running a DNS server, and if it kills DNS, that means the machine is configured to use itself as DNS (this is not the usual case, but still a valid case).  Everything continued for you because you had a device that was not exploitable.  If you had made it through the exploit, there is a step that requires running a DNS server on port 53 that would have failed, which is why we have the check in place.

The only recommendation we generally give is to still kill the DNS but then also add external DNS servers to your resolv.conf file.

closed 2023-02-01 by Cossid

jacobefaust, 2023-02-06:
@craigcurtin-dev, @magoone7 (from Issue #269).  I ran into the same issue when running on Ubuntu 18.04 LTS.  It turns out that `systemd-resolved` is configured to park on port 53 (as @Cossid describes) for many versions of Ubuntu out-of-the-box.  I was able to free up port 53 by changing `systemd-resolved's` settings as described here: https://github.com/systemd/systemd/issues/14700#issuecomment-1046279509 .