tuya-cloudcutter/tuya-cloudcutter #277: SL-SC-01 shutter firmware

openshwprojects, 2023-02-01:
Hello,
A device with strange CB2S module with mixed pad markings, can anyone check and try to add a profile?
[SL-SC-01.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10557882/SL-SC-01.zip)
More info will be added if needed.
Thanks!

kuba2k2, 2023-02-01:
Mixed pad markings? What does that mean?

Cossid, 2023-02-01:
Added as Spacetronik SL-SC-01 Curtain Controller.  I wasn't sure if was generic or not, I could only find it as "Spacetronik", so I added it as that.  If it should be Tuya Generic, let me know and I'll update it.

https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/spacetronik-sl-sc-01-curtain-controller.json contains the device configuration.

closed 2023-02-01 by Cossid

Cossid, 2023-02-01:
Unfortunately, not really.  I don't see any strings in the binary that would give useful information.  We rarely have much to go on beyond that device_configuration pin mapping.  But it shouldn't take much physical trial and error to get that figured out.