tuya-cloudcutter/tuya-cloudcutter #278: Flashing ends too soon

dadavec, 2023-02-02:
Hello,
I am attempting to flash a smart ceiling light with OpenBK7231T/OpenBeken release 1.15.405. I was already able to flash another device of the same model using your tuyacutter method, but I cannot acomplish it again for another piece of device (same model).

I believe the issue is that the script ends the execution too soon. I was running a Wireshark while flashing and I have noticed MQTT messages between my flasher (laptop) and the device. In those messages is written keyword "progress" with a numeric value which is increasing over time and "firmware Type", which is always set to 0.

I have run the flash multiple times using same params and I have noticed that the last MQTT message with progress was always between 30-50 (I assume it is percentage). The time between MQTT message containing "tuya.device.upgrade.status.update" which is the last MQTT message before all those "firmware progress" messagess appear is 30 seconds. This is exactly the same as the timeout mentiond in the bash script ("Device will download and reset. Exiting in 30 seconds.").

Would it be please possible to either extend this 30 seconds exit time? Or could you be reading the progress from the MQTT to find out if the firmware was really succesfully finished?

dadavec, 2023-02-02:
Flashing log:
`Selected Device Slug: cloudy-bay-6inch-recessed-rgbct-downlight-v3-cb2l-1.1.7
Selected Profile: oem-bk7231n-ceiling-light-ty-1.1.7-sdk-2.3.1-40.00
Selected Firmware: /work/custom-firmware/OpenBK7231N_UG_1.15.405.bin
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
...
Found access point name: "A-AFC6", trying to connect..
Device 'wlp5s0' successfully activated with 'f8c04713-7905-4b7a-9dee-3b6c3b1a9db5'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/kJctP8M84CeA.deviceconfig
Saved device config in /work/configured-devices/kJctP8M84CeA.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
..............
Found access point name: "A-AFC6", trying to connect..
Device 'wlp5s0' successfully activated with 'f8c04713-7905-4b7a-9dee-3b6c3b1a9db5'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlp5s0
Feb  2 16:49:49 dnsmasq[15]: started, version 2.80 cachesize 150
Feb  2 16:49:49 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Feb  2 16:49:49 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Feb  2 16:49:49 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlp5s0
Feb  2 16:49:49 dnsmasq[15]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
Using interface wlp5s0 with hwaddr 0c:d2:92:0d:fc:ff and ssid "cloudcutterflash"
wlp5s0: interface state UNINITIALIZED->ENABLED
wlp5s0: AP-ENABLED 
Feb  2 16:49:55 dnsmasq-dhcp[15]: 1804511774 available DHCP range: 10.42.42.10 -- 10.42.42.40
Feb  2 16:49:55 dnsmasq-dhcp[15]: 1804511774 client provides name: wlan0
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 DHCPDISCOVER(wlp5s0) 38:1f:8d:79:af:c6 
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 tags: wlp5s0
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 DHCPOFFER(wlp5s0) 10.42.42.17 38:1f:8d:79:af:c6 
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 next server: 10.42.42.1
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  1 option: 53 message-type  2
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 54 server-identifier  10.42.42.1
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 51 lease-time  12h
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 58 T1  6h
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 59 T2  10h30m
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option:  1 netmask  255.255.255.0
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 28 broadcast  10.42.42.255
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option:  3 router  10.42.42.1
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option:  6 dns-server  10.42.42.1
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 available DHCP range: 10.42.42.10 -- 10.42.42.40
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 client provides name: wlan0
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 DHCPREQUEST(wlp5s0) 10.42.42.17 38:1f:8d:79:af:c6 
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 tags: wlp5s0
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 DHCPACK(wlp5s0) 10.42.42.17 38:1f:8d:79:af:c6 wlan0
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 next server: 10.42.42.1
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  1 option: 53 message-type  5
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 54 server-identifier  10.42.42.1
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 51 lease-time  12h
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 58 T1  6h
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 59 T2  10h30m
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option:  1 netmask  255.255.255.0
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option: 28 broadcast  10.42.42.255
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option:  3 router  10.42.42.1
Feb  2 16:49:59 dnsmasq-dhcp[15]: 1804511774 sent size:  4 option:  6 dns-server  10.42.42.1
Feb  2 16:50:34 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.17
Feb  2 16:50:34 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
[... cut ...]
[I 230202 16:51:02 web:2271] 206 GET /files/OpenBK7231N_UG_1.15.405.bin (10.42.42.17) 122.86ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=309167-616751/616752
[I 230202 16:51:07 web:2271] 206 GET /files/OpenBK7231N_UG_1.15.405.bin (10.42.42.17) 125.54ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=309167-616751/616752
root@david-MSI-NTB-UBUNTU:/home/david/Stažené/tuya-cloudcutter-main# 
`

Example of last MQTT message:
`{"protocol":16,"data":{"progress":"50","firmwareType":0}}`

Cossid, 2023-02-03:
The issue isn't directly that it's ending too soon, it's that it appears to be making multiple requests to update firmware and stepping over itself and none of them finish (actually just flat-out stalling).  When it is working correctly, the sending of firmware usually only takes 5-15 seconds (usually 8-11 depending on size).

I have pushed an update that will limit firmware requests to once per session instead of any time a specific tuya request is made (which could happen multiple times in some cases).  Can you please test it out and see if that makes any difference at all?

dadavec, 2023-02-03:
Thank you a lot for checking the issue. I have downloaded latest code and did another three attempts to flash the device. I can see there is still roughly 30 seconds between start and end of the MQTT firmware related messages.

I have one more observation, it feels like the firmware percentage is increasing with each attempt. I mean that the last MQTT message has the progress higher than previous attempt. Last attempt stopped at 75%. 

I have collected logs of all three tests and also screened a Wireshark window with interesting messages. You can find them all attached here.

[ccuter-testing.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10582788/ccuter-testing.zip)

Cossid, 2023-02-03:
Thanks, I'll look into that.  We don't currently have a hook into the MQTT logs, and maybe that is going to be necessary.  The odd part is that the http response is always the same byte size, never moving.  It either says 0-[filesize-1]/filesize and finishes with no further messages, or it starts repeating a message at a specific offset size that never changes.

I'll look into what it will take to hook into that, as it would give much better progress output, as well as something we can hook onto to verify it is done rather than just assuming a timeframe (I don't prefer just extending the timeframe, that had also been tried before and didn't seem to help much, and made the experience for successful cases worse).

Cossid, 2023-02-04:
Can you please update and give it another run and post log output (successful or not)?

I have added MQTT logging which should give better insights.  Unfortunately, we can't hook on a MQTT message for timing either, as it doesn't give a conclusive "finished" status either.  I have moved where the hook is to start the clock a bit later, and increased the time to 60 seconds.

dadavec, 2023-02-04:
I did another three tests. Here are the results: [cutter-testing2.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10609741/cutter-testing2.zip)

In summary, the progress was significantly lower than before (current top was 14%) and seemed to not really increase over time. Even though the final progress was again increased between each test. 

Note: there is "just a visual" issue with 
> common.sh: line 66: [: ==: (expected unary operator)

Dots are not showing properly when waiting for AP discovery

Cossid, 2023-02-04:
> Note: there is "just a visual" issue with
> 
> > common.sh: line 66: [: ==: (expected unary operator)
> 
> Dots are not showing properly when waiting for AP discovery

Thanks for the report, bug from a different adjustment, fixed now.

> I did another three tests. Here are the results: [cutter-testing2.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10609741/cutter-testing2.zip)
>
> In summary, the progress was [...]

How far is your target device from your cloudcutter device?  AP mode is not very strong, and that's usually the leading case for firmware upgrades failing.  This continues to show the same signs as the usual fail case where it downloads a chunk, then stops, and continually tries to resume but doesn't actually result, requesting the same chunk over and over.

Part of the potential is that it requests and "upgrade" and then follows up later with a "silent upgrade" and it seems to interrupt/restart the process, and we have no control over that flow aside from potentially responding with a different status/response to one of them, but we don't have enough data to know if that is correct/possible.

Sometimes it is a physical thing like distance, sometimes the firmware just has bad networking code, and luck is the only way.

I'll keep looking to see if there is a bug in resuming.  The only odd thing I'm seeing is how quickly it returns (usually the full request takes 5-15 seconds, but all the resumes finish in less than 200ms, no matter the size.

Cossid, 2023-02-04:
@dadavec Would you mind trying an experiment for me?

https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/64d076c11d71309138617b412f2ed49a32129b4d/src/cloudcutter/__main__.py#L125 Could you comment out that line (/src/__main__.py line 125) by adding a `#` character to the beginning of it, and running cloudcutter firmware flash?  Again, post the log success or fail.  Running once is all I need to see.

Cossid, 2023-02-05:
Disregard the request to change a file, just do a `git pull` and try the latest changes instead.

dadavec, 2023-02-05:
I have succeeded to flash the device successfully using latest code. I have observed the time to flash the device was around 30 but dangerously close to 40 seconds for the second device I tested.

> How far is your target device from your cloudcutter device? 

Roughly 1.5 meter. It is a ceiling light an I have the laptop on a table :)

You are correct about device not sending any message after successful flash. But I think you can do two types of monitoring. One is by ICMP (PING). When the device was successfully flashed, it immediately became unreachable, while when it failed (not finished in given time) it remained connected. Second can be monitoring of MQTT progress, where you can measure for how long the progress value did not change. I think it could be enough to have a 10 seconds between not changed progress measurements, which would indicate a failure. The first change between "0" and something was the longest. But that would need testing on much broader set of devices.

Also in both cases the last progress message was 98%. I am not sure how reliable this information is, but it can be the same in other devices.

Here are logs of both devices: 
[success.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10611499/success.txt)
[succes_another_device.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10611501/succes_another_device.txt)

Cossid, 2023-02-05:
Yeah, for me the last successful mqtt message was always 98% as well on the 3 different firmwares I tried in testing.

The problem with ICMP or even MQTT monitoring is that when it works it's simple, but in the failure case we had before, the device would sometimes briefly disconnect (usually indicated by another DHCP request block).

I think I've got the failure case mitigated out, so I think I'm just going to put a 10 second timer after receiving a mqtt status > 95% or something like that, or the on_finish from the file request when the range was 0-[max-1] (which I avoided before because more would come in later with partial requests, but that appears mitigated as well).

dadavec, 2023-02-05:
If all those failure causes are mitigated, the suggested monitoring should be enough. I think only time will tell if there will be any other issues with flashing process. I want to thank you a lot for you great and extremely fast support. I believe this issue can now be considered as fixed and can be closed.

Cossid, 2023-02-06:
Yeah, it seems they are mitigated after further testing.  I've now also changed it so that it won't start the kill timer until it gets notification from MQTT of a progress of > 95%, and that timeout is now configurable with a `-t` parameter just in case (the default is now much shorter because it should start when the update is pretty close to complete).

I've also cleaned up output since this wraps up one of the bigger issues we were still having and had so much output enabled for.  It will now be much cleaner by default and `-v` will enable the rush of useful debugging information.

closed 2023-02-06 by Cossid