tuya-cloudcutter/tuya-cloudcutter #279: Unknown parameter: yes

grumpydolphin, 2023-02-04:
Hi,  I have tried running the script and successfully get to the flashing stage, but this is what happens.  the last two lines just repeat.  Is this expected?  

Selected Device Slug: tuya-generic-e27-rgbct-wb2l-m1-bulb
Selected Profile: oem-bk7231n-light-ty-1.2.16-sdk-2.3.1-40.00
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for open Tuya SmartLife AP
Unknown parameter: --rescan
Unknown parameter: yes
.Unknown parameter: --rescan
Unknown parameter: yes

Cossid, 2023-02-04:
No, that is not expected output.  You may have an outdated version of NetworkManager/nmcli.  Can you post the output of `nmcli --version`?  What OS (and version of it) are you running?  You may need to update your `network-manager`/`NetworkManager` package on your OS.

Cossid, 2023-02-04:
I have move the `--rescan yes` portion before the interface specification, `git pull` and see if it gives the same error or a different one.  This might be an issue with your wifi device name as well.

grumpydolphin, 2023-02-04:
Thanks for the quick reply!

I have the same result after the update.

This is running on Ubuntu 18.04.6
nmcli tool, version 1.10.6

grumpydolphin, 2023-02-04:
I have the same result after the update.

This is running on Ubuntu 18.04.6
nmcli tool, version 1.10.6

Cossid, 2023-02-04:
I couldn't find a way to automatically disable rescan if it's not supported, so I have added a flag to manually be able to disable it.

Please update via `git pull` and add the parameter `-n`.  Please let me know if that works as expected.

Cossid, 2023-02-07:
Closing as fixed, as the fix is in the repo.  If you still have further issues, please reply and re-open this issue.

closed 2023-02-07 by Cossid