tuya-cloudcutter/tuya-cloudcutter #280: Ledvance Smart+ A60 RGBW AC33911

avassdal, 2023-02-06:
https://www.ledvance.com/consumer/products/smart-home/smart-home-products-with-wifi-technology/smart-lamps-with-wifi-technology/smart-classic-lamps-with-wifi-technology/classic-bulb-shape-with-wifi-technology-c6443?productId=83025

[Ledvance_Smartplus-A60-AC33911-RGBW.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10663017/Ledvance_Smartplus-A60-AC33911-RGBW.bin.zip)

avassdal, 2023-02-06:
Regarding the Tuya OEM API, have a look at this: https://github.com/rospogrigio/localtuya/pull/1131

Cossid, 2023-02-07:
This has been added as LEDVANCE SmartPlus AC33911 RGBCT Bulb.

That link is interesting, but is not the API we use in cloudcutter, and I'm not going to attempt to implement that as it requires entering credentials, which is never a good idea.  I have however found that if you follow the profile builder instructions, but instead open the LEDVANCE Smart+ Wifi app instead of the Smart Life app, the token it generates is enough to work over the regular API, so it's restricted by the generated token.

We actually had another dump from #229 but I wasn't able to locate the correct LEDVANCE app until now, as it is region locked and I needed to find an alternate source.

closed 2023-02-07 by Cossid