tuya-cloudcutter/tuya-cloudcutter #281: Profile for LOHAS LED

grumpydolphin, 2023-02-08:
I'm trying to use perform the unlock on some LOHAS LED bulbs and I cannot find the correct profile to use.

It connects (as below) but goes into a loop

Found access point name: "SmartLife-69E3", trying to connect...
Device 'wlx1cbfcebf71bc' successfully activated with '6880ca05-d2d4-4927-93d6-3de67c63dea6'

Cossid, 2023-02-08:
You'll have to provide a full log, showing two lines and stating it goes into a loop is incomplete information, because it has to post something else to keep looping.

If it did not post `Connected to access point.` then it never fully connected are was unable to determine the gateway IP of the device it connected to is 192.168.175.1 and would have posted some other message.

If possible, after it posts the `successfully activated` message, run `nmcli -f IP4.GATEWAY con show "${AP_MATCHED_NAME}"` and post the output.

As for what profile, we can't provide help with no information.  If it doesn't exactly match an existing profile, you'll have to find out the firmware version and go from there.  Are you certain your device is beken based?  Most Lohas bulbs I've seen have been ESP.

grumpydolphin, 2023-02-08:
Unknown parameter: yes
.Unknown parameter: --rescan
Unknown parameter: yes

Found access point name: "SmartLife-69E3", trying to connect...
Device 'wlx1cbfcebf71bc' successfully activated with
'c82ab097-9d23-4492-be48-d9c43358d9ab'.
.Unknown parameter: --rescan
Unknown parameter: yes
.Unknown parameter: --rescan
Unknown parameter: yes
.Unknown parameter: --rescan

nmcli -f IP4.GATEWAY con show "${AP_MATCHED_NAME}"
Error:  - no such connection profile.

PING 192.168.175.1 (192.168.175.1) 56(84) bytes of data.
64 bytes from 192.168.175.1: icmp_seq=1 ttl=255 time=27.7 ms
64 bytes from 192.168.175.1: icmp_seq=2 ttl=255 time=35.5 ms
^C

Cossid, 2023-02-08:
Did you update your git repo since your last ticket?

grumpydolphin, 2023-02-08:
Yes, I did.

Cossid, 2023-02-08:
Then you need to use the `-n` parameter to disable rescan.

Cossid, 2023-02-15:
Closing as stale.  Please re-open with complete logs if `-n` does not resolve your issue.

closed 2023-02-15 by Cossid