tuya-cloudcutter/tuya-cloudcutter #282: Getting Missing argument a when trying to flash MI-BW904-999W

kogeki, 2023-02-10:
Hello,

I am trying to flash the MI-BW904-999W. It seems to work up until it connects to the Pi's AP. It also does this when trying to run the normal script (not trying to flash CFW). I know that this device isn't specfically supported. I used `1.0.10 - BK7231T / oem_bk7231s_light_ty_oldDp` . Any help would be appreciated, thanks for this great project!

```
Scanning for open Tuya SmartLife AP
.
Found access point name: "A-70F3", trying to connect...
Device 'wlan0' successfully activated with '79147445-d951-493a-a618-cf9888554988'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr **:**:**:**:**:** and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 99.76ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 201.08ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 53.54ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 49.20ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 50.61ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 51.01ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 50.90ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 50.77ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 46.78ms
Using PSK v1 - Received PSK ID version 01
400 POST /device/url_config (10.42.42.11): Missing argument a
400 POST /device/url_config (10.42.42.11) 49.77ms
```

Cossid, 2023-02-10:
Sorry about that, bad logging call.  Please `git pull` and try again.

kogeki, 2023-02-10:
Works perfectly now. Thanks!

closed 2023-02-10 by kogeki