tuya-cloudcutter/tuya-cloudcutter #283: Suspect  I need a profile for the Globe 50323 and 50069 down light

Deep-Six, 2023-02-13:
Hey everyone, I'm trying the flash via the version number which was 1.1.0 as noted in the app (after a forced update)

I'm only getting these messages contiously:

Found access point name: "Globe Suite-3DEC", trying to connect...
Error: No network with SSID 'Globe Suite-3DEC' found.
Error: Globe Suite-3DEC - no such connection profile.

I chose this profile as it was the only 1.1.0 that wasn't a curtain switch:
1.1.0 - BK7231T / HYS_03_018_bk7231t_tjsama0rocl95kdn
Tuya Generic
Fairie Lights

Am I understanding things correctly, should this have worked?

thanks for the great work.

Deep-Six, 2023-02-13:
Update I tried using Globe Electric instead and got a little further:

Scanning for open Tuya SmartLife AP
.
Found access point name: "Globe Suite-3DEC", trying to connect...
Device 'wlan0' successfully activated with '96536d52-8304-415a-84c6-efd0ea35713e'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/aDVF2z7YCl3F.deviceconfig
Saved device config in /work/configured-devices/aDVF2z7YCl3F.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
....
Found access point name: "Globe Suite-3DEC", trying to connect...
Device 'wlan0' successfully activated with '96536d52-8304-415a-84c6-efd0ea35713e'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================

Cossid, 2023-02-13:
Unfortunately, there is currently no compatible profile for your device.  Neither curtain nor fairy light will work with your downlights.  We would need a serial or lightleak dump (see https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps).

And be aware that updating your device firmware may patch the vulnerability that Tuya-CloudCutter uses.

Deep-Six, 2023-02-13:
I'll see what I can do to get a dump. Re updating, that's one of the challenges of the suggestions to look for the version numbers in the update field. It ends up forcing you into an update process. Is there any way to identify the version without going into that view?

Cossid, 2023-02-13:
No, there is not, but you also don't have to update, you can decline/cancel, you just won't see your current version number.

When there is a pending update, it will show you what version number is available to update to, but it won't show you what is currently installed anymore, you can only assume it is lower than the update version, but never be sure which exact version.

Either way, Globe Electric seems have custom firmware which means it's not likely to share profiles with other devices.  This is based on the assumption that the one device we have from Globe Electric (a night-light), has the brand "globe" directly in their profile name, which is usually an indicator it is not shared.

Deep-Six, 2023-02-13:
Ok I'll try and do the dump tonight after work!

Cossid, 2023-02-13:
If you have more than one and haven't updated them all, please dump a non-updated device, as we may be able to get both profiles that way.

Deep-Six, 2023-02-13:
I have dozens of them, and several un updated 50069's which are the larger 6" ones. I'll try to do both.

Deep-Six, 2023-02-20:
Trying to generate a profile, this is the guidance given. Is this accurate, I have no Socket (Wi-Fi) network. The AP from the invocation is ap_ssid: Globe Suite but when the light is put in AP mode, it's showing an SSID of  Globe Suite-3DEC. 

I'm a little confused does the device I'm running the script on (an rpi3) need to be on the same network as the light? Ie is I have a network foo2.4 that the Pi is joined to, and I can select it in the Smart Life App, or does both the Pi and the light need to be on the Globe Suite AP above? No network with "Socket Wi-Fi" Shows up. I've tried adding the device while both the light and the rpi is on the foo2.4 network above, and though I add the light, it doesn't ever seem to complete the token flow.

Here's part of the output below:

[!] No token provided.  On the same network, please log into Smart Life, start the add device procedure, select 'Socket (Wi-Fi)', enter your network credentials, next until it asks the status of the indicator and select 'Blink Slowly', select 'Go to Connect', then in your wifi selection screen, hit the back button to return to Smart Life.  A new token should be sent to your network, and this script will continue.
[!] Note: this will join the device to your account.  You can safely delete it afterwards.
[+] Waiting for multicast token from app...

Cossid, 2023-02-20:
The device you are running profile builder from must be on the same network as the device you are running the smart life application from.  At the point you have a firmware dump of a device, that device is no longer a part of the profile building process, just the data pulled from it.  "Socket Wi-Fi" is an option in the smart life app when attempting to manually pair devices.

Deep-Six, 2023-02-20:
Ok, so if I have both my mobile (running smart life app) and the rpi (where I'm running the profile builder on) that should work, but thus far I'm not seeing the multicast aspect of things. Is there Socket Wi-Fi aspect only needed if the automatic joining doesn't work?

Deep-Six, 2023-02-20:
Oh my sorry I realized it's advising me to use Socket Wifi.
Let me try to explicitly read the instructions 👍🏼

Cossid, 2023-02-20:
The smart life app starts the multicast procedure at the end of the instructions provided by the profile builder app.

* log into Smart Life
* start the add device procedure (hit the plus in the upper right hand corner then 'add device')
* select 'Socket (Wi-Fi)'
* enter your network credentials
* hit next until it asks the status of the indicator and select 'Blink Slowly'
* select 'Go to Connect', then in your wifi selection screen
* hit the back button to return to Smart Life.

A new token should be sent to your network, and this script will continue.

Deep-Six, 2023-02-20:
Apologies, I was making far too many assumptions, I at least saw some activity now, but I got this error:

[!] Unable to get a response from Tuya API, or response was malformed.
[!] http://a.tuyaus.com/d.json must not be blocked in order to pull data from the Tuya API.
Traceback (most recent call last):
  File "/home/jhenkel/tuya-cloudcutter/profile-building/tuya_api_connection.py", line 37, in request
    response_body_json = json.loads(response_body)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Unterminated string starting at: line 1 column 11 (char 10)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jhenkel/tuya-cloudcutter/profile-building/build_profile.py", line 48, in <module>
    pull_schema.run_directory(extracted_location, token)
  File "/home/jhenkel/tuya-cloudcutter/profile-building/pull_schema.py", line 265, in run_directory
    run(directory, output_file_prefix, uuid, auth_key, product_key, firmware_key, software_version, baseline_version, cad_version, cd_version, protocol_version, token)
  File "/home/jhenkel/tuya-cloudcutter/profile-building/pull_schema.py", line 184, in run
    response = connection.request(url, params, data, "POST")
  File "/home/jhenkel/tuya-cloudcutter/profile-building/tuya_api_connection.py", line 48, in request
    if (response_body_json is not None):
UnboundLocalError: local variable 'response_body_json' referenced before assignment

I'm going to try and re-do all the steps fresh, hopefully I've messed stuff up, and it'll work next time.

Deep-Six, 2023-02-20:
No luck, still the same error.

Cossid, 2023-02-20:
Did you install all the pip packages per the instructions?

If you did, that can only mean you have a firewall issue, as that message is stating you're getting some sort of response back, but it's not json, and tuya's server only responds with json pass or fail.

Failing that, you can privately send your dump to me via email ( cossid `at` cossid.com ) and I can generate the profile.

Deep-Six, 2023-02-20:
Yes, though the instructions don't seem to match my experience in the app, 
I select Socket Wifi then I get:

Select 2.4Ghz Wi-Fi Network and enter password
Reset the device -> Next
Reset the device (Press and hold the RESET button for 5s) -> Next
Reset the device (Confirm the indicator is blinking) -> Next
Select the status of the indicator light or hear the beep -> Blink Slowly
(A pop up pops up to add some Fans that I haven't yet added, hoping to make those local after!) -> Back
Connect your mobile phone to the devices hotspot -> Go to Connect 
(Wifi Screen) -> Back
Nothing seems to happen... (Reconnect)..nothing seems to happen, but if I select the text  (Confirm hot spot connection, next) I see action on the screen and the error, above.

I'll send you the dump here shortly. Firewall should be stateful, so if it initates outbound it should allow back.

Cossid, 2023-02-20:
Yeah, the instructions are likely changed because you have pending devices broadcasting bluetooth and smart life is reacting differently as a result.  The instructions included are fully for a manual add.

You shouldn't be following any of the instructions in the smart life app (actually putting devices into any specific mode), but rather be following the instructions printed literally.

Cossid, 2023-02-20:
A new profile has been added as `Globe Electric 50323 RGBCT Recessed Light`

Cossid, 2023-02-20:
@Deep-Six is there a difference between 50323 and 50069, or is 50069 just a two-pack of 50323?

Deep-Six, 2023-02-20:
> ~~@Deep-Six is there a difference between 50323 and 50069, or is 50069 just a two-pack of 50323?~~
> Nevermind, apparently 50323 is 4" recessed, and 50069 is 6" recessed.

Correct! I have some 6" I can do later on!

Cossid, 2023-02-20:
Are they the same firmware version?  They might work fine with the 4" profile.
Some other Globe Electric devices were more standard tuya firmware, but that 4" was specific to the globe brand.

Deep-Six, 2023-02-20:
I'll try it on a 6" if you like.

Cossid, 2023-02-25:
Closing issue as complete.  If you find the other device doesn't cut with the same profile, please open a new issue for that device.

closed 2023-02-25 by Cossid

Deep-Six, 2023-03-18:
Hey @Cossid  not sure if you want me to post here or not here was my run:

Found access point name: "Globe Suite-0D56", trying to connect...
Device 'wlan0' successfully activated with '8e2dbefc-7ed7-4025-beee-d716e0d832fa'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/El0CFvEwqQjk.deviceconfig
Saved device config in /work/configured-devices/El0CFvEwqQjk.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "Globe Suite-0D56", trying to connect...
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
..........................
Found access point name: "Globe Suite-0D56", trying to connect...
Device 'wlan0' successfully activated with '8e2dbefc-7ed7-4025-beee-d716e0d832fa'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================

Deep-Six, 2023-03-18:
There's nothing in the logs folder.

Cossid, 2023-03-18:
Which device is this for?  This means it has a different firmware version more than likely.

Deep-Six, 2023-03-18:
This was for the 50323 from which I extracted the fw.

Cossid, 2023-03-18:
I'm not sure what you tell you.  I have verified the dump you sent me adds to smart life as `50232` so it is the right dump, and the profile I generated is from that firmware.  Is this the very same device you did a firmware dump of, or just the same model?

Deep-Six, 2023-03-18:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you can confirm the sender and know the content is safe.

Same model.