tuya-cloudcutter/tuya-cloudcutter #284: Profile request for ECASA Smart Plug 16A

ataa, 2023-02-13:
Here is factory firmware backup of the device (not updated):

https://mega.nz/file/J0pnWDqZ#GwfW2qZwYa3elxuxI-oN03JQuRz8m7Coy-aFHDoTb8U

I was able to flash openbk using serial cable but was not able to flash it using cloudcutter.

[image]

Cossid, 2023-02-13:
There were actually a few profiles already present that would have worked for flashing (Aoycocr X5P Plug, Nedis WIFIP130FWT Smart Plug), but they do have different schemas and configurations.

A new schema has been added as ECASA 16A Smart Plug.

closed 2023-02-13 by Cossid

ataa, 2023-02-13:
Tuya-cloudcutter can not find this device in AP mode.

`Scanning for open Tuya SmartLife AP
................................................................................                                                                                               ................................................................................                                                                                               ................................................................................                                                                                               ................................................................................                                                                                               ................................................................................                                                                                               ................................................................................                                                                                               ................................................................................                                                                                               ................................................................................                                                                                               ................................................................................`

`:~/tuya-cloudcutter $ sudo iwlist wlan0 scan
 
         Cell 03 - Address: D8:1F:12:XX:XX:XX
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
                    Quality=62/70  Signal level=-48 dBm
                    Encryption key:off
                    ESSID:"SmartLife-3088"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000006ace060
                    Extra: Last beacon: 30ms ago`

Cossid, 2023-02-13:
CloudCutter uses NetworkManager/nmcli to scan wifi, it must be set up appropriately.

`nmcli -t -f SSID,SECURITY dev wifi list --rescan yes ifname wlan0` is the command CloudCutter is running to search for wifi networks.  (you can remove -f SSID,SECURITY to see more details)

ataa, 2023-02-14:
Thank you for your reply,

I followed [this guide](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md) to install it, but the command above does not list any AP at all. not sure how to fix that, running `nmcli` is showing:
```
wlan0: unavailable
        "Edimax EW-7811Un"
        wifi (rtl8192cu), CE:3A:5A:XX:XX:90, hw, mtu 1500
```

Cossid, 2023-02-14:
Try `nmcli radio wifi on`

CloudCutter should be doing that as well though.

ataa, 2023-02-14:
Fixed the issue by running:
```
sudo systemctl disable dhcpcd
sudo systemctl stop dhcpcd
```