tuya-cloudcutter/tuya-cloudcutter #286: Need a profile build for BHT-02 with WB3S

djanknecht, 2023-02-15:
Hi,

i sucessfully managed to read the flash of a BHT-02 thermostate with "Lightleak". It worked for me like described in the documentation. Great work, thanks, guys!

Now i need help to build the profile - the image is attached.
May somebody create the profile for me? Next i will go on with flashing OpenBeken...

Thanks and regards, Dirk

[dump_with_storage_bht-02.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10747591/dump_with_storage_bht-02.zip)

Cossid, 2023-02-15:
Which version of the BHT-002 is this?

djanknecht, 2023-02-15:
It's a BHT-002-GBLW
https://www.amazon.de/dp/B087D6YWH6

Cossid, 2023-02-15:
Unfortunately, it looks like most of this dump is null bytes.  Are you using the most recent version of LightLeak android app?
Even the file size is wrong on this one, not sure what happened here

@kuba2k2 Any further insight?

djanknecht, 2023-02-15:
This was the result of doing a "Read Storage".
This seems to be the correct dump:
[dump_bht-002.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10747949/dump_bht-002.zip)

Cossid, 2023-02-15:
Thanks, that dump looks good.
Device added as `Tuya Generic BHT-002GB WIFI Thermostat`
Unfortunately, there is no device configuration for this one.

djanknecht, 2023-02-15:
OK, thank you. No device configuration means what?

Cossid, 2023-02-15:
It means there is no information about GPIO mapping, you will have to trace/reverse that yourself.  I'm guessing it's just TuyaMCU on RX/TX anyway.

closed 2023-02-15 by Cossid

djanknecht, 2023-02-15:
Yes, seems like that. Not much connected here.
[image]

So any chance to get it working?

Cossid, 2023-02-15:
This is issues for CloudCutter, you'll have to consult the support of whatever you put on it.

ESPHome should support it with the Tuya module and climage component for Tuya.  OpenBeken, I can't speak to their climate support over TuyaMCU.