tuya-cloudcutter/tuya-cloudcutter #293: DNS hiccup

tjkolev, 2023-02-24:
Greetings!

Thank you for this great tool, and excellent exploit investigation!

I just finished successfully cloudcutting this smart plug (US):
Model: KMC 30153
Chip: BK72301N
Board: CB2S
Firmware version: 1.1.8

The script was run on Lenovo Ideapad running Linux Mint 20 (Ulyana). Some comments and a couple of issues encountered.

(1) Minor issue. Based on [device list](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/tree/master/devices) I ran the script with
```
-p kmc-30153-smart-plug-mini
```
With this result:
```
Successfully built docker image
Loading options, please wait...
Traceback (most recent call last):
  File "/src/get_input.py", line 252, in <module>
    cli(obj={})
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/src/get_input.py", line 197, in write_profile
    if not (device and profile):
UnboundLocalError: local variable 'device' referenced before assignment
Failed to choose a profile, please run this script again
```
I reran the script dropping the -p option and going with:
```
> By firmware version and name
> 1.1.8 - BK7231N / oem_bk7231n_plug
 > Tuya Generic
 > LSPA7 Plug
```
The last option was picked based finding photos of that model on the net and they looked like my KMC plug.

This worked just fine.

(2) DNS issues.
The scripts takes over port 53 and name resolution. Building of the docker image fails because names can't be resolved. An example:
```
Checking UDP port 53... Occupied by systemd-resolve with PID 526.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image
[+] Building 0.1s (3/3) FINISHED                                                                                  
 => [internal] load build definition from Dockerfile                                                         0.0s
 => => transferring dockerfile: 457B                                                                         0.0s
 => [internal] load .dockerignore                                                                            0.0s
 => => transferring context: 2B                                                                              0.0s
 => ERROR [internal] load metadata for docker.io/library/python:3.9-slim-buster                              0.0s
------
 > [internal] load metadata for docker.io/library/python:3.9-slim-buster:
------
Dockerfile:1
--------------------
   1 | >>> FROM python:3.9-slim-buster AS base
   2 |     
   3 |     RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
--------------------
ERROR: failed to solve: python:3.9-slim-buster: failed to do request: Head "https://registry-1.docker.io/v2/library/python/manifests/3.9-slim-buster": dial tcp: lookup registry-1.docker.io: no such host
Failed to build Docker image, stopping script
```

This happened three times. My workaround for this was to add entries in my hosts file:
```
3.216.34.172    registry-1.docker.io
3.216.34.172    auth.docker.io
104.18.123.25  production.cloudflare.docker.com
```
Of course those may not be valid for the next run.

(3) I initially was going to flash the device over serial. So I had it opened, and wired for that. That didn't work because of a pullup resistor and a capacitor on the Tx pin. Somehow I came across this tool. I kept my harness and the tool ran on open, unplugged plug, just powering (GND/VCC wires only) the CB2S with my USB/UART adaptor. :-D

Cheers!
tjk :)

Cossid, 2023-02-24:
> (1) Minor issue. Based on [device list](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/tree/master/devices) I ran the script with
> 
> ```
> -p kmc-30153-smart-plug-mini
> ```
> 
> With this result:

This is kind of a side-effect/bug of switching to an API for pulling devices.  `-p` assumes you already have the profile locally, so if you've never run it before and haven't manually downloaded the files, it will fail.  Running with `-p` is usually just recommended when you have many of the same device to do repeatedly.  I may look into improving the fallback to check the API in the future, but that's a pretty low priority.

> I reran the script dropping the -p option and going with:
> 
> ```
> > By firmware version and name
> > 1.1.8 - BK7231N / oem_bk7231n_plug
>  > Tuya Generic
>  > LSPA7 Plug
> ```
> 
> The last option was picked based finding photos of that model on the net and they looked like my KMC plug.
> 
> This worked just fine.

Yeah, sounds like there are multiple firmwares for this plug.  I'll look into creating a KMC 1.1.8 profile, but for flashing, what you did works perfectly fine.

> (2) DNS issues.
> The scripts takes over port 53 and name resolution. Building of the docker image fails because names can't be resolved. An example:

> This happened three times. My workaround for this was to add entries in my hosts file:
> 
> ```
> 3.216.34.172    registry-1.docker.io
> 3.216.34.172    auth.docker.io
> 104.18.123.25  production.cloudflare.docker.com
> ```
> 
> Of course those may not be valid for the next run.

Yeah, this is still an issue for linux distros that use a local loop-back for all DNS queries.  The easier / more robust solution would be to add and external nameserver to your resolve.conf file such as `nameserver 1.1.1.1`

There isn't really an easy solution to the DNS issue, as CloudCutter absolutely needs to take over port 53, so those that use loop-back will always have a problem, and it's not really one CloudCutter should be solving.  I will add this to the FAQ page with a short description, there isn't much else for us to do.  Alternately, you can run CloudCutter once without killing port 53, then after device selection is done, re-run it and kill port 53.

closed 2023-02-24 by Cossid

Cossid, 2023-02-24:
A FAQ entry has been added at https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#tuya-cloudcutter-stops-my-dns-but-then-fails-to-download-docker-images-or-profile-data-how-can-this-be-resolved

Cossid, 2023-02-25:
The simpler solution is to just perform the safety checks closer to when they are needed (after docker image building and input API communications).  I have adjusted that moving forward.