tuya-cloudcutter/tuya-cloudcutter #295: Aubess not flashable

FritsOV, 2023-02-26:
for more dan one day i try to flash an aubess mini switch V1.2.1
i followed the steps on: [https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md](url)
even the docker install option in folder tuya-cloudcutter. All with succes. 
i followed the steps on: [https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md](url)

ran the script ./tuya-cloudcutter.sh -w wlan0 -s << myssd>> <<myssd pass>>
After the scriptcycle and getting a succes 
device is after the run disconected from tuya cloud but does not connect to my network. 

tried again without extra parameters ./tuya-cloudcutter.sh
followd the steps but again a succes but device is not up in my network. 

the scriptrun gives at some point an error on the wlan: can't connect to kernel. but gives overall succes with at the end an deviceid and local key. If i put the device in AP mode it shows up as A-728B but i cant connect to it in order to redirect it to the cloudcutter. 
The device is, as mentioned not accessible with a tuya based app but also does it not show up in my network. i'm out of clues. 

log:
Press any key to continue, or CTRL+C to exitBuilding cloudcutter docker image
[+] Building 1.7s (13/13) FINISHED
 => [internal] load .dockerignore                                                                                                                                  0.1s
 => => transferring context: 2B                                                                                                                                    0.0s
 => [internal] load build definition from Dockerfile                                                                                                               0.1s
 => => transferring dockerfile: 457B                                                                                                                               0.0s
 => [internal] load metadata for docker.io/library/python:3.9-slim-buster                                                                                          1.4s
 => [base 1/2] FROM docker.io/library/python:3.9-slim-buster@sha256:c79e572f1ef9e9f30343980472f68986fcaf4c41fee67a3ef7536f20ca1f65ed                               0.0s
 => [internal] load build context                                                                                                                                  0.1s
 => => transferring context: 3.02kB                                                                                                                                0.0s
 => CACHED [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosqui  0.0s
 => CACHED [python-deps 1/4] RUN pip install --upgrade pipenv                                                                                                      0.0s
 => CACHED [python-deps 2/4] COPY src/Pipfile /src/                                                                                                                0.0s
 => CACHED [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                                                           0.0s
 => CACHED [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                                                                       0.0s
 => CACHED [cloudcutter 1/2] COPY src /src                                                                                                                         0.0s
 => CACHED [cloudcutter 2/2] WORKDIR /src                                                                                                                          0.0s
 => exporting to image                                                                                                                                             0.0s
 => => exporting layers                                                                                                                                            0.0s
 => => writing image sha256:1605f19d16f11fcc9bb97e30077c9dba6cc19a4283e9fc0b1a5220fa4afea2cc                                                                       0.0s
 => => naming to docker.io/library/cloudcutter                                                                                                                     0.0s
Successfully built docker image
Loading options, please wait...
[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.2.1 - BK7231N / oem_bk7231n_control_switch
   1.1.71 - BK7231T / bk7231t_common_user_config_ty
   1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.8 - BK7231T / oem_bk7231s_rnd_switch
   1.1.80 - BK7231T / bk7231t_common_user_config_ty
   1.1.9 - BK7231T / bk7231s_common_iot_config_ty
   1.1.9 - BK7231T / oem_bk7231s_ceiling_light_ty
 > 1.2.1 - BK7231N / oem_bk7231n_control_switch
   1.2.16 - BK7231N / oem_bk7231n_light_ty
   1.2.2 - BK7231N / oem_bk7231n_ceiling_light_ty
   1.2.7 - BK7231N / oem_bk7231n_light_ty
   1.3.1 - BK7231N / oem_bk7231n_light_ty
   1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.3.2 - BK7231T / oem_bk7231s_dltj

[?] Select the brand of your device: Aubess
 > Aubess
   Tuya Generic

[?] Select the article number of your device: 16A Mini Smart Switch
 > 16A Mini Smart Switch

Selected Device Slug: aubess-16a-mini-smart-switch
Selected Profile: oem-bk7231n-control-switch-1.2.1-sdk-2.3.3-40.00

================================================================================
[... 38 lines cut ...]
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Cutting device off from cloud...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver             <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< HERE!
Using interface wlan0 with hwaddr b8:27:eb:f8:d6:bf and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.active
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.beacon.key.fetch
!!! Endpoint response not found, using default response - tuya.device.beacon.key.fetch (This is usually okay and safe to ignore unless something isn't working)
Processing endpoint tuya.device.uuid.pskkey.get
Device should be successfully onboarded on WiFi AP!  Please allow up to 2 minutes for the device to connect to your specified network.
Device MAC address: 1c:90:ff:5b:72:8b
Device Id: 6YewozfSajNEjet8CaNT
Local Key: 8rAHPFxtperGB5XE

Cossid, 2023-02-26:
> the scriptrun gives at some point an error on the wlan: can't connect to kernel.

This is just an informational message output by the program that creates an AP, it is not a hard error.  I have not seen a machine that does not display this message.

> The device is, as mentioned not accessible with a tuya based app but also does it not show up in my network. i'm out of clues.

Tuya-CloudCutter is successfully completing, as evidenced by the changing of the AP prefix as well as making it to the end without error.  If it is not joining your network, that is a network error of some sort beyond the control of Tuya-CloudCutter.  Some devices may need to be physically rebooted to join networks however.

There is nothing we can do any differently, we're just setting the ssid and password same as Tuya would be doing.  You'll need to investigate your network further to find potential pitfalls such as not accepting 2.4ghz on the ssid, special characters in ssid or password, or your network doesn't support the security modes of the device you are trying to add.

FritsOV, 2023-02-27:
Thanks for the reply, had to be sure that flashing procedure was well done. 
After playing around for a bit and physically pluged -> use switches, and unpluged the device for at least 10 times 
it now suddenly poped up in my network.  It takes some time for changes get any affect i guess. 
Great job!

closed 2023-02-27 by Cossid