tuya-cloudcutter/tuya-cloudcutter #298: Daybetter bulbs new firmware version V2.9.25

zain1144, 2023-03-01:
There is new firmware version of these bulb, This is a full dump of the chip

https://www.aliexpress.com/item/1005004518945276.html?spm=a2g0o.order_detail.order_detail_item.4.417bf19cYcgCl5

[readResult_BK7231T_QIO_2023-01-3--15-52-18.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10861795/readResult_BK7231T_QIO_2023-01-3--15-52-18.zip)

Cossid, 2023-03-01:
This has been added as `Daybetter RGBCT Bulb v2.9.25`, and all other Daybetter bulbs have been updated to include a firmware version since there are so many of them.

This device has a LED driver, unsure which, but most other Daybetter bulbs have been BP5758D.  Pin configuration can be found here: https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/daybetter-rgbct-bulb-v2.9.25.json

closed 2023-03-01 by Cossid

Cossid, 2023-03-01:
Verified this device is unpatched, Tuya-CloudCutter should work fine.