tuya-cloudcutter/tuya-cloudcutter #299: flash completed but device does not show up

FritsOV, 2023-03-01:
I successfully flashed 1 aubess smartswitch bk7213N, I have 6. It was by no means easy. In the meantime, I have been busy flashing the others for a few days, but after completing the routine. The device does not show up in  my the network. The script  has changed the firmware but afterwards the device does not appear in my network. I don't have any special characters in my wifi ssid or password and have put -s 'myssid' 'mypass' between the characters. I dont have ip or mac filtering in my router. 
After i flash i even tried to connect with the smart app to direct it to my network.  This step wont complete of course hens the tact that Its cut off from the tuya cloud with succes. If i take this step (reconnecting with the smartapp) the device is on my network and pingable but cant be used in homeassistant. 
Also, I tried to direct the AP to the cloudcutterflash using the smartapp but remains too short in the air to complete that. Is there a method to get it up longer and earlier? What am I doing wrong? Attached is the verbose output
[two runs.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10862438/two.runs.txt)

Cossid, 2023-03-02:
Are you cutting or flashing?  The logs provided are only for cutting, and they completed successfully.  If the device doesn't join your network, that is an issue you must figure out how to resolve outside of CloudCutter, as all we're doing is passing the ssid and password into the same fields Tuya uses to connect to a network.

> The script has changed the firmware but afterwards the device does not appear in my network.

Again, are you cutting or flashing?  If flashing, what are you flashing?  If it is ESPHome, are you changing the device name so there is no conflict?  If OpenBeken, a device doesn't join your network, it starts an AP instead.  The `-s` flag does not work with flashing, it is only for cutting.  If you are just cutting, you need to examine your network to find out why the device isn't joining.  There is no other way CloudCutter can pass credentials to join a network.

> After i flash i even tried to connect with the smart app to direct it to my network.

If you are flashing 3rd party firmware, I am not sure what smart app you are referring to, Smart Life would no longer be applicable.  If you are cutting, Smart Life will no longer work as the device never finishes communicating with Smart Life because it has been cut.  The only way to change network credentials (again, only if cut, not if flashed with 3rd party firmware) is by running CloudCutter again and providing different credentials.

> This step wont complete of course hens the tact that Its cut off from the tuya cloud with succes. If i take this step (reconnecting with the smartapp) the device is on my network and pingable but cant be used in homeassistant.

This is because it resets deviceids when you try to join with Smart Life, and it will fail.  Once cut, you should stop using the Smart Life app with that device.  Smart Life when it sets network details attempts to re-activate the device (which will fail), and because it fails, will continue trying to re-activate itself (which will keep failing).

> Also, I tried to direct the AP to the cloudcutterflash using the smartapp but remains too short in the air to complete that. Is there a method to get it up longer and earlier? What am I doing wrong? Attached is the verbose output

I'm not sure what you're even trying to do here.  You should not be doing anything with the Smart Life app at this point.  The CloudCutter script does the network join portion on its own independent of the Smart Life app.

You should not need to use the Smart Life app for any portion of cutting or flashing using Tuya-CloudCutter.

FritsOV, 2023-03-03:
Okay sorry for my bad English. i will try to explain.
i try to cut the mini switch off.

1. sudo ./tuya cloudcutter.sh -r -w wlan0 -s <mySSID> <mypass> -v
2. put mini switch in AP mode (slow blinking)
3. script ->  finds the AP  and setting up wlan0 -> Running initial exploit toolchain
4. device freezes and have to wait till its free to set it in AP mode again 
5. It finds device and configure device for connecting to cloudcutterflash ( i can see cloucutterflash become availeable)
6. script gives message:  Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
7. my question is, during this last step should the device blink? because at this point the device switch is functional. 

it ends with succes and the device should be visible or pingable in my network. but it is not. 

while it isn't working after several times trying and restart the above procedure 
i switched to my android and started the smartlive app.  just for testing purpose. I know its cut off so it cant fullfill its adding procedure anymore. 
1. add device  with loged in to my network wifi settings.
2. putting device in AP mode fast blinking
3. smartlife app connects to device 
4. device stops blinking
5. device is visible in router on local ip 192.168.1.81
6. device is pingable

as mentioned above step with smartlife app should not be done but is more a test cause i'm out of cluess here. 
if i restart tuay-cloudcutter.sh again i get the same results as the first try. device is not visible in router , not pingable. 

so what am i doing wrong or where can i get some pointers how to check if tuya-cloudcutter.sh is actually writing the wifi credentials and use these.

Cossid, 2023-03-03:
> 7\. my question is, during this last step should the device blink? because at this point the device switch is functional.

Not all devices are the same, but no, it should stop blinking once it connects to the cloudcutterflash AP.  Your log shows it goes through the complete motion and finishes successfully.  If the device won't reliably stay on your network, cutting may not be the most viable option (some devices just have poor connectivity code from Tuya).  You may be better off flashing a 3rd party firmware, especially for something as simple as a relay/switch.

> it ends with succes and the device should be visible or pingable in my network. but it is not.

It can't get the the success message without doing all the things that would allow it to join your network.  If it doesn't, then there is either a flaw in your network, or a flaw in the firmware on the device.  It only makes it to the success message if it received a good response to the network join command.

I don't think you're doing anything wrong, but either something is incompatible with your network, or the firmware on the device is flawed in an unusual way.  It wouldn't be the first time, there have been a couple other devices that have failed to join after being sent all data correctly.  My general recommendation at this point would to look into the possibility of a 3rd party firmware.  See https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md#custom-firmware-options for potential 3rd party firmware options.

I'm closing this issue, as there is nothing different Tuya-CloudCutter can do, as unfortunate as that might be.

closed 2023-03-03 by Cossid