tuya-cloudcutter/tuya-cloudcutter #304: The network-manager service needs to be restarted/started at the exit point

FireROR, 2023-03-07:
Hi,

I recently try to flash couple Global smart plug and the tool works fantastic. The only issue I had is once the script finished to flash the firmware, it just exit by itself. If I am trying to run the script again, the docker process will be failed as no dns is working. I have to run "systemctl start network-manager" to run script again.
I hope you can add one more function at the end, either start the service or loop back to see if user like to flash another device.

Cossid, 2023-03-07:
Honestly, the answer is probably not.  While systemctl is the correct command for you system, it's not for all systems.  Starting and stopping isn't the same on all platforms, and we would have to remember all commands issued to stop and re-manipulate them to start, and that is edging on a bit too much work for this type of tool.

Looping isn't really feasible with the way variables and states are currently saved.

This will likely end up getting closed as #wontfix unless someone really wants to create a PR for it.

FireROR, 2023-03-07:
I see.. if possible just post a line at the end will be helpful.

wenoptics, 2023-08-24:
For me, the script kill my systemd-resolved, which result in something like

```
ERROR: failed to solve: python:3.9.12-slim-buster: failed to do request: Head "https://registry-1.docker.io/v2/library/python/manifests/3.9.12-slim-buster": dial tcp: lookup registry-1.docker.io on 127.0.0.53:53: read udp 127.0.0.1:55736->127.0.0.53:53: read: connection refused
Failed to build Docker image, stopping script
```

after stopping the script, and, e.g. run the script again.

A `sudo systemctl start systemd-resolved` solve this.