tuya-cloudcutter/tuya-cloudcutter #305: no such connection profile

12christiaan, 2023-03-07:
Hi all, thanks for the great software.
i did use it to flash OpenBK7231 yesterday into devices. 
Today i start working on a different device. and strugeling to get it working again.
I do not think that for this device there is a device profile yet, but that should not matter, there i am flashing OpenBK7231T into it.

Product Name		Hombli Smart Socket
Model:			HBSS-0209 Smart socket   B2030248
Device update:	Main V 1.0.2
MAC Address		10.5a.17.7e.xx.xx
Chip			WB2S = BK7231T

I try many different options, but there i have no clue what is needed i am stuck.
with and without -f OpenBK7231T_UG_1.15.524.bin in command line
tried with selection by manufacture (tuya generic)
tried with selection by firemware (   1.0.2 - BK7231T / bk7231t_common_user_config_ty &    1.0.2 - BK7231T / cuco_sw5_bk_common &    1.0.2 - BK7231T / oem_bk7231s_control_switch)

every time the script will find the access point name: "SmartLife-3BA1", trying to connect...
and than the error that it could not connect.
while i can connect to that AP with my phone.

why is the error "no such connection profile". the script will make that profile?

`================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP

Found access point name: "SmartLife-3BA1", trying to connect...
Error: No network with SSID 'SmartLife-3BA1' found.
Error: SmartLife-3BA1 - no such connection profile.
.
Found access point name: "SmartLife-3BA1", trying to connect...
Error: No network with SSID 'SmartLife-3BA1' found.
Error: SmartLife-3BA1 - no such connection profile.
.
Found access point name: "SmartLife-3BA1", trying to connect...
Error: No network with SSID 'SmartLife-3BA1' found.
Error: SmartLife-3BA1 - no such connection profile.
.
Found access point name: "SmartLife-3BA1", trying to connect...
Error: No network with SSID 'SmartLife-3BA1' found.
Error: SmartLife-3BA1 - no such connection profile.
.
Found access point name: "SmartLife-3BA1", trying to connect...
Error: No network with SSID 'SmartLife-3BA1' found.
Error: SmartLife-3BA1 - no such connection profile.
Error, could not connect to SSID.
Failed to connect, please run this script again`

Cossid, 2023-03-07:
Are you still running elevated?  You may need to add the `-r` flag to reset NetworkManager.  The error is NetworkManager not being able to create or manipulate the profile it is creating to connect to the wireless network.

12christiaan, 2023-03-07:
Thanks. that helped.

closed 2023-03-08 by Cossid