tuya-cloudcutter/tuya-cloudcutter #312: Unmarked CB2S monitoring smart plug - looks like a LSPA9

Henri-J-Norden, 2023-03-10:
[2023-03-09_23-57-04_lightleak.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10937346/2023-03-09_23-57-04_lightleak.zip)

|  [image]  | [image]   |  [image]  |  [image]  |
|---|---|---|---|

- _I originally tried editing the `tuya-generic-lspa9-smart-plug.json` device to use the `oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00.json` profile, but that didn't work for flashing_
- Dumped using lightleak BK7231N Type 1 / Addr 1 (XOR)
- Flashing tested working now, though the profile is identical to the existing `oem-bk7231n-plug-1.1.8-sdk-2.3.1-40.00.json`

I couldn't get the damn thing open, so I found this teardown video, but it looks it's glued so tight there's no chance (non-destructively): https://www.youtube.com/watch?v=mG4bAAHluMU
It also seems like the same device as https://zry.io/archives/799 which is an Elivco LSPA9 (potentially related issue: https://github.com/tuya-cloudcutter/lightleak/issues/1). 

I edited their ESPHome config back from the ESP-12F they used to the CB2S pinout without much thought and everything (except for current, so the relay and monitoring power, voltage, energy) worked immediately after the first flash with cloudcutter! I don't know where the best place would be to post this, but I'll leave my config here for now just in case:

```
libretuya:
  board: cb2s
  framework:
    version: dev

status_led:  # use the on-board blue LED as status indicator (as it was originally)
  pin:
    number: P8
    inverted: true  # due to it's connected in sink logic

switch:  # the socket relay
  - platform: gpio
    pin: P26
    id: plug_1
    name: 'Plug 1'
    restore_mode: RESTORE_DEFAULT_OFF  # attempt to restore state and default to OFF if failed

binary_sensor:  # the button
  - platform: gpio
    pin:
      number: RX1
      inverted: true
    id: button
    internal: true  # don't expose it to HASS, only use it locally
    on_release:
      then:
        - switch.toggle: plug_1  # toggle the relay

sensor:
  - platform: hlw8012
    model: BL0937  # note that the model must be specified to use special calculation parameters
    voltage_divider: 1600  # adjust it according to the actual resistor values on board
    sel_pin:
      number: P24
      inverted: true  # the logic of BL0937 is opposite from HLW8012
    cf_pin: P7
    cf1_pin: P6
    current:
      name: 'Plug 1 Current'
    voltage:
      name: 'Plug 1 Voltage'
    power:
      name: 'Plug 1 Power'
    energy:
      name: 'Plug 1 Energy'
      # convert it to kWh
      filters:
        - multiply: 0.001
      unit_of_measurement: 'kWh'
      accuracy_decimals: 4
    update_interval: 60s
    change_mode_every: 3
```

Cossid, 2023-03-10:
Despite looking nothing like it, this is a direct match with the `Baytion LSPA7 Smart Plug` (exact schema and pin match) and also a schema compatible match with the `Tuay Generic LSPA7 Plug`

closed 2023-03-10 by Cossid

Henri-J-Norden, 2023-03-10:
Inverting the other pins worked, I still had to tune the voltage_divider value to match my multimeter readings (and just assuming for now that current is close enough with the 1 mOhm resistor setting):
```
sensor:
  - platform: hlw8012
    model: BL0937  # note that the model must be specified to use special calculation parameters
    voltage_divider: 755  # adjust it according to the actual resistor values on board
    # @239V: 1600->500V 2000->630V 900->283V 800->252V
    sel_pin:
      number: P24
      inverted: true  # the logic of BL0937 is opposite from HLW8012
    cf_pin: 
      number: P7
      inverted: true
    cf1_pin: 
      number: P6
      inverted: true
    current:
      name: '$name Current'
      accuracy_decimals: 3
      filters: 
        - median: 
            send_every: 5
            window_size: 5
    voltage:
      name: '$name Voltage'
      filters: 
        - median: 
            send_every: 5
            window_size: 5
    power:
      name: '$name Power'
      filters: 
        - median: 
            send_every: 5
            window_size: 5
    energy:
      name: '$name Energy'
      filters: 
        - median: 
            send_every: 5
            window_size: 5
    update_interval: 0.5s
    change_mode_every: 6
```

Cossid, 2023-03-10:
After looking, it appears we had other people reporting they also had LSPA9s with BK7231N chips, but we hadn't had a dump.  I have renamed the prior BK7231T to include it's version (1.0.3) and added this one as a Tuya Generic LSPA9 Plug v1.1.8

So yes, it appears this model can have either WB2S and CB2S modules inside.  Users will have to match their firmware version to get the correct one.

zethis, 2023-05-11:
Hey i get pretty much the same issue, white branded 16A plug, i opened it and it's a CB2S inside, but i'm not able to successfully flash it, first step of flash successfully passed (with1.1.8 - BK7231N / oem_bk7231n_plug (Exploit run, saved device config too!)), but at the end that failed ([!] The profile you selected did not result in a successful exploit.).
Name: Smart Plug +
Plug principal firmware version: V1.1.15
Other type: V1.0.5
Send you the saved device config:
{"uuid": "170cQEXuRgS7", "auth_key": "C9rE5vrqfMcpLj5C", "local_key": "s6ZVMS1zHNoEQsZr", "sec_key": "Er5lfK72E0fKtMo2", "device_id": "HmN8memb81LlxAnMjzVh", "chip_family": "BK7231N", "profile_name": "1.1.8 - BK7231N / oem_bk7231n_plug", "device_slug": "tuya-generic-lspa9-plug-v1.1.8"}

Thanks for help :)

kuba2k2, 2023-05-11:
If your plug has v1.1.15, then you need to use a profile for v1.1.15. Profiles won't work for non-matching versions.

Since there's no profile for v1.1.15 right now, Cloudcutter won't work. And since you've opened it already, you can try [dumping the firmware](https://docs.libretiny.eu/docs/flashing/dumping/) using UART.

zethis, 2023-05-11:
Thanks for answering :) I take out the soldering iron this week end :)

kuba2k2, 2023-05-11:
@zethis see https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/386#issuecomment-1544169060 - 1.1.15 apparently is unexploitable.

Cossid, 2023-05-11:
Correct, 1.1.15 for plugs is patched and serial is your only option.
To expand awareness, I have added a Wiki page that will track known patched firmwares: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/Known-Patched-Firmware