tuya-cloudcutter/tuya-cloudcutter #313: After first try, the second try isnt working due to APname mismatch

euro2, 2023-03-10:
I tried to flash OpenBeken but halfway down the script it couldnt find the firmware file due to me not putting it in the correct dir.

Now trying again it fails once more because now it has an AP called A-6349 and not Tuya Smartlife AP.....

Any workaround?

Cossid, 2023-03-10:
It shouldn't fail to find an A- prefixed AP, it is designed to pick that up just the same.  What is the actual log output?  Simply restarting the process should have worked fine.  You may need to use the `-r` parameter.

euro2, 2023-03-10:
Sorry, but i cannot reproduce because i went oldschool and went UART. 

Thank you for your reply,. Good to know. Ill link this info to a forum page i made about dismantling:
https://www.elektroda.com/rtvforum/topic3963536.html

closed 2023-03-10 by Cossid