tuya-cloudcutter/tuya-cloudcutter #316: Support for lenovo-se-242dc-rgbct-bulb-v1.2.21

hackoder, 2023-03-12:
Hello,
I'm trying to use cloudcutter to detach the mentioned Lenovo Smart Bulb, but it never connects to the provided SSID. Logs and some comments:

```
$ sudo ./tuya-cloudcutter.sh -v -r -s 'ASUS_[...]' '[..]' -p lenovo-se-242dc-rgbct-bulb-v1.2.21

Building cloudcutter docker image
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  22.59MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 5896ec7c9c41
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 552dcf7a0f29
Step 3/10 : FROM base AS python-deps
 ---> 552dcf7a0f29
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 0d1984aa30ca
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> ffd48689cec0
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 53c383ad688d
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> ece7a2d99d96
Step 8/10 : FROM python-deps AS cloudcutter
 ---> ece7a2d99d96
Step 9/10 : COPY src /src
 ---> Using cache
 ---> cae58465b692
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 9c69a28e88d2
Successfully built 9c69a28e88d2
Successfully tagged cloudcutter:latest
Successfully built docker image
Loading options, please wait...
Selected Device Slug: lenovo-se-242dc-rgbct-bulb-v1.2.21
Selected Profile: oem-bk7231n-light-ty-1.2.21-sdk-2.3.1-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

```
At this point, I switch to AP mode (6 off/on cycles):

```
Wiping NetworkManager configs
Scanning for open Tuya SmartLife AP
....
Found access point name: "A-A513", trying to connect...
Device 'wlan0' successfully activated with 'deb2018e-e096-4373-a287-0cf40141689a'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/jYvyjzJjMyag.deviceconfig
Saved device config in /work/configured-devices/jYvyjzJjMyag.deviceconfig

================================================================================
[... 191 lines cut ...]
Mar 12 17:02:13 dnsmasq-dhcp[15]: 775596154 sent size:  4 option: 51 lease-time  12h
Mar 12 17:02:13 dnsmasq-dhcp[15]: 775596154 sent size:  4 option: 58 T1  6h
Mar 12 17:02:13 dnsmasq-dhcp[15]: 775596154 sent size:  4 option: 59 T2  10h30m
Mar 12 17:02:13 dnsmasq-dhcp[15]: 775596154 sent size:  4 option:  1 netmask  255.255.255.0
Mar 12 17:02:13 dnsmasq-dhcp[15]: 775596154 sent size:  4 option: 28 broadcast  10.42.42.255
Mar 12 17:02:13 dnsmasq-dhcp[15]: 775596154 sent size:  4 option:  3 router  10.42.42.1
Mar 12 17:02:13 dnsmasq-dhcp[15]: 775596154 sent size:  4 option:  6 dns-server  10.42.42.1
Mar 12 17:03:06 dnsmasq[15]: query[A] a3.tuyaus.com from 10.42.42.29
Mar 12 17:03:06 dnsmasq[15]: config a3.tuyaus.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
Mar 12 17:03:06 dnsmasq[15]: query[A] m2.tuyaus.com from 10.42.42.29
Mar 12 17:03:06 dnsmasq[15]: config m2.tuyaus.com is 10.42.42.1

[17:03:07.016917 Log (Client)] Request: HTTPServerRequest(protocol='https', host='a3.tuyaus.com', method='POST', uri='/d.json?a=tuya.device.uuid.pskkey.get&et=1&t=57&uuid=jYvyjzJjMyag&v=1.0&sign=a4535f50112120eb66489cfcf7f4097a', version='HTTP/1.1', remote_ip='10.42.42.29')
[17:03:07.016953 LOG (Client)] ==== Request body ===
{"t":57}
[17:03:07.016960 LOG (Client)] ==== End request body ===
[17:03:07.017187 LOG (Server)] Response:  {'result': {'pskKey': ''}, 't': 1678640587, 'success': True}
Device should be successfully onboarded on WiFi AP!  Please allow up to 2 minutes for the device to connect to your specified network.
Device MAC address: 
Device Id: qo[...]
Local Key: 2A[...]
```

I check my SSID, and do not see an IP getting assigned for this device. No incoming DHCP requests either. Anything else I can try to determine what is going wrong? Given the requests, it seems like it does connect to the cloudcutterflash AP, but doesn't connect to the configured SSID (tried two different ones).

Cossid, 2023-03-12:
What is the proximity of your cloudcutter device to your bulb?  It's clearly connecting as evidenced by the fact it does make a dhcp request, but then it makes a less than normal DNS request.  I'm wondering if it isn't dropping some packets.  Instead of looking up a dns server, it appears to be looking up the mqtt server, then eventually the API server.  And once it does the API server, it's not making the usual activate request, but instead API endpoints that should come later.

This device is already cut, so you can't reconnect it to cloud services, but are you aware if this worked with the Tuya Smart Life app, or did you have to use the Lenovo app (some 3rd party branded stuff restricts this).

If you wouldn't mind and have it available, could you send the full firmware to me privately at [ cossid -at - cossid.com ]?

hackoder, 2023-03-12:
Thank you for the response. 

> What is the proximity of your cloudcutter device to your bulb? 

The cloudcutter device (laptop) is right next to the bulb (makes it easy to toggle the light switch). 

> are you aware if this worked with the Tuya Smart Life app

Yup, I used it initially with the Tuya app, it was able to connect correctly. Used tinytuya to pull the id/key, and then placed it on an isolated subnet and used homeassistant/localtuya to control it for a bit as well. The device profile was added a couple of days ago (https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/commit/5a2fb3c75a11dc32d5eb8c2c80eee748d961a882) so I attempted to cut it then.

Unfortunately I don't have the firmware available, but maybe @kr1os from the PR would have it available. I will also attempt to use lightleak to see if I can dump the firmware.

Cossid, 2023-03-12:
Did you try using the device with the device and local key displayed after the last attempt to cut?  Your device may not be attempting to activate, but if it's already configured, that data may be valid.  The actual cut in cloudcutter actually happens when the device changes it's AP prefix to `A-` and since it didn't call `tuya.device.active` it should still have the schema that was present prior to cutting.  Doesn't explain why it's not trying to activate, but I'd be curious what is and isn't working.

hackoder, 2023-03-12:
How would I attempt to use it without an active wifi connection? I guess I can set up a dummy cloudcutterflash AP (since its configured to use it now?) and see where I get.

Cossid, 2023-03-12:
The time you got

```
[17:03:07.016917 Log (Client)] Request: HTTPServerRequest(protocol='https', host='a3.tuyaus.com', method='POST', uri='/d.json?a=tuya.device.uuid.pskkey.get&et=1&t=57&uuid=jYvyjzJjMyag&v=1.0&sign=a4535f50112120eb66489cfcf7f4097a', version='HTTP/1.1', remote_ip='10.42.42.29')
[17:03:07.016953 LOG (Client)] ==== Request body ===
{"t":57}
[17:03:07.016960 LOG (Client)] ==== End request body ===
[17:03:07.017187 LOG (Server)] Response:  {'result': {'pskKey': ''}, 't': 1678640587, 'success': True}
Device should be successfully onboarded on WiFi AP!  Please allow up to 2 minutes for the device to connect to your specified network.
Device MAC address: 
Device Id: qo[...]
Local Key: 2A[...]
```
Should have put it back on your local network.  Alternately, CloudCutter is actually using tinytuya to change the wifi credentials as well, so you could create the AP temporarily and use tinytuya to change them again as well.

hackoder, 2023-03-12:
So, I'm actually not seeing it on my local network _at all_. No DHCP requests at power on. I'm looking at the System log for my Asus router, which shows other (uncut) bulbs getting IPs, but not this one. Thanks for the tip on using tinytuya, I'll play around with that then.

Cossid, 2023-03-12:
For reference, here is how we use tinytuya to change network credentials:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/96bb16dd4ac10448dcfe5e2fdf54f279c3a4995c/src/cloudcutter/__main__.py#L39

hackoder, 2023-03-12:
So, apparently this line:

https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/96bb16dd4ac10448dcfe5e2fdf54f279c3a4995c/src/cloudcutter/__main__.py#L51

is returning `{'Error': 'Invalid JSON Response from Device', 'Err': '900', 'Payload': ''}`, so the while loop below is not getting engaged at all:

```
[23:36:46.810342 Log (Client)] Request: HTTPServerRequest(protocol='https', host='a3.tuyaus.com', method='POST', uri='/d.json?a=tuya.device.uuid.pskkey.get&et=1&t=56&uuid=KMucyoN0tEst&v=1.0&sign=2224f995735dfaec485406b946f7ab5b', version='HTTP/1.1', remote_ip='10.42.42.29')
[23:36:46.810378 LOG (Client)] ==== Request body ===
{"t":56}
[23:36:46.810390 LOG (Client)] ==== End request body ===
[23:36:46.810665 LOG (Server)] Response:  {'result': {'pskKey': ''}, 't': 1678664206, 'success': True}
INFO:tornado.access:200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=56&uuid=KMucyoN0tEst&v=1.0&sign=2224f995735dfaec485406b946f7ab5b (10.42.42.29) 77.94ms
DEBUG:tinytuya.tinytuya:updatedps() entry (dev_type is default)
DEBUG:tinytuya.tinytuya:building payload=b'{"dpId":[1]}'
DEBUG:tinytuya.tinytuya:payload generated=b'000055aa0000000000000012000000189cb525b63b7e94b720c6e7a834064251c67acbb30000aa55'
DEBUG:tinytuya.tinytuya:Timeout or exception in _send_receive() - retry 1/5
DEBUG:tinytuya.tinytuya:received data=b'000055aa00000000000000120000002c000000012c0e3d4cfdaedd9765195da6c6967f7702da05bbd0b437016f6ca9e5be3cec54a193bd4e0000aa55'
DEBUG:tinytuya.tinytuya:raw unpacked message = TuyaMessage(seqno=0, cmd=18, retcode=1, payload=b',\x0e=L\xfd\xae\xdd\x97e\x19]\xa6\xc6\x96\x7fw\x02\xda\x05\xbb\xd0\xb47\x01ol\xa9\xe5\xbe<\xecT', crc=2710814030)
DEBUG:tinytuya.tinytuya:decode payload=b',\x0e=L\xfd\xae\xdd\x97e\x19]\xa6\xc6\x96\x7fw\x02\xda\x05\xbb\xd0\xb47\x01ol\xa9\xe5\xbe<\xecT'
DEBUG:tinytuya.tinytuya:decrypting=b',\x0e=L\xfd\xae\xdd\x97e\x19]\xa6\xc6\x96\x7fw\x02\xda\x05\xbb\xd0\xb47\x01ol\xa9\xe5\xbe<\xecT'
DEBUG:tinytuya.tinytuya:decrypted 3.3 payload=''
DEBUG:tinytuya.tinytuya:payload type = <class 'str'>
DEBUG:tinytuya.tinytuya:decoded results=''
DEBUG:tinytuya.tinytuya:ERROR Invalid JSON Response from Device - 900 - payload: ""
DEBUG:tinytuya.tinytuya:updatedps received data={'Error': 'Invalid JSON Response from Device', 'Err': '900', 'Payload': ''}
{'Error': 'Invalid JSON Response from Device', 'Err': '900', 'Payload': ''}
Device should be successfully onboarded on WiFi AP!  Please allow up to 2 minutes for the device to connect to your specified network.
Device MAC address: 
```

If I manually set `parsed_data = {}` before the while loop, then it returns:

```
Mar 12 23:19:37 dnsmasq[15]: config a3.tuyaus.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying is false - bailing out
[!] Unable to decrypt device reponse.  PSKKEY/AUTHKEY do not match device.
Oh no, something went wrong with detaching from the cloud! Try again I guess...
```

Cossid, 2023-03-13:
That means it's using the wrong device id/local key.  It changes every time the device is exploited.  The most recent file in `/configured-devices/` should have the correct credentials.  If not, something else entirely is going on.

Cossid, 2023-03-21:
Closing as stale.  Not sure there is anything CloudCutter can do either way, this may be a bug in Tuya's firmware.  Feel free to respond/re-open if you find anything concrete.

closed 2023-03-21 by Cossid

fleminra, 2023-04-03:
Same situation here with same bulb. The bulb is connected to the "cloudcutterflash" AP, as created by `tuya-cloudcutter.sh`.

I modified `src/tinytuya/examples/getstatus.py` as follows:
```python
import tinytuya
import time

import logging

logging.basicConfig(level=logging.DEBUG)

DEVICEID='AhkrCso4SCz1WLsvPmFu'
DEVICEIP='10.42.42.30'
DEVICEKEY='hBCwiUBMxBbfcGvd'

d=tinytuya.BulbDevice(DEVICEID, DEVICEIP, DEVICEKEY)
d.set_version(3.3)

# Alternative connection - for some devices with 22 character IDs they require a special handling                                
#    d=tinytuya.OutletDevice(DEVICEID, DEVICEIP, DEVICEKEY, 'device22')                                                          
#    d.set_dpsUsed({"1": None})                                                                                                  
#    d.set_version(3.3)                                                                                                          

# Option for Power Monitoring Smart Plugs - Some require UPDATEDPS to update power data points                                   
# payload = d.generate_payload(tinytuya.UPDATEDPS,['18','19','20'])                                                              
# d.send(payload)                                                                                                                
# sleep(1)                                                                                                                       

# Get the status of the device                                                                                                   
# e.g. {'devId': '0071299988f9376255b', 'dps': {'1': True, '3': 208, '101': False}}                                              
data = d.status()
print(data)
```
Here's its output:
```
DEBUG:tinytuya:status() entry (dev_type is default)
DEBUG:tinytuya:building payload=b'{"gwId":"AhkrCso4SCz1WLsvPmFu","devId":"AhkrCso4SCz1WLsvPmFu","uid":"AhkrCso4SCz1WLsvPmFu","t":"1680504371"}'
DEBUG:tinytuya:payload generated=b'000055aa000000000000000a00000078e5baff8fa0b8f7e57506bcb12dc420dd865fe85d5ebc1d473abd3fc69a6caef480eeccf2b475ce922cbdf2e426021081169ad23c730aa2abc42538cceae0b360440e01bc1d27f3d7e4b2719b22244eea2917322155e2a7baca1a2c19540de57cd8a3669dbd455af33eb2bd71069fce440c4c89690000aa55'
DEBUG:tinytuya:received data=b'000055aa000000000000000a0000002c00000001bc18c1d54acded3ee9ff36299dee2e83bc33a911d61484509083ad07df91333704d410f00000aa55'
DEBUG:tinytuya:raw unpacked message = TuyaMessage(seqno=0, cmd=10, retcode=1, payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137', crc=81006832)
DEBUG:tinytuya:decode payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137'
DEBUG:tinytuya:decrypting=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137'
DEBUG:tinytuya:decrypted 3.3 payload=b"\xef=\xc6G\xc6\xe5\x18\xe9j'\x8dD;N\xda\x96\xa2\xb21O\x0em"
DEBUG:tinytuya:payload type = <class 'bytes'>
DEBUG:tinytuya:payload was not string type and decoding failed
DEBUG:tinytuya:ERROR Invalid JSON Response from Device - 900 - payload: ""
DEBUG:tinytuya:status() received data={'Error': 'Invalid JSON Response from Device', 'Err': '900', 'Payload': ''}
DEBUG:tinytuya:bulb type set to B
DEBUG:tinytuya:status() entry (dev_type is default)
DEBUG:tinytuya:building payload=b'{"gwId":"AhkrCso4SCz1WLsvPmFu","devId":"AhkrCso4SCz1WLsvPmFu","uid":"AhkrCso4SCz1WLsvPmFu","t":"1680504371"}'
DEBUG:tinytuya:payload generated=b'000055aa000000010000000a00000078e5baff8fa0b8f7e57506bcb12dc420dd865fe85d5ebc1d473abd3fc69a6caef480eeccf2b475ce922cbdf2e426021081169ad23c730aa2abc42538cceae0b360440e01bc1d27f3d7e4b2719b22244eea2917322155e2a7baca1a2c19540de57cd8a3669dbd455af33eb2bd71069fce44e0be75a10000aa55'
DEBUG:tinytuya:received data=b'000055aa000000010000000a0000002c00000001bc18c1d54acded3ee9ff36299dee2e83bc33a911d61484509083ad07df913337a74238bc0000aa55'
DEBUG:tinytuya:raw unpacked message = TuyaMessage(seqno=1, cmd=10, retcode=1, payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137', crc=2806134972)
DEBUG:tinytuya:decode payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137'
DEBUG:tinytuya:decrypting=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137'
DEBUG:tinytuya:decrypted 3.3 payload=b"\xef=\xc6G\xc6\xe5\x18\xe9j'\x8dD;N\xda\x96\xa2\xb21O\x0em"
DEBUG:tinytuya:payload type = <class 'bytes'>
DEBUG:tinytuya:payload was not string type and decoding failed
DEBUG:tinytuya:ERROR Invalid JSON Response from Device - 900 - payload: ""
DEBUG:tinytuya:status() received data={'Error': 'Invalid JSON Response from Device', 'Err': '900', 'Payload': ''}
{'Error': 'Invalid JSON Response from Device', 'Err': '900', 'Payload': ''}
```

Thoughts? It seems to be using the right DEVICEID and DEVICEKEY because I don't get even this far with other values.

Cossid, 2023-04-03:
Try setting version to 3.4 instead of 3.3?

fleminra, 2023-04-03:
With 3.4:
```
DEBUG:tinytuya:status() entry (dev_type is default)
DEBUG:tinytuya:building payload=b'{"gwId":"E2nkjrVpbhEyrmmvJzRJ","devId":"E2nkjrVpbhEyrmmvJzRJ","uid":"E2nkjrVpbhEyrmmvJzRJ","t":"1680535539"}'
DEBUG:tinytuya:payload generated=b'000055aa000000000000000a000000747b2267774964223a2245326e6b6a72567062684579726d6d764a7a524a222c226465764964223a2245326e6b6a72567062684579726d6d764a7a524a222c22756964223a2245326e6b6a72567062684579726d6d764a7a524a222c2274223a2231363830353335353339227dfac9479c0000aa55'
DEBUG:tinytuya:received data=b'000055aa000000000000000a0000002c00000001bc18c1d54acded3ee9ff36299dee2e83bc33a911d61484509083ad07df91333704d410f00000aa55'
DEBUG:tinytuya:raw unpacked message = TuyaMessage(seqno=0, cmd=10, retcode=1, payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137', crc=81006832)
DEBUG:tinytuya:decode payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137'
DEBUG:tinytuya:Unexpected payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137'
DEBUG:tinytuya:ERROR Unexpected Payload from Device - 904 - payload: ""
DEBUG:tinytuya:status() received data={'Error': 'Unexpected Payload from Device', 'Err': '904', 'Payload': ''}
DEBUG:tinytuya:bulb type set to B
DEBUG:tinytuya:status() entry (dev_type is default)
DEBUG:tinytuya:building payload=b'{"gwId":"E2nkjrVpbhEyrmmvJzRJ","devId":"E2nkjrVpbhEyrmmvJzRJ","uid":"E2nkjrVpbhEyrmmvJzRJ","t":"1680535539"}'
DEBUG:tinytuya:payload generated=b'000055aa000000010000000a000000747b2267774964223a2245326e6b6a72567062684579726d6d764a7a524a222c226465764964223a2245326e6b6a72567062684579726d6d764a7a524a222c22756964223a2245326e6b6a72567062684579726d6d764a7a524a222c2274223a2231363830353335353339227d5419b5b00000aa55'
DEBUG:tinytuya:received data=b'000055aa000000010000000a0000002c00000001bc18c1d54acded3ee9ff36299dee2e83bc33a911d61484509083ad07df913337a74238bc0000aa55'
DEBUG:tinytuya:raw unpacked message = TuyaMessage(seqno=1, cmd=10, retcode=1, payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137', crc=2806134972)
DEBUG:tinytuya:decode payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137'
DEBUG:tinytuya:Unexpected payload=b'\xbc\x18\xc1\xd5J\xcd\xed>\xe9\xff6)\x9d\xee.\x83\xbc3\xa9\x11\xd6\x14\x84P\x90\x83\xad\x07\xdf\x9137'
DEBUG:tinytuya:ERROR Unexpected Payload from Device - 904 - payload: ""
DEBUG:tinytuya:status() received data={'Error': 'Unexpected Payload from Device', 'Err': '904', 'Payload': ''}
{'Error': 'Unexpected Payload from Device', 'Err': '904', 'Payload': ''}
```

I see that with 3.3, it at least has success decrypting the message. With 3.4, there's no mention of even trying to decrypt it.

Could the message be in, e.g. MessagePack or Protocol Buffers instead of JSON?

Cossid, 2023-04-03:
Nevermind, I though I read 1.2.21 had SDK 2.3.3, but it is 2.3.1, which is still 3.3.

Any reason your devId changed?

I managed to get a hold of a dump for this device so I'll give it a try when I get some free time and see if I can see anything strange with it.

fleminra, 2023-04-03:
> Any reason your devId changed?

Yeah I just turned the bulb off overnight and had to start over in the morning.

> I managed to get a hold of a dump for this device so I'll give it a try when I get some free time and see if I can see anything strange with it.

Awesome, yeah there was a deal on these bulbs a couple months so I got a few, knowing it'd be a gamble. Thanks for your help on it.

Cossid, 2023-04-03:
As usual, when it comes to Tuya not acting as expected, we generally recommend looking into [3rd part firmware alternatives](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md#custom-firmware-options) instead, as they are much more predictable/controllable, and usually give a better experience.

Cossid, 2023-04-03:
I'm not sure what to tell you, I just tried it locally, and everything went pretty much as expected.

Cut log:
```
Selected Device Slug: lenovo-se-242dc-rgbct-bulb-v1.2.21
Selected Profile: oem-bk7231n-light-ty-1.2.21-sdk-2.3.1-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.......
Found access point name: "SmartLife-73AB", trying to connect...
Device 'wlan0' successfully activated with 'a388c849-3bb6-4b15-a6bc-4e6ac7efc6a5'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/bTUDC2LdxTmb.deviceconfig
Saved device config in /work/configured-devices/bTUDC2LdxTmb.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "SmartLife-73AB", trying to connect...
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Expected AP gateway = 192.168.175.1 or 192.168.176.1 but got
..
Found access point name: "A-73AB", trying to connect...
Device 'wlan0' successfully activated with '9e5c7d3d-d3ba-4b09-8571-a6ff295e6f02'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Cutting device off from cloud...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:93:00:86 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.active
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.uuid.pskkey.get
Device should be successfully onboarded on WiFi AP!  Please allow up to 2 minutes for the device to connect to your specified network.
Device MAC address: 84:e3:42:f4:73:ab
Device Id: 5TNvqg5PndYAc2OiKJhO
Local Key: wx7ZoW2Pb0riJld4
[... 46 lines cut ...]
DEBUG:tinytuya.core:raw unpacked message = TuyaMessage(seqno=1, cmd=10, retcode=0, payload=b'w\xe1\xd0\xa9\xaa6\xbbM[\x13f\x99\xce\xf6$\xfb\x80*?\xb6\xe5\x99\n\x11lh\xab!)\x07\xf62\x1a\x1b\xc3\x15\xee\x8e\xd7\xcd\x10hR\xc4\xa9\t\x1f\x93\x02\xbeR5\x03\xeb\x9b$\x9bw\xda\xd2\x11P%MP\xd6.\x93\x15\xb8\xac\xe4r{\xdb%]~1\xa6\xe6oz\x880\x8f\x97\xef\x8d\'"\x88\xc0\xc2\xa5\xb5\xa4\xb3\xb3\xbfT\xe8eo\xee\xcd\x83]a\xe5\xdb\xa6\x13{\x16\xd8>b\x13c=d\xefi\x87\xda^T', crc=4012970555, crc_good=True, prefix=21930, iv=None)
DEBUG:tinytuya.core:decode payload=b'w\xe1\xd0\xa9\xaa6\xbbM[\x13f\x99\xce\xf6$\xfb\x80*?\xb6\xe5\x99\n\x11lh\xab!)\x07\xf62\x1a\x1b\xc3\x15\xee\x8e\xd7\xcd\x10hR\xc4\xa9\t\x1f\x93\x02\xbeR5\x03\xeb\x9b$\x9bw\xda\xd2\x11P%MP\xd6.\x93\x15\xb8\xac\xe4r{\xdb%]~1\xa6\xe6oz\x880\x8f\x97\xef\x8d\'"\x88\xc0\xc2\xa5\xb5\xa4\xb3\xb3\xbfT\xe8eo\xee\xcd\x83]a\xe5\xdb\xa6\x13{\x16\xd8>b\x13c=d\xefi\x87\xda^T'
DEBUG:tinytuya.core:decrypting=b'w\xe1\xd0\xa9\xaa6\xbbM[\x13f\x99\xce\xf6$\xfb\x80*?\xb6\xe5\x99\n\x11lh\xab!)\x07\xf62\x1a\x1b\xc3\x15\xee\x8e\xd7\xcd\x10hR\xc4\xa9\t\x1f\x93\x02\xbeR5\x03\xeb\x9b$\x9bw\xda\xd2\x11P%MP\xd6.\x93\x15\xb8\xac\xe4r{\xdb%]~1\xa6\xe6oz\x880\x8f\x97\xef\x8d\'"\x88\xc0\xc2\xa5\xb5\xa4\xb3\xb3\xbfT\xe8eo\xee\xcd\x83]a\xe5\xdb\xa6\x13{\x16\xd8>b\x13c=d\xefi\x87\xda^T'
DEBUG:tinytuya.core:decrypted 3.x payload='{"dps":{"20":true,"21":"white","22":1000,"23":1000,"24":"000003e803e8","25":"000e0d0000000000000000c80000","26":0,"34":false}}'
DEBUG:tinytuya.core:payload type = <class 'str'>
[... cut ...]
DEBUG:tinytuya.core:decoded results='{"dps":{"20":true,"21":"white","22":1000,"23":1000,"24":"000003e803e8","25":"000e0d0000000000000000c80000","26":0,"34":false}}'
DEBUG:tinytuya.core:status() received data={'dps': {'20': True, '21': 'white', '22': 1000, '23': 1000, '24': '000003e803e8', '25': '000e0d0000000000000000c80000', '26': 0, '34': False}}
{'dps': {'20': True, '21': 'white', '22': 1000, '23': 1000, '24': '000003e803e8', '25': '000e0d0000000000000000c80000', '26': 0, '34': False}}
```

About all I can recommend is make sure the devices are as close together as possible, and double check you're using the correct values from your last cut.

Cossid, 2023-04-03:
Well, I ran it a second time, and found it did not connect to my local wifi.  It appears tinytuya acknowledged the setting of wifi credentials, but it never took.  As a result, it was fully cut, but it was still trying to connect to the `cloudcutterflash` AP instead of my local wifi.

I'll have to see if I can improve the logic around sending the wifi join command via tinytuya, but I'm not sure what can really be changed at this point.

At the very least, you run setup_ap.sh on your cutting machine, and in another session, communicate with the device via tinytuya manually.

Cossid, 2023-04-04:
So here is what I can clarify:

If your device goes through the exploit (get an A- prefix), but won't connect to the cloudcutterflash AP, you can use the Smart Life app to force it to connect to the cloudcutterflash AP, and it will put the device into activation mode, and because CloudCutter can handle that, it will activate and complete the process then (hopefully) connect to your wifi via a command delivered by tinytuya while still connected to the cloudcutterflash AP

Once the device has been activated, if you put it back into slow or fast blink mode and send credentials via Smart Life, it will connect to your wifi, but it will have put the device back into activation mode, which it will never finish because even though it is on your network, it has been completely cut from Tuya's service, and expects to find the address cloudcutter set as an alternative.  It actually tries the alternate Tuya servers, but it fails because the security values have been changed and don't match what Tuya has on record anymore, leaving the device stuck in the activate mode.

The only solution, really, appears to be to cut the device again until it works.  The alternative is just too much work it looks like.

I have tried extending the tinytuya timeout values a bit to see if that helps any.

Cossid, 2023-04-04:
> So I took some steps to try to prevent NetworkManager from automatically connecting to my home network, and things went much more smoothly after that.
> 
> So this could be classified as "operator error" I guess. Or maybe `tuya-cloudcutter.sh` could try to detect unexpected changes happening to the Wi-Fi device? Up to you guys.

There is already the `-r` flag that likely helps with that situation, but it's a bit 'brute-force' in that it just removes all network manager profiles, so you'd have to re-add any you normally would use.

> Now to just figure out how to make that firmware do something to control these bulbs...

https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/27effd6b3d5d09c2e49655335e4d01a5ce9d97e5/devices/lenovo-se-242dc-rgbct-bulb-v1.2.21.json#L160 has the pins you'll need to assign PWM values for, search for `_pin`, the letter prefix represents the color of the channel.

hackoder, 2023-05-17:
Wanted to leave a comment here if someone else runs into the same issue as me; I was running tuya-cloudcutter on Arch and was consistently hitting the same issue (no activation). I live booted off a Ubuntu live usb and was able to cut without any issues. 

The only difference that I could see in the script execution was that dnsmasq took a really long time to start up (~2 minutes?) on the arch install while pegging a cpu at 100%. Tried this on 4 different (!) arch installs in the house and had the exact same experience with all of them. My guess is that in those two minutes the bulb bails out of the activation flow which is why the cut wasn't succeeding. I'm still not sure why dnsmasq would behave differently inside a guest on Ubuntu vs Arch, but for now I'm just live booting off a usb to cut which is working as expected.