tuya-cloudcutter/tuya-cloudcutter #317: new binary from lightleak: Globe-Electric_50329-Smart-Plug.bin

bmarr, 2023-03-15:
[Globe-Electric_50329-Smart-Plug.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/files/10977217/Globe-Electric_50329-Smart-Plug.zip)

Tuya product number: 50207

The included ZIP file contains the 2MB binary Flash dump using LightLeak (on ESP-WROOM-32) and the CloudCutter Android app. The dumped device was already set up and working with Tuya Smart and tuya-local in Home Assistant.

The firmware version reported by Tuya Smart is 1.1.8 and it also said it was the most recent version. The version should match out of the box as I prefer to not apply Tuya updates. I have another never-powered smart plug still in the box that I could try to dump.

It took several attempts to get the full Flash dump as the device AP mode would sometimes freeze or timeout. After the dump succeeded the AP mode did not timeout.

I attempted to use the build_profile.py but it was only able to retrieve some information, and so could not follow through to the profile JSON outputs. The first run of build_profile.py crashed while it appeared to be looking for keys in the "storage file".

```
[+] Processing file='../../Globe-Electric_50329-Smart-Plug.bin' as Globe-Electric_50329-Smart-Plug
[+] Encrypted bin has already been extracted
[+] Haxomatic has already been run
[!] Storage file not found
[+] SDK: 1.0.2
[+] Device class: oem_bk7231s_rnd_switch
[!] uuid was not found
[!] Unable to build complete profile as schema remains missing.

```
I could follow with a separate dump of the "storage" file from the CloudCutter Android app if needed - the instructions only required the Flash file.

Cossid, 2023-03-15:
> I attempted to use the build_profile.py but it was only able to retrieve some information, and so could not follow through to the profile JSON outputs. The first run of build_profile.py crashed while it appeared to be looking for keys in the "storage file".

Can you provide the crash/failure log?  I'm guessing you only had bk7231tools installed but not the optional encryption packages, or you were running from a platform that doesn't support the encryption methods (32bit arm)

Everything needed was present, a profile has been added as `Globe Electric 50329 Smart Plug`

bmarr, 2023-03-15:
Thanks for adding the profile, that was quicker than I anticipated!

I had indeed followed instructions and used `pip install .` before using build_profile.py. Today I followed your suggestion and tried `pip install bk7231tools[cli]` but that only reported that it was already installed.

I am running on a 64 bit x86_64 system (core2duo laptop with 4GB RAM, booted to Kali Linux installed on hard drive):
```
$ uname -a
Linux kali-linuxnx7400 6.1.0-kali5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.12-lkali1 (2023-02-20) x86_64 GNU/Linux
```
The python version used is:
`Python 3.11.1 (main, Dec 31 2022, 10:23:59) [GCC 12.2.0] on linux`

The first-pass output from build_profile.py is:
```
[+] Processing file='../../Globe-Electric_50329-Smart-Plug.bin' as Globe-Electric_50329-Smart-Plug
RBL containers:
	0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xdd20]
		extracted to /home/kali/Globe-Electric_50329-Smart-Plug
	0x129f0a: app - [encoding_algorithm=NONE, size=0xeb7a0]
		extracted to /home/kali/Globe-Electric_50329-Smart-Plug
Storage partition:
Traceback (most recent call last):
  File "/home/kali/tuya-cloudcutter/profile-building/build_profile.py", line 44, in <module>
    extract.run(file)
  File "/home/kali/tuya-cloudcutter/profile-building/extract.py", line 50, in run
    bk7231tools.__main__.dissect_dump_file(input)
  File "/usr/local/lib/python3.11/dist-packages/bk7231tools/__main__.py", line 240, in dissect_dump_file
    keys = storage.find_all_keys()
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bk7231tools/analysis/storage.py", line 180, in find_all_keys
    self.sf_find_index("abcdef")
  File "/usr/local/lib/python3.11/dist-packages/bk7231tools/analysis/storage.py", line 251, in sf_find_index
    label.label_12
AttributeError: 'NoneType' object has no attribute 'label_12'

```

Cossid, 2023-03-15:
Hmm, i'm not familiar with that code.  It's not an error with encryption, but rather not understanding goto instructions.  I'm going to close this issue, as all CloudCutter-related issues are resolved.  Please create an issue under bk7231tools ( https://github.com/tuya-cloudcutter/bk7231tools/issues ) with the details you posted above.

closed 2023-03-15 by Cossid