tuya-cloudcutter/tuya-cloudcutter #318: New Device - Feit Electric BR30 650lm RGBW Bulb (BR30/RGBW/CA/AG)

bmcsorley, 2023-03-16:
I'm newish to Git and not sure how to add a new profile to devices page.

But I was able to successfully flash my Fiet BR30/RGBW/CA/AG bulb using an existing profile for a similar bulb.

The feit-om60-rgbw-ca-ag-3-bulb works for this bulb.

Cossid, 2023-03-16:
We only add new devices with verified schemas (usually consisting of a full firmware dump from the device).  The `By firmware version and name` option exists for devices like this we can't verify a schema for, we do not add new devices based on flashing compatibility (because it may not work for just cutting).

closed 2023-03-16 by Cossid

bmcsorley, 2023-03-16:
Thank you for the information.

I tried to get a dump via lightleak but was unsuccessful, and don't think I can connect via uart without destroying the bulb.

I will try lightleak on another bulb.  I have 8 of them.

Thank you for all your hard work on this though!

bmcsorley, 2023-03-17:
@Cossid.  

Since there is a profile similar enough that I can write new firmware, would it be possible to also read/dump the flash after the exploit runs during the cloudcutter process since the device ID and local key work?

Cossid, 2023-03-17:
@bmcsorley Unfortunately, no, you're still cutting along the way to process to flash firmware, so it writes the schema of whatever device profile you used to flash, and the device id/local key are the ones generated by CloudCutter that do not work for activating externally with Smart Life.  It would only work for devices that were first activated on Smart Life and then serially flashed, at which point, you could have just gotten a serial dump first anyway.