tuya-cloudcutter/tuya-cloudcutter #323: new binary from bk7231tools read_flash: Globe-Electric_37783-RGBCWWW-A19-Smart-LED

bmarr, 2023-03-20:
[Globe-Electric_37783-RGBCWWW-A19-Smart-LED.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11014654/Globe-Electric_37783-RGBCWWW-A19-Smart-LED.zip)

The included ZIP file contains the 2MB binary Flash dump using the bk7231tools and a FTDI USB cable.  The dumped device was straight from the Globe packaging (from Costco, Canada) to the workbench, so it has never been set up with an AP and the Tuya Smart app.

However, I did first verify that it was exploitable using the LightLeak AP and the Android Cloudcutter app, though when it got to the download flash screen, the error message was "couldn't receive packets from the device".

The firmware version reported by Tuya Smart on other identical bulbs from this same package is v2.0.3, but I can't be certain these ones had not had firmware updates applied at some point.

I attempted to use the build_profile.py in several passes but it was only able to retrieve bootloader and application binaries. No JSON output was generated. In the second pass it reported:

`[+] Matched pattern for BK7231T version SDK 2.0.0 8710_2M, payload type datagram`

And haxomatic failed with:

`[!] Failed to find datagram payload address (found 2, expected 3)`

Cossid, 2023-03-20:
So this is an interesting one.

I can confirm that the dump you provided is version 2.0.0 and there is an update available to 2.0.3
You received an error from haxomatic because this build has an older baseline version than the versions supported in haxomatic.  I'll look to see if I can find the haxomatic offsets for this baseline or reach out for help if needed, but that may take some time.

Additionally, using the 2.0.0 firmware, I cannot get this device to connect to my network in any way, it continually fails with Tuya giving the message "net work unavailable" over and over.  I tried with both the Globe Suite app as well as Smart Life, both failed the same (though Global Suite wouldn't even let me attempt to add via slow blink AP mode, it would crash).

I can confirm that the 2.0.3 firmware is still exploitable, and if you can update your devices to that, go ahead.  I will post a 2.0.3 profile shortly.

Cossid, 2023-03-20:
A profile for `Globe Electric 37783 RGBCT Bulb v2.0.3` has been added, I'll continue to look into the v2.0.0

Cossid, 2023-03-20:
A profile for `Globe Electric 37783 RGBCT Bulb v2.0.0` has been added.  Haxomatic has been updated to work with Baseline 30.04 as well.

Even with cutting the 2.0.0 firmware will not connect to my network, not sure what's up with that.  2.0.3 seems to connect fine, so 2.0.0 may just be a buggy release.  Either way, it will 3rd party flash now with CloudCutter.

closed 2023-03-20 by Cossid

MRobi1, 2023-04-26:
This may not be the best spot to ask but I've done some searching and haven't found it.
Wondering if either of you have a LibreTuya config for this bulb? The ESPHome config uses the esp8266_pwm platform which I assume doesn't directly convert.

Cossid, 2023-04-26:
https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/a877e69c8f3829be545e9a5a9d76cab9d5fd7fd2/devices/globe-electric-37783-rgbct-bulb-v2.0.3.json#L109

Looks like it's probably SM2135 LED driver, you'll need to add external component with ESPHome PR #3850

```yaml
external_components:
  - source: github://pr#3850
    components: [ sm2135 ]
    refresh: 10 min

sm2135:
  data_pin: P26
  clock_pin: P8
  rgb_current: 20mA
  cw_current: 50mA

output:
  - platform: sm2135
    id: output_red
    channel: 2
  - platform: sm2135
    id: output_green
    channel: 1
  - platform: sm2135
    id: output_blue
    channel: 0
  - platform: sm2135
    id: output_coldwhite
    channel: 3
  - platform: sm2135
    id: output_warmwhite
    channel: 4

light:
  - platform: rgbww
    name: "Light"
    red: output_red 
    green: output_green 
    blue: output_blue 
    cold_white: output_coldwhite 
    warm_white: output_warmwhite 
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    color_interlock: true
```

White channels may be mixed up, just switch channels 3/4 if so.

MRobi1, 2023-05-05:
White channels were good, blue and green were reversed.
I also found when power was restored they booted to red by default so I added a on_turn_on command to force them into white. Here's what I ended up with for future searchers.

Any idea if this would work with the globe-electric BR30 bulbs as well? I can't see the internals being much different than the A19 bulbs.
```
external_components:
  - source: github://pr#3850
    components: [ sm2135 ]
    refresh: 10 min

sm2135:
  data_pin: P26
  clock_pin: P8
  rgb_current: 20mA
  cw_current: 50mA

output:
  - platform: sm2135
    id: output_red
    channel: 2
  - platform: sm2135
    id: output_green
    channel: 0
  - platform: sm2135
    id: output_blue
    channel: 1
  - platform: sm2135
    id: output_coldwhite
    channel: 3
  - platform: sm2135
    id: output_warmwhite
    channel: 4

light:
  - platform: rgbww
    name: "Light"
    id: globelight1
    red: output_red 
    green: output_green 
    blue: output_blue 
    cold_white: output_coldwhite 
    warm_white: output_warmwhite 
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    color_interlock: true
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
      light.control:
        id: globelight1
        state: on
        color_mode: COLD_WARM_WHITE
```

Cossid, 2023-05-06:
Oh right, I forgot the channel orders on the SM2135 driver are out of order, so yeah, you just have to change channels until it works.

As for the on color, that's most likely due to the last saved preference.  If you changed it to your preferred color, then turned it off for 10 minutes, it would re-save the preference.  But the on_turn_on also works fine.

As for the BR30s, best answer I can give is maybe.  Depends on which firmware they come with.  Also, most BR30 bulbs I've seen have been PWM.