tuya-cloudcutter/tuya-cloudcutter #325: [!] The profile you selected did not result in a successful exploit.

fabio-s-franco, 2023-03-21:
I am going insane trying to figure out what's wrong. Tried many things, followed instructions to the letter, reflashed the raspberry pi 4b several times as I tinkered with configurations, but I am about to give up, but really didn't want to. Hopefully you guys have some idea on what I could try.

So, my setup is as follows:
- Raspberry Pi 4B, 4GB RAM, connected to my home network via ethernet
- I have several light bulbs that match to the profiles: oem-bk7231s-light-ty-2.9.16-sdk-1.0.8-40.00
- I have two variations of those bulbs, and I can confirm both have firmware version 2.9.16. The models match exactly the ones with the slugs lsc-2578539-970724-white-and-color-ambiance-e27-806lumen and lsc-3000272-970716-tunable-white-e27-806lumen
- My goal is to flash the bulb with [OpenBK7231T_UG_1.15.621.bin](https://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.15.621/OpenBK7231T_UG_1.15.621.bin), but trying to release it to my network renders the same results, so I am just focusing on trying to flash this thing.
- I connected first with external mouse, keyboard and montior, but now I am on ssh, it's just more practical.

To keep things simple, I simply followed the interactive menu, which ensures I am not making any booboo with the cli parameters. So just run sudo cloud-cutter.sh (after setting up the Pi exactly as instructed).
The process goes smoothly, until the time I need to power cycle off and on again, so maybe I am doing something wrong at this stage, but everything seems to going nicely. The bulb freezes, after sometimes it reboots itself and enters AP mode again. When the script connects to it again, that's when the failure occurs with the error: `[!] The profile you selected did not result in a successful exploit.`

I tried a variety of alternate profiles, without success (although some of them wouldn't even cause the bulb to freeze and reboot). I tried letting the bulb reboot itself. Tried turning it off, then on again (it re-enters AP mode without having to do anything extra). Also tried forcing it get out of AP mode, then back. All with the same result.

Here is an example output that I can repeat over and over again:

```
 sudo ./tuya-cloudcutter.sh
Building cloudcutter docker image
[+] Building 295.3s (13/13) FINISHED
 => [internal] load build definition from Dockerfile                                                               0.1s
 => => transferring dockerfile: 457B                                                                               0.0s
 => [internal] load .dockerignore                                                                                  0.1s
 => => transferring context: 2B                                                                                    0.0s
 => [internal] load metadata for docker.io/library/python:3.9-slim-buster                                          1.7s
 => [base 1/2] FROM docker.io/library/python:3.9-slim-buster@sha256:ef30644d6bf6f30a2efceac16e48b25c0cd8af0398bb  11.4s
 => => resolve docker.io/library/python:3.9-slim-buster@sha256:ef30644d6bf6f30a2efceac16e48b25c0cd8af0398bb5d4dd4  0.0s
 => => sha256:0f4365aec03b6f4d8a22da20cd2e75fb2814c403828af44983ccf46c7e1586c8 10.43MB / 10.43MB                   1.2s
 => => sha256:ef30644d6bf6f30a2efceac16e48b25c0cd8af0398bb5d4dd43c2ef2d241e1bc 988B / 988B                         0.0s
 => => sha256:5067e690991fc8c55ca1a70600fc2ac3e3784cc3185bc902f5b13aa36ae9c9e8 1.37kB / 1.37kB                     0.0s
 => => sha256:b3c9f6209e764d9636a468ef8825d92e08582b47c61769e3b2b5eea52ad0a6c9 7.92kB / 7.92kB                     0.0s
 => => sha256:7138f15de53476a9a56af46eac63ea2f39c78b20aa9ad61ce3018e0eddb748f8 22.75MB / 22.75MB                   3.2s
 => => sha256:3263bcfdfa8cf12a4cdb3701ea31c7cc9e9b61291cbf7f31c3412cb6a1ea737e 2.37MB / 2.37MB                     0.7s
 => => sha256:282d6f755df1142a03dfbe41c47edf702f5da81db297268dd49eecfc96550d75 235B / 235B                         1.3s
 => => sha256:7e12a3594b78bbee8e759c855b31dd76dba471750a6d360f904dd832f4b2b9cb 3.19MB / 3.19MB                     2.2s
 => => extracting sha256:7138f15de53476a9a56af46eac63ea2f39c78b20aa9ad61ce3018e0eddb748f8                          2.8s
 => => extracting sha256:3263bcfdfa8cf12a4cdb3701ea31c7cc9e9b61291cbf7f31c3412cb6a1ea737e                          0.4s
 => => extracting sha256:0f4365aec03b6f4d8a22da20cd2e75fb2814c403828af44983ccf46c7e1586c8                          1.3s
 => => extracting sha256:282d6f755df1142a03dfbe41c47edf702f5da81db297268dd49eecfc96550d75                          0.0s
 => => extracting sha256:7e12a3594b78bbee8e759c855b31dd76dba471750a6d360f904dd832f4b2b9cb                          0.8s
 => [internal] load build context                                                                                  0.2s
 => => transferring context: 831.24kB                                                                              0.1s
 => [base 2/2] RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq   47.0s
 => [python-deps 1/4] RUN pip install --upgrade pipenv                                                            20.9s
 => [python-deps 2/4] COPY src/Pipfile /src/                                                                       0.2s
 => [python-deps 3/4] COPY src/Pipfile.lock /src/                                                                  0.2s
 => [python-deps 4/4] RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy                            206.8s
 => [cloudcutter 1/2] COPY src /src                                                                                0.3s
 => [cloudcutter 2/2] WORKDIR /src                                                                                 0.2s
 => exporting to image                                                                                             6.5s
 => => exporting layers                                                                                            6.5s
 => => writing image sha256:e70cded68e3375de2d0f608a30a7ba1302ecdd81b5016be5db36ce9a5e348bc5                       0.0s
 => => naming to docker.io/library/cloudcutter                                                                     0.0s
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] Select your custom firmware file: OpenBK7231T_UG_1.15.621.bin
 > OpenBK7231T_UG_1.15.621.bin

[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 2.9.16 - BK7231T / oem_bk7231s_light_ty
   2.0.3 - BK7231T / oem_bk7231s_light5_sm2135e_yamao
   2.0.6 - BK7231T / WB3S_RMD_Night_Light
   2.1.1 - BK7231T / HYS_03_018_bk7231t_r1e1zch4rds9ypqq
[... 95 lines cut ...]
Checking TCP port 8886... Available.
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wlan0
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr d8:3a:dd:02:79:12 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Using PSK v1 - Received PSK ID version 02
SSL Error on 15 ('10.42.42.21', 56640): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
SSL Error on 15 ('10.42.42.21', 56641): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
SSL Error on 15 ('10.42.42.21', 56642): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
....# this keeps going until eventually freezes
```
I tried fiddling with what I could find on the internet to try the only thing I thought may be wrong with my setup:` wlan0: Could not connect to kernel driver`
But causes, symptoms and circumstances were all over the place, nothing really helped.
Maybe chat GPT has some idea, because I am out of them, or hopefully you guys can put me out my misery and I will focus on a plumbing career or something.
I also very much appreciate the work put into this. This thing is so tiny I can't get around to solder a wire to the impossibly tiny UART ports these chips have. I don't know how people succeed doing this manually, should be brain surgeons or something. So having this work OTA, would be amazing.

Cossid, 2023-03-21:
Unfortunately, I don't have any solid advice for you.

> If I comment out this function on common-run.sh:

Commenting out the successful exploit check will be of no help.  If that fails, everything after is irrelevant, it would fail exactly as you have shown.

> I tried fiddling with what I could find on the internet to try the only thing I thought may be wrong with my setup:` wlan0: Could not connect to kernel driver`

That is a message you can ignore, all users see that, it's a message that comes from hostapd that we cannot control, but apparently doesn't mean anything useful.

The only advice I can really give is to double-check the firmware version in Smart Life.  Beyond that, you can try wirelessly dumping the firmware with LightLeak ( https://github.com/tuya-cloudcutter/lightleak ) if you're able.  It is possible your device shares a firmware version but has a unique device class, though rather unlikely for a light bulb.

fabio-s-franco, 2023-03-21:
Thanks for the clarifications, I will definitely give it a shot. I did perform the exploit test and the result was "it's almost definitely exploitable.". So really annoys me to always find myself on the short end of the stick, so I will put some effort to dodge this one. Will come back to it here.

fabio-s-franco, 2023-03-21:
@Cossid Not a lot of luck I must say... This bulb is beating me to the corner. On lightleak it fails on the last stage, everything goes right.

[image]

fabio-s-franco, 2023-03-21:
I can't believe it. In the last random profile I tried, I got it working (at least with the white light only, not the RGB, this still gonna try).

So this is the bulb model.LSC E27 Tunnable White. With the model number as in the other profile. But the profile that actually worked is this one:

2578539 970719 RGBCT Bulb 2.9.27

So there must be a mismatch in definitions of LSC bulbs. Nevertheless these are sold in the Netherlands ([Action](https://www.action.com/nl-nl/p/lsc-smart-connect-slimme-ledlamp3/)).

[image]

How can we make this so others could be spared of my cruzade?

Cossid, 2023-03-21:
That means it has 2.9.27 firmware on it, and Smart Life should have reported it as such.  I'm not sure how you concluded that it was 2.9.16 aside from finding a model that matched and assuming they were the same, but that is not always the case, that is why we have instructions for verifying your firmware version with Smart Life.

Without a verified dump, we generally don't add new profiles because of the differences between cutting and flashing.  LSC doesn't offer an upgrade from 2.9.16 to 2.9.27, so they could be different hardware/software configurations.

closed 2023-03-21 by Cossid

fabio-s-franco, 2023-03-21:
I just tried every profile and this was the last one that I tried. I have another exact same bulb still ins smart life:

[image]

I guess information there is either unreliable or this bulbs just come randomly flashed.

Cossid, 2023-03-21:
I'm not sure what to tell you.  Without a firmware dump, there isn't a lot more we can do without guessing.  It's possible LSC registered these devices as 2.9.16 with Smart Life despite shipping with 2.9.27.  When a device is joined to Smart Life, it sends it's firmware version to Tuya, and that's part of what determines what devices have upgrades available or not.  LSC doesn't offer an upgrade path from 2.9.16 to 2.9.27.  They seem to only support the version that comes shipping on the bulb and never upgrade OTA.

fabio-s-franco, 2023-03-21:
I would add dumps for the other ones, so it can be verified, but it simply doesn't work. I go through the entire process within the cloud cutter app, but when it's time to read flash it errors. Tried multiple times. Not sure if the fact that I only have esp8266 d1 mini mcu's could be the cause. If when it tries to pull the firmware it tries to store it in the mcu, it will likely not work as it is very limited on that matter. I could try something else, if that is the case, but will need some time to get find proper stuff for this.

fabio-s-franco, 2023-03-21:
>  They seem to only support the version that comes shipping on the bulb and never upgrade OTA.

That definitely seems to be the case, I never seen it upgrade, even when I had the auto upgrade enabled before started tinkering with this stuff

Cossid, 2023-03-21:
LightLeak isn't perfect or universal, it just doesn't work on some devices, not much we can do about that.  It never stores anything on the intermediate device, all firmware backups are directly to the phone.  Until we get a successful lightleak dump or a serial dump, nothing is going to change.  This thread is all we can do to document that some LSC bulbs might indicate an incorrect version.  This isn't something we've seen before.

fabio-s-franco, 2023-03-21:
I completely understand, too many variables to account for. And as I mentioned before, I am often "lucky" with the short end of the stick of things, but on the other end, I am very stubborn, so whatever I figure out as I go I will try to share here. If eventually I manage to get a dump of these, I will open an issue in the other repo (or maybe even try to profile it myself). Thanks for the help anyway. Got to have some fun with more tinkering.

kuba2k2, 2023-03-21:
Maybe the 2.9.27 profile has wrong version in its description? Or you mistakenly thought that 2.9.27 worked after having tried several profiles one after another, but the one that actually cut your bulb is 2.9.16?

Cossid, 2023-03-21:
> Maybe the 2.9.27 profile has wrong version in its description?

Nope
The 2.9.27 profile is correct and belongs to 2.9.27, against SDK 2.3.2
The 2.9.16 profile is also correct an belongs to 2.9.16, against SDK 1.0.8
They are not mixed up, and can't be, they use different address patterns.

I'm a bit more confused now than before.

When you join a device to Smart Life, the device sends the current version to Tuya and that *is* the version Smart Life will display on the `Device Update` screen.  I just tested it by sending a completely custom version number and it displayed it, so it's not something Tuya is saving ahead of activation.

That can only mean the firmware on his device has the version number wrong and it's being sent to Tuya like that.  Firmware version is stored in 3 total places (that I'm aware of);
2 in storage as `gw_di.swv` and `gw_di.dev_swv`, and once in the firmware itself (There is a separate place in the rbl table layout, but it's always 1.00 there).  I'm not sure which one of those 3 it sends with the active command, and once again, only having the actual firmware could confirm.