tuya-cloudcutter/tuya-cloudcutter #335: Add better detection for patched devices

no2chem, 2023-03-28:
I bought some new Lumary "Smart Disk" Lights:

https://www.amazon.com/Lumary-Dimmable-Recessed-Retrofit-Assistant/dp/B0BBLXPYGZ

Some of the lights in the package were an older firmware oem_bk7231n_light_ty 1.3.21

Unfortunately, it seems that the other lights have a new firmware 1.5.10. I dumped the firmware, which contains a string with TuyaOS in it:
```
 strings Lumary-DiskLight_US---DS6B--1.5.10-RGBCT-Bulb_app_1.00_decrypted.bin | grep Tuya
TuyaIoT-OS
< TuyaOS V:3.3.2 BS:40.00_PT:2.3_LAN:3.5_CAD:1.0.5_CD:1.0.0 >
```

I *think* this firmware is probably patched. `haxomatic.py` should say so and abort.

Cossid, 2023-03-28:
Can you please attach a dump so we can verify?  This is the first we've seen of 3.x.x SDK, we do not yet know if it's patched or not.

Cossid, 2023-03-28:
Yeah, after a preliminary look, this device appears patched.  I have added a stop condition to haxomatic.

Cossid, 2023-03-31:
Re-confirmed TuyaOS V:3.* is not vulnerable, and haxomatic now has a proper stop for that case, closing this ticket.  Thank you for supplying the dump!

closed 2023-03-31 by Cossid