tuya-cloudcutter/tuya-cloudcutter #340: dry relay CB3S SWT10 OXT 240V 150W 5A

openshwprojects, 2023-03-28:
Hello, this is a device where you can connect 12V or any other circuit, the relay output is separated from mains. 
[ SWT10 OXT 240V 150W 5A .zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/11093930/SWT10.OXT.240V.150W.5A.zip)
Thanks!

PS: don't worry, only one more device left in this flash dumps batch!

Cossid, 2023-03-28:
Can you provide some product info for this one, can't seem to find/verify this one.

Cossid, 2023-03-28:
Nevermind, I found it.  Unfortunately, this device is patched an not vulnerable to CloudCutter

Device configuration is as follows:
```json
"user_param_key": {
		"ch_cddpid1": 9,
		"ch_dpid1": 1,
		"ch_num": 1,
		"clean_t": 6,
		"crc": 66,
		"cyc_dpid": 43,
		"inch_dp": 44,
		"init_conf": 38,
		"jv": 102,
		"module": "CB3S",
		"net_trig": 4,
		"net_type": 0,
		"netled1_lv": 1,
		"netled1_pin": 9,
		"netn_led": 0,
		"nety_led": 0,
		"on_off_cnt": 10,
		"onoff1": 8,
		"onoff_clear_t": 10,
		"onoff_n": 5,
		"onoff_rst_m": 1,
		"onoff_rst_type": 2,
		"onoff_type": 0,
		"rand_dpid": 42,
		"remote_add_dp": 49,
		"remote_list_dp": 50,
		"remote_select": 0,
		"reset_t": 3,
		"rl1_lv": 1,
		"rl1_pin": 6,
		"series_ctrl": 0,
		"total_bt_lv": 0,
		"total_bt_pin": 26,
		"total_stat": 2,
		"zero_select": 0
	}
```

closed 2023-03-28 by Cossid

openshwprojects, 2023-03-28:
That's interesting. The device was never connected to the cloud. So basically it's from the batch so new that it actually has updated SDK version?

Cossid, 2023-03-28:
Yes, this must be a newly produced device.  The SDK date is `2022_12_07_15_29_01` and the firmware compile date is `Dec 12 2022 14:09:13`